Hamlib/Hamlib #227: rig_get_channel inconsistent

mdblack98, 2020-04-26:
rig_get_channel is inconsistently implemented.
Some backends update the rig and some don't.
So we will add a read_only flag to the call so the client can ask for a channel and leave the rig as it was. 
Each backend will have to allow for read only and restoring the rig values if needed.
For the moment just the flag has been added and a number of #warn comments inserted about backends that need more changes.
We will also add a flag to rigctl/rigctld on the "h" command to allow read only.  So "h 2 0" will read channel 2 and update the rig and "h 2 1" will read the channel and leave the rig as it was.

dl8fcl, 2020-04-30:
This commit is breaking the build with lua-bindings and python-bindings enabled

mdblack98, 2020-04-30:
Update and try again...should be fixed now..
de Mike W9MDB

closed 2020-06-15 by mdblack98