Hamlib/Hamlib #230: TS-590S Power On on startup missing

hlnurse, 2020-04-30:
It appears the code to power up the TS-590S is missing.  Hamlib 4.0 latest.  Log is below.

Howard W6HN

--------------------------
pi@rigpi4:~ $ sudo rigctl -m 2031 -r /dev/ttyUSB0 -t 4532 -s 115200 -vvvvv -Z
2020-04-30:19:00:46.648225: rigctl, Hamlib 4.0~git Apr 27 2020 15:16:32
2020-04-30:19:00:46.648456: Report bugs to <hamlib-developer@lists.sourceforge.net>
2020-04-30:19:00:46.648683: rig_init called
2020-04-30:19:00:46.648996: initrigs4_kenwood called
2020-04-30:19:00:46.649168: rig_register called
2020-04-30:19:00:46.649276: rig_register: rig_register (2012)
2020-04-30:19:00:46.649414: rig_register called
2020-04-30:19:00:46.649546: rig_register: rig_register (2013)
2020-04-30:19:00:46.649651: rig_register called
2020-04-30:19:00:46.649749: rig_register: rig_register (2001)
2020-04-30:19:00:46.649853: rig_register called
2020-04-30:19:00:46.649954: rig_register: rig_register (2025)
2020-04-30:19:00:46.650056: rig_register called
2020-04-30:19:00:46.650157: rig_register: rig_register (2003)
2020-04-30:19:00:46.650258: rig_register called
2020-04-30:19:00:46.650358: rig_register: rig_register (2004)
2020-04-30:19:00:46.650460: rig_register called
2020-04-30:19:00:46.650571: rig_register: rig_register (2016)
2020-04-30:19:00:46.650677: rig_register called
2020-04-30:19:00:46.650772: rig_register: rig_register (2024)
2020-04-30:19:00:46.650874: rig_register called
2020-04-30:19:00:46.650975: rig_register: rig_register (2005)
2020-04-30:19:00:46.651075: rig_register called
2020-04-30:19:00:46.651176: rig_register: rig_register (2007)
2020-04-30:19:00:46.651278: rig_register called
2020-04-30:19:00:46.651379: rig_register: rig_register (2009)
2020-04-30:19:00:46.651504: rig_register called
2020-04-30:19:00:46.651589: rig_register: rig_register (2010)
2020-04-30:19:00:46.651696: rig_register called
2020-04-30:19:00:46.651789: rig_register: rig_register (2022)
2020-04-30:19:00:46.651892: rig_register called
2020-04-30:19:00:46.651975: rig_register: rig_register (2014)
2020-04-30:19:00:46.652081: rig_register called
2020-04-30:19:00:46.652180: rig_register: rig_register (2030)
2020-04-30:19:00:46.652281: rig_register called
2020-04-30:19:00:46.652382: rig_register: rig_register (2021)
2020-04-30:19:00:46.652493: rig_register called
2020-04-30:19:00:46.652605: rig_register: rig_register (2029)
2020-04-30:19:00:46.652716: rig_register called
2020-04-30:19:00:46.652808: rig_register: rig_register (2043)
2020-04-30:19:00:46.652912: rig_register called
2020-04-30:19:00:46.653013: rig_register: rig_register (2044)
2020-04-30:19:00:46.653116: rig_register called
2020-04-30:19:00:46.653217: rig_register: rig_register (2045)
2020-04-30:19:00:46.653319: rig_register called
2020-04-30:19:00:46.653405: rig_register: rig_register (2038)
2020-04-30:19:00:46.653519: rig_register called
2020-04-30:19:00:46.653626: rig_register: rig_register (2002)
2020-04-30:19:00:46.653728: rig_register called
2020-04-30:19:00:46.653829: rig_register: rig_register (2011)
2020-04-30:19:00:46.653931: rig_register called
2020-04-30:19:00:46.654032: rig_register: rig_register (2006)
2020-04-30:19:00:46.654133: rig_register called
2020-04-30:19:00:46.654233: rig_register: rig_register (2008)
2020-04-30:19:00:46.654333: rig_register called
2020-04-30:19:00:46.654492: rig_register: rig_register (2015)
2020-04-30:19:00:46.654582: rig_register called
2020-04-30:19:00:46.654665: rig_register: rig_register (2026)
2020-04-30:19:00:46.654769: rig_register called
2020-04-30:19:00:46.654877: rig_register: rig_register (2017)
2020-04-30:19:00:46.654972: rig_register called
2020-04-30:19:00:46.655056: rig_register: rig_register (2033)
2020-04-30:19:00:46.655161: rig_register called
2020-04-30:19:00:46.655261: rig_register: rig_register (2042)
[... 206 lines cut ...]
2020-04-30:19:00:51.683299: kenwood_transaction: read_string(len=0)=''
2020-04-30:19:00:51.683337: kenwood_transaction: read_string retval < 0, retval = -5, retry_read=8, retry=10
2020-04-30:19:00:51.683376: kenwood_transaction: cmdstr = FV
2020-04-30:19:00:51.683413: serial_flush called
2020-04-30:19:00:51.683446: serial_flush: tcflush
2020-04-30:19:00:51.683483: write_block called
2020-04-30:19:00:51.683541: write_block(): TX 3 bytes
2020-04-30:19:00:51.683580: 0000    46 56 3b                                            FV;             
2020-04-30:19:00:51.683615: read_string called
2020-04-30:19:00:52.184216: read_string(): Timed out 0.500546 seconds after 0 chars
2020-04-30:19:00:52.184279: kenwood_transaction: read_string(len=0)=''
2020-04-30:19:00:52.184321: kenwood_transaction: read_string retval < 0, retval = -5, retry_read=9, retry=10
2020-04-30:19:00:52.184361: kenwood_transaction: cmdstr = FV
2020-04-30:19:00:52.184400: serial_flush called
2020-04-30:19:00:52.184434: serial_flush: tcflush
2020-04-30:19:00:52.184476: write_block called
2020-04-30:19:00:52.184534: write_block(): TX 3 bytes
2020-04-30:19:00:52.184574: 0000    46 56 3b                                            FV;             
2020-04-30:19:00:52.184611: read_string called
2020-04-30:19:00:52.685210: read_string(): Timed out 0.500546 seconds after 0 chars
2020-04-30:19:00:52.685269: kenwood_transaction: read_string(len=0)=''
2020-04-30:19:00:52.685308: kenwood_transaction: read_string retval < 0, retval = -5, retry_read=10, retry=10
2020-04-30:19:00:52.685362: kenwood_transaction: returning retval=-5
2020-04-30:19:00:52.685405: kenwood_open: cannot get f/w version
2020-04-30:19:00:52.885538: main: error opening rig, try#1

mdblack98, 2020-04-30:
The "-t 4532" is wrong.  -t on rigctl is the terminator character so get rid of that switch.
It looks to me like the rig is turned on as it did eventually get a good response to the ID command so there was no powersup sequence tried....
But the rig's reponses don't look stable at all and may just because of the -t switch. 
de Mike W9MDB

hlnurse, 2020-05-01:
Mike, removing the -t 4532 command helped, thanks.  It now appears the TS-590s is not responding to the FV command preventing startup.  The radio is in the Kenwood power standby mode when the rigctl command is sent.  Since the ID number is being returned correctly, with no evidence of power up commands, something is strange (log snippet follows):

2020-05-01:16:06:12.659755: 0000    46 56 3b                                            FV;             
2020-05-01:16:06:12.659804: read_string called
2020-05-01:16:06:13.160417: read_string(): Timed out 0.500544 seconds after 0 chars
2020-05-01:16:06:13.160531: kenwood_transaction: read_string(len=0)=''
2020-05-01:16:06:13.160591: kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, retry=10

I've asked another Beta tester to try with his TS-590GS and will report back.

Howard W6HN

mdblack98, 2020-05-01:
This code has been in there for a long time....can you send me the complete log?  I just made a small change but it likely won't help....maybe it will.  Update andry again and send me the whole log.
Sure looks like the FV command used to work....
    if (RIG_IS_TS590S)    {        /* we need the firmware version for these rigs to deal with f/w defects */        static char fw_version[7];        char *dot_pos;
        err = kenwood_transaction(rig, "FV", fw_version, sizeof(fw_version));

hlnurse, 2020-05-01:
More clues.  Another Beta tester has a TS-590SG and I asked him to run rigctl from terminal.  The SG is not asked for FV, but it is asked for ID and properly returns the ID.  But when rigctl asks for IF, the request times out and startup fails.  When the rig prompt comes up, the radio is not on.  So it is returning ID while in the standby power mode and the ID request doesn't power on the radio.  Sending w PS1; from the prompt does turn on the power.  I don't want to confuse issues between the two radios, but the results are similar.  ID; returns the ID but power is not on.

The 590S user is in Sweden so getting feedback takes a bit longer.

Howard W6HN

[TS590SG-startup.zip](https://github.com/Hamlib/Hamlib/files/4566410/TS590SG-startup.zip)

mdblack98, 2020-05-01:
So when the rig is in standby does "PS;" return anything?
Mike

hlnurse, 2020-05-01:
With TS-590SG off, w PS; returns PS0; (from rig prompt:)
With power on,, w PS; returns PS1;

--Howard

hlnurse, 2020-05-02:
BINGO!  The TS-590S power on is now working as expected.  Thanks, Mike!

--Howard

 rigctl -m 2031 -r /dev/ttyUSB0 -s 115200 -vvvvv -Z
2020-05-02:21:43:42.643409: rigctl, Hamlib 4.0~git May  2 2020 14:41:20
2020-05-02:21:43:42.643548: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-05-02:21:43:42.643672: rig_init called
2020-05-02:21:43:42.643955: initrigs4_kenwood called
2020-05-02:21:43:42.644063: rig_register called
2020-05-02:21:43:42.644115: rig_register: rig_register (2012)
2020-05-02:21:43:42.644191: rig_register called
2020-05-02:21:43:42.644225: rig_register: rig_register (2013)
2020-05-02:21:43:42.644259: rig_register called
2020-05-02:21:43:42.644291: rig_register: rig_register (2001)
2020-05-02:21:43:42.644327: rig_register called
2020-05-02:21:43:42.644358: rig_register: rig_register (2025)
2020-05-02:21:43:42.644390: rig_register called
2020-05-02:21:43:42.644422: rig_register: rig_register (2003)
2020-05-02:21:43:42.644455: rig_register called
2020-05-02:21:43:42.644488: rig_register: rig_register (2004)
2020-05-02:21:43:42.644520: rig_register called
2020-05-02:21:43:42.644551: rig_register: rig_register (2016)
2020-05-02:21:43:42.644583: rig_register called
2020-05-02:21:43:42.644617: rig_register: rig_register (2024)
2020-05-02:21:43:42.644648: rig_register called
2020-05-02:21:43:42.644680: rig_register: rig_register (2005)
2020-05-02:21:43:42.644712: rig_register called
2020-05-02:21:43:42.644744: rig_register: rig_register (2007)
2020-05-02:21:43:42.644776: rig_register called
2020-05-02:21:43:42.644808: rig_register: rig_register (2009)
2020-05-02:21:43:42.644840: rig_register called
2020-05-02:21:43:42.644872: rig_register: rig_register (2010)
2020-05-02:21:43:42.644904: rig_register called
2020-05-02:21:43:42.644936: rig_register: rig_register (2022)
2020-05-02:21:43:42.644968: rig_register called
2020-05-02:21:43:42.644999: rig_register: rig_register (2014)
2020-05-02:21:43:42.645031: rig_register called
2020-05-02:21:43:42.645063: rig_register: rig_register (2030)
2020-05-02:21:43:42.645095: rig_register called
2020-05-02:21:43:42.645127: rig_register: rig_register (2021)
2020-05-02:21:43:42.645159: rig_register called
2020-05-02:21:43:42.645190: rig_register: rig_register (2029)
2020-05-02:21:43:42.645222: rig_register called
2020-05-02:21:43:42.645254: rig_register: rig_register (2043)
2020-05-02:21:43:42.645285: rig_register called
2020-05-02:21:43:42.645317: rig_register: rig_register (2044)
2020-05-02:21:43:42.645349: rig_register called
2020-05-02:21:43:42.645380: rig_register: rig_register (2045)
2020-05-02:21:43:42.645415: rig_register called
2020-05-02:21:43:42.645446: rig_register: rig_register (2038)
2020-05-02:21:43:42.645477: rig_register called
2020-05-02:21:43:42.645509: rig_register: rig_register (2002)
2020-05-02:21:43:42.645540: rig_register called
2020-05-02:21:43:42.645574: rig_register: rig_register (2011)
2020-05-02:21:43:42.645606: rig_register called
2020-05-02:21:43:42.645637: rig_register: rig_register (2006)
2020-05-02:21:43:42.645669: rig_register called
2020-05-02:21:43:42.645700: rig_register: rig_register (2008)
2020-05-02:21:43:42.645732: rig_register called
2020-05-02:21:43:42.645763: rig_register: rig_register (2015)
2020-05-02:21:43:42.645795: rig_register called
2020-05-02:21:43:42.645827: rig_register: rig_register (2026)
2020-05-02:21:43:42.645859: rig_register called
2020-05-02:21:43:42.645893: rig_register: rig_register (2017)
2020-05-02:21:43:42.645925: rig_register called
2020-05-02:21:43:42.645956: rig_register: rig_register (2033)
2020-05-02:21:43:42.645988: rig_register called
2020-05-02:21:43:42.646020: rig_register: rig_register (2042)
[... 191 lines cut ...]
2020-05-02:21:43:43.730255: elapsed_ms: elapse_secs=-1.58846e+06
2020-05-02:21:43:43.730298: kenwood_transaction: returning retval=0
2020-05-02:21:43:43.730338: kenwood_get_split_vfo_if: priv->tx_vfo=VFOA
2020-05-02:21:43:43.730376: kenwood_open: priv->tx_vfo=VFOA
2020-05-02:21:43:43.730419: rig_get_vfo called
2020-05-02:21:43:43.730453: elapsed_ms: start = 0,0
2020-05-02:21:43:43.730490: rig_get_vfo: cache check age=1000000ms
2020-05-02:21:43:43.730526: rig_get_vfo: cache miss age=1000000ms
2020-05-02:21:43:43.730559: kenwood_get_vfo_if called
2020-05-02:21:43:43.730594: kenwood_get_if called
2020-05-02:21:43:43.730630: kenwood_safe_transaction called
2020-05-02:21:43:43.730666: kenwood_transaction called
2020-05-02:21:43:43.730703: elapsed_ms: start = 1588455823,730196370
2020-05-02:21:43:43.730747: elapsed_ms: elapse_secs=0.000535952
2020-05-02:21:43:43.730786: kenwood_transaction: cache hit, age=0ms
2020-05-02:21:43:43.730822: kenwood_get_vfo_if: priv->tx_vfo=VFOA
2020-05-02:21:43:43.730856: elapsed_ms: start = 0,0
2020-05-02:21:43:43.730893: elapsed_ms: after gettime, start = 1588455823,730883430
2020-05-02:21:43:43.730933: elapsed_ms: elapse_secs=1.47539e+06
2020-05-02:21:43:43.730979: rig_set_parm called
Opened rig model 2031, 'TS-590S'
2020-05-02:21:43:43.731059: Backend version: 20200501.0, Status: Stable
2020-05-02:21:43:43.731100: rigctl_parse: called, interactive=1

Rig command:

closed 2020-05-02 by mdblack98