Hamlib/Hamlib #237: IC-910H acts differently now (snapshot build 20200504 vs 20200405)

jharwinbarrozo, 2020-05-05:
To give you insight, I'm using Gpredict + rigctld to control my radio + rotctld to control my rotator. 

Previously, I was using **20200405 snapshot build** which worked incredibly well on IC-910 and GQRX. In Gpredict radio controller, I always select "GQRX" localhost, port 7356 as my receive, and "IC910" localhost, port 4532 as my transmit (this requires only the MAIN band to be active). My radio is set up to only show MAIN BAND, the SUB BAND is hidden/disabled. After pressing "ENGAGE" from Gpredict, then I turn the physical vfo knob on the radio, I could see whatever frequency I have on the radio will be displayed in Gpredict radio control. And the frequency stays to where ever I left it. **I could control the freq in two ways, gpredict to radio or radio to gpredict.**

Sometimes I don't use the "GQRX". and have the "IC910" be my rx and tx, so my radio needs to be in satellite mode for this prior to running rigctld. After pressing ENGAGE, I could see whatever frequency I have on the MAIN and SUB band from the radio will be displayed in Gpredict radio control both for Downlink and Uplink. Turn the knob from radio, GPredict will update. Stop turning the knob, frequency stays where ever I left it. **I could control the freq in two ways, gpredict to radio or radio to gpredict.**

The above **previously** worked great! Until...

Using the latest **20200505 snapshot build**, after engaging - then I turn the physical vfo knob on the radio, I could see the frequency reflected back to Gpredict "momentarily" then go back again to original frequency (the update speed is controlled by XXX msec "cycle" from Gpredict). 
Right now, I couldn't use the knob on my radio to control/update Gpredict uplink/downlink anymore, it's only one-way (Gpredict to radio,YES) but (radio to gpredict "no longer usable")

I tried to report this as clear as possible, but if you have any questions or need anymore clarification, I am willing to provide more feedback.

mdblack98, 2020-05-05:
Could you please provide a debug log adding "-vvvvv -Z" to the rigctld command line?
You can redirect the output like this...
>& log.txt
And send me the log.txt file

Doesn't need to be a long test...just enough to demonstrate the problem.

I thought we had the IC-910 working on just one rigctld instance.  Is there a problem with that where you have to have 2 instances?

de Mike W9MDB

mdblack98, 2020-05-05:
That didn't show up right in the mail
redirect>& log.txt

jharwinbarrozo, 2020-05-05:
> I thought we had the IC-910 working on just one rigctld instance. Is there a problem with that where you have to have 2 instances?
> de Mike W9MDB

I completely uninstall the snapshot build 20200405 prior to installing build 20200505. And the way I run rigctld command is via a script that it makes sure first there's no running instance of rigctld. So I could be certain that there's only one rigctld instance running. 

```
#!/bin/bash
RigPID=$(pgrep rigctld)
[[ -n $RigPID ]] && { echo "Stopping rigctld PID $RigPID"; sudo kill $RigPID; }
```

I will provide you the log soon. Thanks again

jharwinbarrozo, 2020-05-06:
Here's the debug log for the rigctld snapshot build 20200505.

The problem is after engaging gpredict radio controller, the main and sub band on the IC-910 is swapping back and forth by itself. If I turn my radio's vfo knob, the swapping stops, and I could see the frequencies from radio will be reflected back to gpredict. But as soon as I release and stop turning the VFO knob from the radio, the swapping of main and sub band goes back again.

[rigctld-snapshot-build-20200505.txt](https://github.com/Hamlib/Hamlib/files/4586255/rigctld-snapshot-build-20200505.txt)

[image]

[image]

mdblack98, 2020-05-06:
Try adding this switch to rigctld and see if it behaves.
--set-conf=cache_timeout=0
de Mike W9MDB

jharwinbarrozo, 2020-05-06:
Hi @mdblack98 ,

Nothing change. Main and Sub band still keep on swapping back and forth. Similar to the previous one without the --set-conf=cache_timeout=0

Here's the log
[loglog.txt](https://github.com/Hamlib/Hamlib/files/4587927/loglog.txt)

mdblack98, 2020-05-06:
Can you please follow the directions for git bisect to determine where this regression happens?

https://git-scm.com/docs/git-bisect

de Mike W9MDB

jharwinbarrozo, 2020-05-07:
@mdblack98  Okay I did the git bisect. And found out this is that last commit b244fe2 although it requires the --set-conf=cache_time=0 for to rigctld to work on IC-910. Without this --setconf, rigctld is totally useless in IC-910

latest commit 067f95d down to d75272 were bad w/ and w/o --set-conf=cache_time=0

b244fe2 only good with --set-conf=cache_time=0

Hope this helps.

mdblack98, 2020-05-07:
Can you now give me 3 debug logs with b244fe2 cache_timeout=0, one with b244fe2 cache_timeout=500, and one with the latest commit with cache_timeout=500  where it's not working?
I assume you misspelled cache_timeout as cache_time in your email as that should give an error.
The VFO interaction with gpredict and the VFO swapping is a little complex so have to find out where this happening.
Mike

 
 

latest commit 067f95d down to d75272d were bad w/ and w/o --set-conf=cache_time=0

jharwinbarrozo, 2020-05-08:
@mdblack98 apology i just mispelled the "time". I check my command link, it showed correctly

`/home/ubuntu/local-test/bin/rigctld -m 3044 -r /dev/ttyUSB0 -t 4532 -s 19200 --set-conf=cache_timeout=0 -vvvvv -Z 2>&1 | tee cache_timeout0.txt`

Anyway, here are the logs:

b244fe2 w/ cache_timeout=0 
[cache_timeout0.txt](https://github.com/Hamlib/Hamlib/files/4597116/cache_timeout0.txt)

b244fe2 w/ cache_timeout=500
[cache_timeout500.txt](https://github.com/Hamlib/Hamlib/files/4597121/cache_timeout500.txt)
Run gpredict, open radio controller window then engage. As soon as I start to rotate my radio's vfo knob, the swapping of main and sub band starts to happen. Completely, unuseable for IC-910

latest commit 067f95d w/ cache_timeout=500
[latest_commit_w_cache_timeout500.txt](https://github.com/Hamlib/Hamlib/files/4597161/latest_commit_w_cache_timeout500.txt)

Hope this helps.

mdblack98, 2020-05-09:
It would help a lot if we can figure out a rigctl cmd sequence that shows the problem.

I suspect it has to do with getting/setting split or such

So try rigctl and do these commands

V VFOA
F 14074000
V VFOB
F 14075000
V VFOA
i
x
s
I 14076000
X USB 3000
S 1 VFOB
S 0 VFOA

Mike

N6RFM, 2020-05-10:
Hi Mike and Jharwin,

Have been watching this thread intently.  Thanks efforts.  Did a git pull for latest and install.  Just tried your requested sequence Mike.  Ran into some issues, and only made it part way. Log attached.  Hope that is helpful.  Best,  Bob
[IC-910.txt](https://github.com/Hamlib/Hamlib/files/4606333/IC-910.txt)

mdblack98, 2020-05-10:
Your command line is wrong, the -c is unnecessary and the -t is the terminator char for rigctl so is wrong.Try this and change the baud rate to match your setup
rigctl -m 3044 -r /dev/ttyUSB0 -s 19200 -vvvvv -Z
Then send me the log from doing the commands.
Or contact me off list and we can hook up and do some tests on your system.
Mike

N6RFM, 2020-05-10:
Hi Mike,

Thanks.  Here is log.  Not sure I had all commands correct.  For first
part, radio was not in split mode (using button on rig), then tried with
split mode on.  Radio freq was not updating as expected.

Cheers,
Bob

bob@bob:~$
bob@bob:~$ rigctl -m 3044 -r /dev/ttyUSB0 -s 19200 -vvvvv -Z
2020-05-10:18:55:34.464589: rigctl, Hamlib 4.0~git May 10 2020 13:28:19
2020-05-10:18:55:34.464606: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-05-10:18:55:34.464618: rig_init called
2020-05-10:18:55:34.464634: initrigs4_icom: _init called
2020-05-10:18:55:34.464642: rig_register called
2020-05-10:18:55:34.464648: rig_register: rig_register (3055)
2020-05-10:18:55:34.464653: rig_register called
2020-05-10:18:55:34.464659: rig_register: rig_register (3009)
2020-05-10:18:55:34.464665: rig_register called
2020-05-10:18:55:34.464669: rig_register: rig_register (3010)
2020-05-10:18:55:34.464673: rig_register called
2020-05-10:18:55:34.464678: rig_register: rig_register (3011)
2020-05-10:18:55:34.464684: rig_register called
2020-05-10:18:55:34.464688: rig_register: rig_register (3013)
2020-05-10:18:55:34.464693: rig_register called
2020-05-10:18:55:34.464699: rig_register: rig_register (3014)
2020-05-10:18:55:34.464704: rig_register called
2020-05-10:18:55:34.464708: rig_register: rig_register (3015)
2020-05-10:18:55:34.464713: rig_register called
2020-05-10:18:55:34.464718: rig_register: rig_register (3019)
2020-05-10:18:55:34.464723: rig_register called
2020-05-10:18:55:34.464728: rig_register: rig_register (3020)
2020-05-10:18:55:34.464733: rig_register called
2020-05-10:18:55:34.464738: rig_register: rig_register (3021)
2020-05-10:18:55:34.464742: rig_register called
2020-05-10:18:55:34.464747: rig_register: rig_register (3022)
2020-05-10:18:55:34.464752: rig_register called
2020-05-10:18:55:34.464756: rig_register: rig_register (3067)
2020-05-10:18:55:34.464761: rig_register called
2020-05-10:18:55:34.464766: rig_register: rig_register (3023)
2020-05-10:18:55:34.464771: rig_register called
2020-05-10:18:55:34.464776: rig_register: rig_register (3046)
2020-05-10:18:55:34.464781: rig_register called
2020-05-10:18:55:34.464785: rig_register: rig_register (3024)
2020-05-10:18:55:34.464790: rig_register called
2020-05-10:18:55:34.464794: rig_register: rig_register (3028)
2020-05-10:18:55:34.464799: rig_register called
2020-05-10:18:55:34.464804: rig_register: rig_register (3030)
2020-05-10:18:55:34.464809: rig_register called
2020-05-10:18:55:34.464814: rig_register: rig_register (3026)
2020-05-10:18:55:34.464818: rig_register called
2020-05-10:18:55:34.464822: rig_register: rig_register (3027)
2020-05-10:18:55:34.464827: rig_register called
2020-05-10:18:55:34.464831: rig_register: rig_register (3047)
2020-05-10:18:55:34.464836: rig_register called
2020-05-10:18:55:34.464841: rig_register: rig_register (3057)
2020-05-10:18:55:34.464846: rig_register called
2020-05-10:18:55:34.464850: rig_register: rig_register (3063)
2020-05-10:18:55:34.464855: rig_register called
2020-05-10:18:55:34.464860: rig_register: rig_register (3029)
2020-05-10:18:55:34.464865: rig_register called
2020-05-10:18:55:34.464869: rig_register: rig_register (3062)
2020-05-10:18:55:34.464873: rig_register called
2020-05-10:18:55:34.464878: rig_register: rig_register (3045)
2020-05-10:18:55:34.464883: rig_register called
2020-05-10:18:55:34.464887: rig_register: rig_register (3056)
2020-05-10:18:55:34.464892: rig_register called
2020-05-10:18:55:34.464897: rig_register: rig_register (3075)
[... 1143 lines cut ...]
Frequency: 14074000
2020-05-10:18:58:25.513641: rigctl_parse: debug5
2020-05-10:18:58:25.513690: rigctl_parse: debug10
2020-05-10:18:58:25.513748: rigctl_parse: vfo_mode=0
2020-05-10:18:58:25.513784: rig_set_freq called
2020-05-10:18:58:25.513808: icom_get_freq called for currVFO
2020-05-10:18:58:25.513833: set_vfo_curr: vfo=currVFO, curr_vfo=VFOA
2020-05-10:18:58:25.513866: set_vfo_curr: Asking for currVFO,  currVFO=VFOA
2020-05-10:18:58:25.513884: set_vfo_curr: curr_vfo now=VFOA
2020-05-10:18:58:25.513899: icom_get_freq: VFO_RX requested, vfo=currVFO
2020-05-10:18:58:25.513917: icom_get_freq: using vfo=VFOA
2020-05-10:18:58:25.513938: serial_flush called
2020-05-10:18:58:25.513955: serial_flush: tcflush
2020-05-10:18:58:25.513979: write_block called
2020-05-10:18:58:25.514016: write_block(): TX 6 bytes
2020-05-10:18:58:25.514040: 0000    fe fe 60 e0 03 fd                                   ..`...          
2020-05-10:18:58:25.514060: read_string called
2020-05-10:18:58:25.522148: read_string(): RX 6 characters
2020-05-10:18:58:25.522185: 0000    fe fe 60 e0 03 fd                                   ..`...          
2020-05-10:18:58:25.522201: read_string called
2020-05-10:18:58:25.554049: read_string(): RX 11 characters
2020-05-10:18:58:25.554100: 0000    fe fe e0 60 03 63 95 25 44 01 fd                    ...`.c.%D..     
2020-05-10:18:58:25.554131: from_bcd called
2020-05-10:18:58:25.554162: icom_set_freq called currVFO=14074000.000000
2020-05-10:18:58:25.554186: icom_set_freq: currVFO

mdblack98, 2020-06-02:
Is this working now?

N6RFM, 2020-06-02:
Hi Mike,

Did git pull, install etc.  When using GPredict, enabling split mode via
rigctld interface works.  What seems to be missing is when asking for
satmode from GPredict, the rig goes to or stays in split mode.   If
manually place rig is satmode, things works. And then if ask for split
mode, rig goes to split mode.  But, unable to go back to satmode via
software interface.

Cheers,

Bob