Hamlib/Hamlib #239: FT-991 FM mode, rptr_off, w cmd

mdblack98, 2020-05-05:
Hi

I am using Michael Black's fork from May 3rd  on a RigPi (Raspbian
Buster Pi4)

I am playing with my FT-991A and I am surprised that I cannot set FM
mode.  I can set PKTFM.
caps show:
Mode list: AM CW USB LSB RTTY CWR RTTYR PKTLSB PKTUSB PKTFM C4FM
and
FM    Normal: 16 kHz,    Narrow: 9 kHz,    Wide: 0 Hz
It looks like it is supported in the Ft-991A Cat manual with MD04; (page 11)

Also repeater offset does not seem to be supported

Rig command: O 600
set_rptr_offs: error = Feature not available
caps show
Can set Repeater Offset:    N
Can get Repeater Offset:    N

It looks like it is supported in the Ft-991A Cat manual with
EX 082 60 (page 7-8) but unfortunately it looks like it is band specific.

I also cannot seem to be able to use the w command, most likely I am
doing something wrong. A pointer would be appreciated.
Rig command: w fa;
send_cmd: error = Invalid parameter

Rig command:

I was expecting it to return the current frequency.

Any -vvvvv log required just ask.

I have a TS-590SG, FT-991A and a FT-857D and more than willing to do any
testing required to improve Hamlib. I really apreciate all the years of
work on this great project!

73
Glenn VE9GJ

GoatMon, 2020-05-06:
Regarding the w send_cmd, its looking for hex.
rig w \\0x49\\0x46\\0x3b
IF033053090000+000000411002;
or your example
rig w \\0x66\\0x61\\0x3b
FA053090000;

mdblack98, 2020-05-06:
Yeah...it's not recognizing the terminator ';' for Yaesu rigs.
Gotta' fix that.
Mike

ve9gj, 2020-05-06:
I figure out w would work with hex last evening Thanks
I had expected it to take the acscii chars.
I rebuilt hamlib but FM is still not working as I expect.

git show 
commit e9ffb2991d2c3b8f1cef47ab6c6516bb713b28bf (HEAD -> master, origin/master, origin/HEAD)
Author: mdblack98 <mdblack98@yahoo.com>
Date:   Wed May 6 08:54:34 2020 -0500
Radio is tuned to 147.150Mhz FM 16khz BW

Rig command: m
Mode: FM
Passband: 1700 <--??

Rig command: M USB 2400
Mode: 
Rig command:   (Worked OK)

Rig command: M FM 16000
Mode: set_mode: error = Invalid parameter

Rig command: M FM 9000
Mode: set_mode: error = Invalid parameter

rigctl -vvvvvvvv -m 1035 -r /dev/radio2  >hamlib.txt 2>&1
I do a m followed by a M FM 16000
The radio does switch to FM but with the set_mode: error = Invalid parameter
[hamlib.txt](https://github.com/Hamlib/Hamlib/files/4587838/hamlib.txt)

Anything else you need?

73
Glenn 

:

ve9gj, 2020-05-06:
I added to newcat.c at line 5196
`case RIG_MODE_FM: //ve9gj
  //case RIG_MODE_FM: // Can you set passband on FM or FMN for FT991? Returns error for now.`

This stopped the invalid parameter errors

mdblack98, 2020-05-07:
The FT-991A has one mode FM with a Nar/Wide button on the F-Menu Touch
Screen. (9khz/16khz) Yaesu's SH0; will return SH001; for 9khz and SH002;
  for 16khz but doesn't seem to allow it to be set for FM.

Glenn VE9GJ

ve9gj, 2020-05-07:
working on a patch to fix the FT991 passband issues

mdblack98, 2020-05-12:
All working now....may be some more enhancements coming but no bugs right now.

closed 2020-05-12 by mdblack98