Hamlib/Hamlib #257: Many warnings with gcc 8.3.0 on Debian 10

N0NB, 2020-05-21:
I just did a clean build on Debian 10 that resulted in quite a number of warnings.  I will put the warnings for each file in a separate comment in this issue.

```
  CC       netrigctl.lo
../../hamlib/dummy/netrigctl.c: In function ‘netrigctl_set_freq’:
../../hamlib/dummy/netrigctl.c:577:10: warning: unused variable ‘vfostr’ [-Wunused-variable]
     char vfostr[6] = "";
          ^~~~~~
../../hamlib/dummy/netrigctl.c: In function ‘netrigctl_get_freq’:
../../hamlib/dummy/netrigctl.c:609:10: warning: unused variable ‘vfotmp’ [-Wunused-variable]
     char vfotmp[16];
          ^~~~~~
```

N0NB, 2020-05-21:
```
  CC       flrig.lo
In file included from ../../hamlib/dummy/flrig.c:32:
../../hamlib/dummy/flrig.c: In function ‘read_transaction.constprop’:
../../hamlib/dummy/flrig.c:412:36: warning: ‘%s’ directive output may be truncated writing up to 8191 bytes into a region of size 730 [-Wformat-truncation=]
         rig_debug(RIG_DEBUG_TRACE, "%s: string='%s'\n", __func__, tmp_buf);
                                    ^~~~~~~~~~~~~~~~~~~            ~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 29 and 8220 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/dummy/flrig.c:412:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_TRACE, "%s: string='%s'\n", __func__, tmp_buf);
         ^~~~~~~~~
../../hamlib/dummy/flrig.c:417:38: warning: ‘%s’ directive output may be truncated writing up to 8191 bytes into a region of size 705 [-Wformat-truncation=]
             rig_debug(RIG_DEBUG_ERR, "%s: Expected 'HTTP/1.1 200 OK', got '%s'\n", __func__,
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                       tmp_buf);
                       ~~~~~~~         
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 54 and 8245 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/dummy/flrig.c:417:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_ERR, "%s: Expected 'HTTP/1.1 200 OK', got '%s'\n", __func__,
             ^~~~~~~~~
../../hamlib/dummy/flrig.c:438:23: warning: ‘%s’ directive output may be truncated writing up to 8191 bytes into a region of size between 693 and 696 [-Wformat-truncation=]
                       "%s: xml buffer overflow!!\nTrying to add len=%d\n%sTo len=%d\n%s", __func__,
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                       (int)strlen(tmp_buf), tmp_buf, (int)strlen(xml), xml);
                                             ~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/dummy/flrig.c:438:23: note: using the range [-2147483648, 2147483647] for directive argument
                       "%s: xml buffer overflow!!\nTrying to add len=%d\n%sTo len=%d\n%s", __func__,
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output 70 or more bytes (assuming 8264) into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/dummy/flrig.c:437:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_ERR,
             ^~~~~~~~~
../../hamlib/dummy/flrig.c: In function ‘flrig_open’:
../../hamlib/dummy/flrig.c:681:34: warning: ‘%s’ directive output may be truncated writing up to 8191 bytes into a region of size 744 [-Wformat-truncation=]
     rig_debug(RIG_DEBUG_VERBOSE, "Transceiver=%s\n", value);
                                  ^~~~~~~~~~~~~~~~~~  ~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 14 and 8205 bytes into a destination of size 756
[... cut ...]
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 19 and 8210 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/dummy/flrig.c:1158:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_TRACE, "%s: '%s'\n", __func__, value);
             ^~~~~~~~~
```

N0NB, 2020-05-21:
```
  CC       network.lo
In file included from ../../hamlib/src/network.c:70:
../../hamlib/src/network.c: In function ‘network_flush’:
../../hamlib/src/network.c:315:23: warning: ‘%s’ directive output may be truncated writing up to 8191 bytes into a region of size between 694 and 704 [-Wformat-truncation=]
                       "%s: network data clear d: ret=%d, len=%d, '%s'\n",
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/network.c:317:38:
                       ret, (int)len, buffer);
                                      ~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 55 and 8256 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/network.c:314:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_WARN,
             ^~~~~~~~~
../../hamlib/src/network.c:328:23: warning: ‘%s’ directive output may be truncated writing up to 8191 bytes into a region of size between 679 and 695 [-Wformat-truncation=]
                       "%s: network data cleared: ret=%d, len_read=%d/0x%x, '%s'\n",
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/network.c:330:48:
                       ret, len_read, len_read, buffer);
                                                ~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 64 and 8271 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/network.c:327:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_WARN,
             ^~~~~~~~~
```

N0NB, 2020-05-21:
```
  CC       gpio.lo
In file included from ../../hamlib/src/gpio.h:25,
                 from ../../hamlib/src/gpio.c:29:
../../hamlib/src/gpio.c: In function ‘gpio_open’:
../../hamlib/src/gpio.c:88:19: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size between 221 and 732 [-Wformat-truncation=]
                   "GPIO%s opening value file %s: %s\n",
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/gpio.c:90:19:
                   pathname,
                   ~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output 28 or more bytes (assuming 1562) into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/gpio.c:87:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_ERR,
         ^~~~~~~~~
../../hamlib/src/gpio.c:66:19: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size between 223 and 734 [-Wformat-truncation=]
                   "GPIO%s direction (using %s): %s\n",
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/gpio.c:68:19:
                   pathname,
                   ~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output 27 or more bytes (assuming 1561) into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/gpio.c:65:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_ERR,
         ^~~~~~~~~
../../hamlib/src/gpio.c:47:19: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size between 226 and 737 [-Wformat-truncation=]
                   "Export GPIO%s (using %s): %s\n",
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/gpio.c:49:19:
                   pathname,
                   ~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output 24 or more bytes (assuming 1558) into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/src/gpio.c:46:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_ERR,
         ^~~~~~~~~
```

N0NB, 2020-05-21:
```
  CC       rigctld-rigctld.o
In file included from ../../hamlib/tests/rigctld.c:73:
../../hamlib/tests/rigctld.c: In function ‘handle_socket’:
../../hamlib/tests/rigctld.c:1101:15: warning: ‘%s’ directive output may be truncated writing up to 1024 bytes into a region of size 733 [-Wformat-truncation=]
               "Connection closed from %s:%s\n",
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               host,
               ~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 26 and 1081 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctld.c:1100:5: note: in expansion of macro ‘rig_debug’
     rig_debug(RIG_DEBUG_VERBOSE,
     ^~~~~~~~~
../../hamlib/tests/rigctld.c: In function ‘main’:
../../hamlib/tests/rigctld.c:896:23: warning: ‘%s’ directive output may be truncated writing up to 1024 bytes into a region of size 733 [-Wformat-truncation=]
                       "Connection opened from %s:%s\n",
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                       host,
                       ~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 26 and 1081 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctld.c:895:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_VERBOSE,
             ^~~~~~~~~
```

N0NB, 2020-05-21:
```
  CC       rotctld-rotctld.o
In file included from ../../hamlib/include/hamlib/rotator.h:25,
                 from ../../hamlib/tests/rotctld.c:64:
../../hamlib/tests/rotctld.c: In function ‘handle_socket’:
../../hamlib/tests/rotctld.c:694:15: warning: ‘%s’ directive output may be truncated writing up to 1024 bytes into a region of size 733 [-Wformat-truncation=]
               "Connection closed from %s:%s\n",
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               host,
               ~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 26 and 1081 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rotctld.c:693:5: note: in expansion of macro ‘rig_debug’
     rig_debug(RIG_DEBUG_VERBOSE,
     ^~~~~~~~~
../../hamlib/tests/rotctld.c: In function ‘main’:
../../hamlib/tests/rotctld.c:588:19: warning: ‘%s’ directive output may be truncated writing up to 1024 bytes into a region of size 733 [-Wformat-truncation=]
                   "Connection opened from %s:%s\n",
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   host,
                   ~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 26 and 1081 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rotctld.c:587:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_VERBOSE,
         ^~~~~~~~~
```

N0NB, 2020-05-21:
```
  CC       rigctlcom-rigctlcom.o
In file included from ../../hamlib/tests/rigctlcom.c:82:
../../hamlib/tests/rigctlcom.c: In function ‘main’:
../../hamlib/tests/rigctlcom.c:1554:19: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 694 [-Wformat-truncation=]
                   "*********************************\n%s: unknown cmd='%s'\n",
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 65 and 1088 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctlcom.c:1553:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_ERR,
         ^~~~~~~~~
../../hamlib/tests/rigctlcom.c:1428:38: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 726 [-Wformat-truncation=]
             rig_debug(RIG_DEBUG_ERR, "%s: error parsing '%s'\n", __func__, (char *)arg);
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 33 and 1056 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctlcom.c:1428:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_ERR, "%s: error parsing '%s'\n", __func__, (char *)arg);
             ^~~~~~~~~
../../hamlib/tests/rigctlcom.c:1374:38: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 715 [-Wformat-truncation=]
             rig_debug(RIG_DEBUG_ERR, "%s: squelch cmd parse failed: %s\n", __func__,
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 43 and 1066 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctlcom.c:1374:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_ERR, "%s: squelch cmd parse failed: %s\n", __func__,
             ^~~~~~~~~
../../hamlib/tests/rigctlcom.c:1323:38: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 719 [-Wformat-truncation=]
             rig_debug(RIG_DEBUG_ERR, "%s: AGC cmd parse failed: %s\n", __func__,
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 39 and 1062 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctlcom.c:1323:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_ERR, "%s: AGC cmd parse failed: %s\n", __func__,
             ^~~~~~~~~
../../hamlib/tests/rigctlcom.c:1265:38: warning: ‘%s’ directive output may be truncated writing up to 1023 bytes into a region of size 710 [-Wformat-truncation=]
             rig_debug(RIG_DEBUG_ERR, "%s: speech level cmd parse failed: %s\n", __func__,
                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
[... cut ...]
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 44 and 1067 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/rigctlcom.c:1021:13: note: in expansion of macro ‘rig_debug’
             rig_debug(RIG_DEBUG_ERR, "%s: error parsing preamp cmd '%s'\n", __func__,
             ^~~~~~~~~
```

N0NB, 2020-05-21:
```
  CC       ampctld-ampctld.o
In file included from ../../hamlib/include/hamlib/amplifier.h:25,
                 from ../../hamlib/tests/ampctld.c:64:
../../hamlib/tests/ampctld.c: In function ‘handle_socket’:
../../hamlib/tests/ampctld.c:680:15: warning: ‘%s’ directive output may be truncated writing up to 1024 bytes into a region of size 733 [-Wformat-truncation=]
               "Connection closed from %s:%s\n",
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               host,
               ~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 26 and 1081 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/ampctld.c:679:5: note: in expansion of macro ‘rig_debug’
     rig_debug(RIG_DEBUG_VERBOSE,
     ^~~~~~~~~
../../hamlib/tests/ampctld.c: In function ‘main’:
../../hamlib/tests/ampctld.c:575:19: warning: ‘%s’ directive output may be truncated writing up to 1024 bytes into a region of size 733 [-Wformat-truncation=]
                   "Connection opened from %s:%s\n",
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   host,
                   ~~~~
../../hamlib/include/hamlib/rig.h:2646:93: note: in definition of macro ‘rig_debug’
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                                                             ^~~
../../hamlib/include/hamlib/rig.h:2646:62: note: ‘snprintf’ output between 26 and 1081 bytes into a destination of size 756
 #define rig_debug(debug_level,fmt,...) { char xxxbuf[756]="";snprintf(xxxbuf,sizeof(xxxbuf),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); }
                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../hamlib/tests/ampctld.c:574:9: note: in expansion of macro ‘rig_debug’
         rig_debug(RIG_DEBUG_VERBOSE,
         ^~~~~~~~~
```

mdblack98, 2020-05-21:
Do an update and try again...Had to restore the bigger buffer size in rig_debug
Mike W9MDB

N0NB, 2020-05-21:
Excellent!

That quelled all but the unused variable warning from netrigctl.c

Thanks, Mike.

closed 2020-05-22 by mdblack98