Hamlib/Hamlib #260: FT-DX101 Improvements

mdblack98, 2020-05-24:
The FT-DX101MP works OK but there are several places where there are some potential improvements in the newcat.c driver where the DX5000 is referenced.  Perhaps the DX101 models should also be referenced.  Need to examine the manuals.

mdblack98, 2020-05-26:
Changed this to a bug now as have a JTDX user reporting bad behavior on the FTDX101D
Quote:

I've tested a brand new Yaesu FTdx101D radio and found quite some severe issues with the CAT control with JTDX v151_1:

To name a few:

With the FTdx101 radio selected:
1 - When MONO is selected (only main receiver in display) the two meters are displayed together with the Filter display (the Width graph). When I use the USB cable from the radio direct to the laptop, during TX the second S meter is removed! Very inconvenient because that removed meter displays in example the ALC or SWR.
 
2 - When MONO is DE-selected (both receivers in display and the two S meters are displayed together with the Filter display (the Width graph)). When I use the Yaesu SCU-17 interface over USB cable from the radio to the laptop, during TX the next band stack is selected!!! (very severe bug/ issue, as also items like the internal tuner can be disabled on other band stack while radio goes into TX mode etc). Also can be seen that during the start of TX, the second TX/RX LEDs blink and the TFT display flickers White for a split second accordingly. (all indicated a swap between VFO A > B and back to A within a second time at start of TX).

3 - When MONO is DE-selected (both receivers in display and the two S meters are displayed together with the Filter display (the Width graph)) the blinking of the VFO-B does also occur and also TFT display does flash white when I use the USB direct connection. VFO-B flashes for a split second. (all indicated a swap between VFO A > B and back to A within a second time at start of TX).

With the FTdx5000 selected as radio (using on the same FTdx101D radio):
 4 - When MONO is DE-selected (both receivers in display and the two S meters are displayed together with the Filter display (the Width graph)) the blinking of the VFO-B does not occur and also TFT display does not flash white when I use the USB direct connection.

vk4tux, 2020-06-02:
Tested above scenarios and also the SUB Data-U cat set , and find there are no issues with today’s latest release on my FTDX101MP.

closed 2020-06-04 by mdblack98