Hamlib/Hamlib #262: rotcrl rotcgtld 603 rotator error

mdblack98, 2020-05-25:
Subject: | rotctl rotctld and 603 rotator error
-- | --
Sun, 24 May 2020 21:05:38 -0400
John LeRoy <w4jklleroy@aim.com>
hamlib-developer@lists.sourceforge.net

I pulled the nightly snapshot yesterday and I'm using it with cqrlog right now. cqrlog starts /usr/local/bin/rotctld at log startup. I have a script (rotor) that I use to manually rotate the beam. The rotctl for the 603 Yaesu GS-232B when called by

#!/bin/bash
#
sleep 1
#
/usr/local/bin/rotctl -m 2 -r 127.0.0.1:4533 -vvvv

gives

john@hamradio1:~$ ./rotor
rotctl, Hamlib 4.0~git
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_dummy: _init called
rot_register (1)
rot_register (2)
set_conf: called
rot_open called
rot_open: using network address 127.0.0.1:4533
network_open called
network_open version 1.0
netrotctl_open called
write_block called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.   
read_string called
read_string(): RX 2 characters
0000    30 0a                                               0.             
read_string called
read_string(): RX 4 characters
0000    36 30 33 0a                                         603.           
read_string called
read_string(): RX 12 characters
0000    2d 31 38 30 2e 30 30 30 30 30 30 0a                 -180.000000.   
read_string called
read_string(): RX 11 characters
0000    34 35 30 2e 30 30 30 30 30 30 0a                    450.000000.    
read_string called
read_string(): RX 9 characters
0000    30 2e 30 30 30 30 30 30 0a                          0.000000.      
read_string called
read_string(): RX 11 characters
0000    31 38 30 2e 30 30 30 30 30 30 0a                    180.000000.    
Opened rot model 2, 'NET rotctl'
Backend version: 0.3, Status: Beta

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
write_block called
write_block(): TX 2 bytes
0000    70 0a                                               p.             
read_string called
read_string(): RX 2 characters
0000    30 0a                                               0.             
read_string called
read_string(): RX 7 characters
0000    33 30 30 2e 30 30 0a                                300.00.        
rot_get_position: got az=0.00, el=300.00
Azimuth: 0.00
Elevation: 300.00

Rotator command:

which is reversed az and el. Azimuth is 301 degrees on the hy-gain controller. Elevation is random for controller here, which is an ERC card in the hy-gain controller box.
However, cqrlog reports the correct azimuth, so rotctld reports correctly.
73,

John W4JKL

ve9gj, 2020-05-25:
This looks like the same issue I emailed the dev list about on 2020-05-22
When running rotctld there appears to be a few problems.
[rotctld.txt](https://github.com/Hamlib/Hamlib/files/4677487/rotctld.txt)
[rotctl_net.txt](https://github.com/Hamlib/Hamlib/files/4677488/rotctl_net.txt)
[rotctldirect.txt](https://github.com/Hamlib/Hamlib/files/4677489/rotctldirect.txt)

using rotctl directly (rotctldirect.txt) connected to model 401 (Idiom RotarEZ) all seems OK.

Now using rotctl -m2 there are issues. See rotctl_net.txt

Line 13  ???
connect to localhost:4533 failed, (trying next interface): Network error 111: Connection refused

line 51  after doing a p
rot_get_position: got az=0.00, el=196.00
Azimuth: 0.00
Elevation: 196.00
My rotor does not do elev at all?

Line 66 while  doing a P 230 0
set_pos: error = Protocol error
The rotor does turn to 230 NP.

After using rotctl to turn (P) once from now on az and el will be reported correctly.

I have also attached the debug from the rotctld session.

Another user is also having the exact same issues with his Yaesu rotor. We are both using a current hamlib build (as of yesterday)

Any test or further debug log just ask!

w4jkl, 2020-05-28:
> John LeRoy w4jklleroy@aim.com <mailto:w4jklleroy@aim.com>

Update2:

You have to use the P command before the az/el inversion goes away.

73,

John W4JKL

mdblack98, 2020-05-28:
Can you show that in some debug?  What rotor model do you have?
The GS232B manual doesn't show any inversion that I can see

Mike

mdblack98, 2020-05-28:
Wait a minute...I think the problem is with rotctld....
Your debug is with rot_dummy -- do you have a problem with using it on a real rotor?
I'm looking at it...
Mike

ve9gj, 2020-05-28:
Yes the problem only exists when using rotctld. Direct rotctl works OK.
I am using idiom press rotorEZ
Glenn

mdblack98, 2020-05-28:
Just posted a patch that should fix it.  Please testProblem has been there forever apparently....
Mike

ve9gj, 2020-05-28:
Hi Mike

Your patch appears to fix the issue.  AZ / El is correct everytime and 
the read errors are gone!
Thanks and 73
Glenn VE9GJ

closed 2020-05-28 by mdblack98

w4jkl, 2020-05-29:
I compiled today the

/home/john/Downloads/hamlib-4.0~git-889d80ab-20200529.tar.gz

Started CQRLOG which started /usr/local/bin/rotctld for 603 Yaesu 
GS-232B on /dev/ttyUSB1 at 19200 polling every 1 second

Then ran

#!/bin/bash
#
sleep 1
#
/usr/local/bin/rotctl -m 2 -r 127.0.0.1:4533 -vvvvv

(-m 2 is the 2 Hamlib NET rotctl )

Here's the dump.

john@hamradio1:~$ ./rotor
rotctl Hamlib 4.0~git
Last commit was Thu May 28 23:55:44 2020 -0500
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_dummy: _init called
rot_register (1)
rot_register (2)
set_conf: called
rot_open called
rot_open: using network address 127.0.0.1:4533
network_open called
network_open version 1.0
netrotctl_open called
network_flush called
write_block called
write_block(): TX 12 bytes
0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a \dump_state.
read_string called
read_string(): RX 2 characters
0000    30 0a 0.
read_string called
read_string(): RX 4 characters
0000    36 30 33 0a 603.
read_string called
read_string(): RX 12 characters
0000    2d 31 38 30 2e 30 30 30 30 30 30 0a -180.000000.
read_string called
read_string(): RX 11 characters
0000    34 35 30 2e 30 30 30 30 30 30 0a 450.000000.
read_string called
read_string(): RX 9 characters
0000    30 2e 30 30 30 30 30 30 0a 0.000000.
read_string called
read_string(): RX 11 characters
0000    31 38 30 2e 30 30 30 30 30 30 0a 180.000000.
Opened rot model 2, 'NET rotctl'
Backend version: 0.3, Status: Beta

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
network_flush called
network_flush: network data clear d: ret=0, len=2, ''
network_flush: network data cleared: ret=0, len_read=2/0x2, '0
'
write_block called
write_block(): TX 2 bytes
0000    70 0a p.
read_string called
read_string(): RX 5 characters
0000    32 2e 30 30 0a 2.00.
read_string called
read_string(): RX 6 characters
0000    36 36 2e 30 30 0a 66.00.
rot_get_position: got az=2.00, el=66.00
Azimuth: 2.00
Elevation: 66.00

Rotator command: P 45 0
rotctl_parse: input_line: P 45 0
rot_set_position called az=45.00 el=0.00
rot_set_position: south_zero=0
netrotctl_set_position called: 45.000000 0.000000
network_flush called
write_block called
write_block(): TX 21 bytes
0000    50 20 34 35 2e 30 30 30 30 30 30 20 30 2e 30 30     P 45.000000 0.00
0010    30 30 30 30 0a 0000.
read_string called
read_string(): RX 7 characters
0000    52 50 52 54 20 30 0a                                RPRT 0.

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
network_flush called
write_block called
write_block(): TX 2 bytes
0000    70 0a p.
read_string called
read_string(): RX 6 characters
0000    34 38 2e 30 30 0a 48.00.
read_string called
read_string(): RX 6 characters
0000    37 34 2e 30 30 0a 74.00.
rot_get_position: got az=48.00, el=74.00
Azimuth: 48.00
Elevation: 74.00

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
network_flush called
write_block called
write_block(): TX 2 bytes
0000    70 0a p.
read_string called
read_string(): RX 6 characters
0000    34 36 2e 30 30 0a 46.00.
read_string called
read_string(): RX 6 characters
0000    36 38 2e 30 30 0a 68.00.
rot_get_position: got az=46.00, el=68.00
Azimuth: 46.00
Elevation: 68.00

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
network_flush called
write_block called
write_block(): TX 2 bytes
0000    70 0a p.
read_string called
read_string(): RX 6 characters
0000    34 35 2e 30 30 0a 45.00.
read_string called
read_string(): RX 6 characters
0000    36 39 2e 30 30 0a 69.00.
rot_get_position: got az=45.00, el=69.00
Azimuth: 45.00
Elevation: 69.00

Rotator command: P 0 0
rotctl_parse: input_line: P 0 0
rot_set_position called az=0.00 el=0.00
rot_set_position: south_zero=0
netrotctl_set_position called: 0.000000 0.000000
network_flush called
write_block called
write_block(): TX 20 bytes
0000    50 20 30 2e 30 30 30 30 30 30 20 30 2e 30 30 30     P 0.000000 0.000
0010    30 30 30 0a 000.
read_string called
read_string(): RX 7 characters
0000    52 50 52 54 20 30 0a                                RPRT 0.

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
network_flush called
write_block called
write_block(): TX 2 bytes
0000    70 0a p.
read_string called
read_string(): RX 7 characters
0000    33 35 38 2e 30 30 0a 358.00.
read_string called
read_string(): RX 6 characters
0000    36 30 2e 30 30 0a 60.00.
rot_get_position: got az=358.00, el=60.00
Azimuth: 358.00
Elevation: 60.00

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
netrotctl_get_position called
network_flush called
write_block called
write_block(): TX 2 bytes
0000    70 0a p.
read_string called
read_string(): RX 7 characters
0000    33 35 38 2e 30 30 0a 358.00.
read_string called
read_string(): RX 6 characters
0000    36 36 2e 30 30 0a 66.00.
rot_get_position: got az=358.00, el=66.00
Azimuth: 358.00
Elevation: 66.00

Rotator command:

Works. The protocol errors are gone. Thanks Mike!

73,

John W4JKL