Hamlib/Hamlib #263: Add rig test program

mdblack98, 2020-05-26:
Add a rig test program..."rigtest" perhaps.
It will do several tests of critical functions and allow debug info to be shown.  This should allow better isolation of certain problems.

#1 Determine available VFOs and the ability to get/set freq and get/set mode on them
#2 Determine that RX/TX VFOs operate as intended.
#3 Determine split modes work as appropriate
#4 Determine sat mode works as appropriate
#5 Determine PTT modes work as appropriate

mikaelnousiainen, 2020-09-14:
@mdblack98 This would be very welcome, have you started working on this tool already?  In addition to the critical tests, this tool could simply go through all defined levels, funcs, parms and other features to see that they work on a particular rig model.

Regressions would be way easier to spot that way.

mdblack98, 2020-09-14:
No...have not started working on this....very low priority.  Feel free to attack it if you want.  I'm busy enough with other things.
As far as the IC-7300 PARMS/LEVELS and such that logic has been broken for a LONG time for all Icom rigs and was just noticed by a tester.  It's a not-often-used section of code.
Mike

dforsi, 2025-08-12:
As `tests/testrig.c` was created before this issue was open, however it does simpler tests than those in the [first message above](https://github.com/Hamlib/Hamlib/issues/263#issue-625069802) and it doesn't return a failure code if any of them fails, so it can't be used as-is in CI pipeline.

It calls the following functions with predefined arguments, it prints the result or the error message:
```
grep -Eo "rig_[gs]et[^( ]+\(" testrig.c | sort -u
rig_get_caps_int(
rig_get_conf2(
rig_get_freq(
rig_get_func(
rig_get_mode(
rig_get_strength(
rig_get_vfo(
rig_set_conf(
rig_set_debug_level(
rig_set_freq(
rig_set_func(
rig_set_mode(
rig_set_ptt(
rig_set_vfo(
```