Hamlib/Hamlib #266: Rotctl get_pos and set_pos errors with Meade driver

rkinnett, 2020-05-29:
Hi,

I am running rotctl and rotctld (release 3.3) on Win 10 64-bit with a Meade Starnavigator az/el mount with a 497EP Audiostar controller.  Per multiple sources, this hand controller is fully backward compatible with classic Meade Autostar protocol, and this is more or less confirmed by successful operation in other applications.

The connection looks fine, and I can confirm the mount receives command, per Issue 2 below.
```
c:\Program Files (x86)\hamlib-w32-3.3\bin>rotctl -m 1801 -r COM3 -vvvv
rotctl, Hamlib 3.3
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
meade: _init called
rot_register (1801)
meade_init called
rot_open called
serial_open called
serial_setup called
meade_open called
serial_flush called
write_block called
Opened rot model 1801, 'LX200'
rig_strstatus called
Backend version: 0.1, Status: Alpha
```

**Issue 1**:  get_pos returns "Invalid parameter"
```
Rotator command: p
rot_get_position called
meade_get_position called
serial_flush called
write_block called
read_string called
get_pos: error = Invalid parameter
```

This appears to be the same issue as previously reported here: 
https://sourceforge.net/p/hamlib/mailman/message/36799731/  

**Issue 2**:  set_pos fails if _elevation_ argument greater than 1.50 deg (regardless of az which works fine for elevation values <=1.5).  Here's a successful slew to 1.5 deg and failure at 1.51:
```
Rotator command: P 10 1.50
Azimuth: Elevation: rot_set_position called
meade_set_position called: 10.00 1.50
serial_flush called
write_block called
read_string called
serial_flush called
write_block called
read_string called

Rotator command: P 10 1.51
Azimuth: Elevation: rot_set_position called
meade_set_position called: 10.00 1.51
serial_flush called
write_block called
read_string called
serial_flush called
write_block called
read_string called
set_pos: error = Invalid parameter
```

This seems suspiciously close to 1.57 rad, which is 90 deg elevation, which one might expect to get rejected.  I wonder if there might be a missing deg->rad conversion?

Logs available if needed, but the above snippets capture the parts of interest.

Thanks

mdblack98, 2020-05-29:
Please try 4.0 and see if it works now. 3.3 is no longer being maintained
http://n0nb.users.sourceforge.net/

There have been a few commits relevant to Meade since then.
commit 49fde34f2f14699897b26c77b31f857e81401519Author: Michael Black <mdblack98@yahoo.com>Date:   Sat Nov 23 23:04:54 2019 -0600
    Tested with LX200 & Autostar    Move command during active rotation ignored    Added --conf=south_zero=1 option for LX200 south orientation    LX200 now moves smoothly through 180 degrees    Parses both low and high precision answers
commit b8da8cd4a80ae2ef9bf82f468652991be64b0573Author: Michael Black <mdblack98@yahoo.com>Date:   Thu Nov 7 07:52:56 2019 -0600
    astyle meade and prosistel
commit 3bccc571e7ddccb5d42bcf0669208d41f2643f8bAuthor: Michael Black <mdblack98@yahoo.com>Date:   Wed Nov 6 23:19:46 2019 -0600
    Fix Meade parsing and promote to stable

 
 

https://sourceforge.net/p/hamlib/mailman/message/36799731/

rkinnett, 2020-05-29:
Thanks for the quick reply and the link to 4.0.

Issue #1 above seems to have been fixed in w64 4.0.  set_pos is now working.

Issue #2 persists - I can slew to any azimuth but elevation is limited to 1.5 deg.

mdblack98, 2020-05-29:
Please use debug switches and send me some debug info.
Try this command and send me the log.txt file
rotctl -m 1801 -r com3 -vvvvv -Z P 180 30 pause 1 p pause 1 p pause 1 p >log.txt 2>&1

Mike

rkinnett, 2020-05-29:
Here ya' go:
(note the all-zero positions are legitimate)

```
2020-05-29:04:28:11.457233: rotctl Hamlib 4.0~git
Last commit was Wed May 27 10:56:40 2020 -0500
2020-05-29:04:28:11.458207: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-05-29:04:28:11.458207: rot_init called
2020-05-29:04:28:11.458207: initrots4_meade: _init called
2020-05-29:04:28:11.458207: rot_register (1801)
2020-05-29:04:28:11.458207: meade_init called version 0.2
2020-05-29:04:28:11.458207: rot_open called
2020-05-29:04:28:11.458207: serial_open called
2020-05-29:04:28:11.459215: serial_setup called
2020-05-29:04:28:11.459215: serial_flush called
2020-05-29:04:28:11.459215: serial_flush: tcflush
2020-05-29:04:28:11.459215: meade_open called
2020-05-29:04:28:11.459215: serial_flush called
2020-05-29:04:28:11.459215: serial_flush: tcflush
2020-05-29:04:28:11.460305: write_block called
2020-05-29:04:28:12.093500: meade_open product_name=Autostar
2020-05-29:04:28:12.093500: serial_flush called
2020-05-29:04:28:12.093500: serial_flush: tcflush
2020-05-29:04:28:12.093500: write_block called
2020-05-29:04:28:12.294009: Backend version: 0.2, Status: Stable
2020-05-29:04:28:12.294009: rot_set_position called az=180.00 el=30.00
2020-05-29:04:28:12.294009: rot_set_position: south_zero=0 
2020-05-29:04:28:12.294009: meade_set_position called: 180.00 30.00
2020-05-29:04:28:12.294009: serial_flush called
2020-05-29:04:28:12.294009: serial_flush: tcflush
2020-05-29:04:28:12.294009: write_block called
2020-05-29:04:28:12.924281: serial_flush called
2020-05-29:04:28:12.924281: serial_flush: tcflush
2020-05-29:04:28:12.924281: write_block called
2020-05-29:04:28:13.124860: meade_set_position: expected 110, got 112
Opened rot model 1801, 'LX200/Autostar'
set_pos: error = Invalid parameter
2020-05-29:04:28:14.126097: rot_get_position called
2020-05-29:04:28:14.126097: meade_get_position called
2020-05-29:04:28:14.126097: serial_flush called
2020-05-29:04:28:14.126097: serial_flush: tcflush
2020-05-29:04:28:14.126097: write_block called
2020-05-29:04:28:14.756379: meade_get_position: returned '000�00:00#+00�00:00#'
2020-05-29:04:28:14.756379: meade_get_position: az=000:00:00, el=000:00:00
2020-05-29:04:28:14.756379: dmmm2dec called
2020-05-29:04:28:14.756379: dmmm2dec called
2020-05-29:04:28:14.756379: rot_get_position: got az=0.00, el=0.00
0.00
0.00
2020-05-29:04:28:15.756705: rot_get_position called
2020-05-29:04:28:15.756705: meade_get_position called
2020-05-29:04:28:15.756705: serial_flush called
2020-05-29:04:28:15.756705: serial_flush: tcflush
2020-05-29:04:28:15.756705: write_block called
2020-05-29:04:28:16.387057: meade_get_position: returned '000�00:00#+00�00:00#'
2020-05-29:04:28:16.387057: meade_get_position: az=000:00:00, el=000:00:00
2020-05-29:04:28:16.387057: dmmm2dec called
2020-05-29:04:28:16.387057: dmmm2dec called
2020-05-29:04:28:16.387057: rot_get_position: got az=0.00, el=0.00
0.00
0.00
2020-05-29:04:28:17.388375: rot_get_position called
2020-05-29:04:28:17.388375: meade_get_position called
2020-05-29:04:28:17.388375: serial_flush called
2020-05-29:04:28:17.388375: serial_flush: tcflush
2020-05-29:04:28:17.388375: write_block called
2020-05-29:04:28:18.022643: meade_get_position: returned '000�00:00#+00�00:00#'
2020-05-29:04:28:18.022643: meade_get_position: az=000:00:00, el=000:00:00
2020-05-29:04:28:18.022643: dmmm2dec called
2020-05-29:04:28:18.022643: dmmm2dec called
2020-05-29:04:28:18.022643: rot_get_position: got az=0.00, el=0.00
0.00
0.00
2020-05-29:04:28:18.022643: rot_close called
2020-05-29:04:28:18.022643: meade_close called
2020-05-29:04:28:18.022643: serial_flush called
2020-05-29:04:28:18.022643: serial_flush: tcflush
2020-05-29:04:28:18.022643: write_block called
2020-05-29:04:28:18.223109: ser_close called
2020-05-29:04:28:18.223109: ser_close: restoring options
2020-05-29:04:28:18.223109: rot_cleanup called
2020-05-29:04:28:18.223109: meade_cleanup called
```

mdblack98, 2020-05-29:
It looks like you only used 4 v's instead of 5.
And it's showing no movement at all in this log.  
So add "pause 5 p" to the command line too.

mdblack98, 2020-05-29:
Can you compile hamlib?
If so, clone from here and try it
https://github.com/Hamlib/Hamlib

If not I'll have to compile a version for you

mdblack98, 2020-05-29:
Try this binary set and see if it works OK.  It appears the elevation command was wrong.
https://www.dropbox.com/s/0zjw58t4svippt2/bin.zip?dl=0

Mike W9MDB

rkinnett, 2020-05-29:
> It looks like you only used 4 v's instead of 5. And it's showing no movement at all in this log.   So add "pause 5 p" to the command line too. 

Sorry, I should have copy-pasta'd the cmd line you provided.  Here's with 5 v's and the additional pause 5 p at the end:
```
2020-05-29:05:25:30.137463: rotctl Hamlib 4.0~git
Last commit was Wed May 27 10:56:40 2020 -0500
2020-05-29:05:25:30.137463: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-05-29:05:25:30.137463: rot_init called
2020-05-29:05:25:30.137463: initrots4_meade: _init called
2020-05-29:05:25:30.138461: rot_register (1801)
2020-05-29:05:25:30.138461: meade_init called version 0.2
2020-05-29:05:25:30.138461: set_conf: called
2020-05-29:05:25:30.138461: rot_open called
2020-05-29:05:25:30.138461: serial_open called
2020-05-29:05:25:30.138461: serial_open: OPEN before
2020-05-29:05:25:30.139459: serial_open: OPEN after
2020-05-29:05:25:30.139459: serial_open: serial_setup before
2020-05-29:05:25:30.139459: serial_setup called
2020-05-29:05:25:30.139459: serial_setup: tcgetattr
2020-05-29:05:25:30.139459: serial_setup: cfsetispeed
2020-05-29:05:25:30.139459: serial_setup: cfsetospeed
2020-05-29:05:25:30.139459: serial_setup: tcsetattr TCSANOW
2020-05-29:05:25:30.140456: serial_open: serial_setup after
2020-05-29:05:25:30.140456: serial_open: serial_flush before
2020-05-29:05:25:30.140456: serial_flush called
2020-05-29:05:25:30.140456: serial_flush: tcflush
2020-05-29:05:25:30.140456: serial_open: serial_flush before
2020-05-29:05:25:30.140456: meade_open called
2020-05-29:05:25:30.140456: serial_flush called
2020-05-29:05:25:30.140456: serial_flush: tcflush
2020-05-29:05:25:30.140456: write_block called
2020-05-29:05:25:30.341082: write_block(): TX 5 bytes
2020-05-29:05:25:30.341082: 0000    3a 47 56 50 23                                      :GVP#           
2020-05-29:05:25:30.341082: read_string called
2020-05-29:05:25:30.774793: read_string(): RX 9 characters
2020-05-29:05:25:30.774793: 0000    41 75 74 6f 73 74 61 72 23                          Autostar#       
2020-05-29:05:25:30.774793: meade_open product_name=Autostar
2020-05-29:05:25:30.774793: serial_flush called
2020-05-29:05:25:30.774793: serial_flush: tcflush
2020-05-29:05:25:30.774793: write_block called
2020-05-29:05:25:30.975227: write_block(): TX 16 bytes
2020-05-29:05:25:30.975227: 0000    3a 41 4c 23 3a 53 6f 30 30 23 3a 53 68 39 30 23     :AL#:So00#:Sh90#
2020-05-29:05:25:30.975227: Backend version: 0.2, Status: Stable
2020-05-29:05:25:30.975227: rot_set_position called az=180.00 el=30.00
2020-05-29:05:25:30.975227: rot_set_position: south_zero=0 
2020-05-29:05:25:30.975227: meade_set_position called: 180.00 30.00
2020-05-29:05:25:30.975227: serial_flush called
2020-05-29:05:25:30.975227: serial_flush: tcflush
2020-05-29:05:25:30.975227: write_block called
2020-05-29:05:25:31.175893: write_block(): TX 3 bytes
2020-05-29:05:25:31.175893: 0000    3a 44 23                                            :D#             
2020-05-29:05:25:31.175893: read_string called
2020-05-29:05:25:31.608692: read_string(): RX 1 characters
2020-05-29:05:25:31.608692: 0000    23                                                  #               
2020-05-29:05:25:31.608692: serial_flush called
2020-05-29:05:25:31.608692: serial_flush: tcflush
2020-05-29:05:25:31.608692: write_block called
2020-05-29:05:25:31.810168: write_block(): TX 25 bytes
2020-05-29:05:25:31.810168: 0000    3a 53 7a 20 31 38 30 2a 30 30 23 3a 53 61 2b 33     :Sz 180*00#:Sa+3
2020-05-29:05:25:31.810168: 0010    30 2a 30 30 23 3a 4d 41 23                          0*00#:MA#       
2020-05-29:05:25:31.810168: read_string called
2020-05-29:05:25:31.810168: read_string(): RX 3 characters
2020-05-29:05:25:31.810168: 0000    31 31 32                                            112             
2020-05-29:05:25:31.810168: meade_set_position: expected 110, got 112
Opened rot model 1801, 'LX200/Autostar'
set_pos: error = Invalid parameter
2020-05-29:05:25:32.810940: rot_get_position called
2020-05-29:05:25:32.810940: meade_get_position called
2020-05-29:05:25:32.810940: serial_flush called
[... 57 lines cut ...]
2020-05-29:05:25:41.906062: 0000    3a 47 5a 23 3a 47 41 23                             :GZ#:GA#        
2020-05-29:05:25:41.906062: read_string called
2020-05-29:05:25:42.334915: read_string(): RX 20 characters
2020-05-29:05:25:42.334915: 0000    30 30 31 df 32 32 3a 35 30 23 2b 31 31 df 31 34     001.22:50#+11.14
2020-05-29:05:25:42.334915: 0010    3a 35 38 23                                         :58#            
2020-05-29:05:25:42.334915: meade_get_position: returned '001�22:50#+11�14:58#'
2020-05-29:05:25:42.334915: meade_get_position: az=001:22:50, el=011:14:58
2020-05-29:05:25:42.334915: dmmm2dec called
2020-05-29:05:25:42.334915: dmmm2dec called
2020-05-29:05:25:42.334915: rot_get_position: got az=1.37, el=11.23
1.37
11.23
2020-05-29:05:25:42.334915: rot_close called
2020-05-29:05:25:42.334915: meade_close called
2020-05-29:05:25:42.334915: serial_flush called
2020-05-29:05:25:42.334915: serial_flush: tcflush
2020-05-29:05:25:42.335922: write_block called
2020-05-29:05:25:42.536412: write_block(): TX 3 bytes
2020-05-29:05:25:42.536412: 0000    3a 51 23                                            :Q#             
2020-05-29:05:25:42.536412: ser_close called
2020-05-29:05:25:42.536412: ser_close: restoring options
2020-05-29:05:25:42.537376: rot_cleanup called
2020-05-29:05:25:42.537376: meade_cleanup called

```

rkinnett, 2020-05-29:
> Try this binary set and see if it works OK.  It appears the elevation command was wrong. 

Success!

```
2020-05-29:05:39:56.196473: rotctl Hamlib 4.0~git
Last commit was Thu May 28 23:55:44 2020 -0500
2020-05-29:05:39:56.196473: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-05-29:05:39:56.196473: rot_init called
2020-05-29:05:39:56.196473: initrots4_meade: _init called
2020-05-29:05:39:56.196473: rot_register (1801)
2020-05-29:05:39:56.196473: meade_init called version 20200528
2020-05-29:05:39:56.196473: set_conf: called
2020-05-29:05:39:56.196473: rot_open called
2020-05-29:05:39:56.196473: serial_open called
2020-05-29:05:39:56.196473: serial_open: OPEN before
2020-05-29:05:39:56.197442: serial_open: OPEN after
2020-05-29:05:39:56.197442: serial_open: serial_setup before
2020-05-29:05:39:56.197442: serial_setup called
2020-05-29:05:39:56.198439: serial_setup: tcgetattr
2020-05-29:05:39:56.198439: serial_setup: cfsetispeed
2020-05-29:05:39:56.198439: serial_setup: cfsetospeed
2020-05-29:05:39:56.198439: serial_setup: tcsetattr TCSANOW
2020-05-29:05:39:56.198439: serial_open: serial_setup after
2020-05-29:05:39:56.198439: serial_open: serial_flush before
2020-05-29:05:39:56.198439: serial_flush called
2020-05-29:05:39:56.198439: serial_flush: tcflush
2020-05-29:05:39:56.198439: serial_open: serial_flush before
2020-05-29:05:39:56.198439: meade_open called
2020-05-29:05:39:56.198439: serial_flush called
2020-05-29:05:39:56.198439: serial_flush: tcflush
2020-05-29:05:39:56.198439: write_block called
2020-05-29:05:39:56.398936: write_block(): TX 5 bytes
2020-05-29:05:39:56.398936: 0000    3a 47 56 50 23                                      :GVP#           
2020-05-29:05:39:56.398936: read_string called
2020-05-29:05:39:56.829757: read_string(): RX 9 characters
2020-05-29:05:39:56.829757: 0000    41 75 74 6f 73 74 61 72 23                          Autostar#       
2020-05-29:05:39:56.829757: meade_open product_name=Autostar
2020-05-29:05:39:56.829757: serial_flush called
2020-05-29:05:39:56.829757: serial_flush: tcflush
2020-05-29:05:39:56.829757: write_block called
2020-05-29:05:39:57.030418: write_block(): TX 16 bytes
2020-05-29:05:39:57.030418: 0000    3a 41 4c 23 3a 53 6f 30 30 23 3a 53 68 39 30 23     :AL#:So00#:Sh90#
2020-05-29:05:39:57.030418: Backend version: 20200528, Status: Stable
2020-05-29:05:39:57.030418: rot_set_position called az=180.00 el=30.00
2020-05-29:05:39:57.030418: rot_set_position: south_zero=0 
2020-05-29:05:39:57.030418: meade_set_position called: 180.00 30.00
2020-05-29:05:39:57.030418: serial_flush called
2020-05-29:05:39:57.030418: serial_flush: tcflush
2020-05-29:05:39:57.030418: write_block called
2020-05-29:05:39:57.231853: write_block(): TX 3 bytes
2020-05-29:05:39:57.231853: 0000    3a 44 23                                            :D#             
2020-05-29:05:39:57.231853: read_string called
2020-05-29:05:39:57.664522: read_string(): RX 1 characters
2020-05-29:05:39:57.664522: 0000    23                                                  #               
2020-05-29:05:39:57.664522: serial_flush called
2020-05-29:05:39:57.664522: serial_flush: tcflush
2020-05-29:05:39:57.664522: write_block called
2020-05-29:05:39:57.864787: write_block(): TX 26 bytes
2020-05-29:05:39:57.864787: 0000    3a 53 7a 20 31 38 30 2a 30 30 23 3a 53 61 73 2b     :Sz 180*00#:Sas+
2020-05-29:05:39:57.864787: 0010    33 30 2a 30 30 23 3a 4d 41 23                       30*00#:MA#      
2020-05-29:05:39:57.864787: read_string called
2020-05-29:05:39:57.864787: read_string(): RX 3 characters
2020-05-29:05:39:57.864787: 0000    31 31 30                                            110             
Opened rot model 1801, 'LX200/Autostar'
2020-05-29:05:39:58.865575: rot_get_position called
2020-05-29:05:39:58.865575: meade_get_position called
2020-05-29:05:39:58.865575: serial_flush called
2020-05-29:05:39:58.865575: serial_flush: tcflush
[... 65 lines cut ...]
2020-05-29:05:40:08.400142: rot_get_position: got az=42.77, el=2.13
42.77
2.13
2020-05-29:05:40:08.400142: rot_close called
2020-05-29:05:40:08.400142: meade_close called
2020-05-29:05:40:08.400142: serial_flush called
2020-05-29:05:40:08.400142: serial_flush: tcflush
2020-05-29:05:40:08.400142: write_block called
2020-05-29:05:40:08.600612: write_block(): TX 3 bytes
2020-05-29:05:40:08.600612: 0000    3a 51 23                                            :Q#             
2020-05-29:05:40:08.600612: ser_close called
2020-05-29:05:40:08.600612: ser_close: restoring options
2020-05-29:05:40:08.601609: rot_cleanup called
2020-05-29:05:40:08.601609: meade_cleanup called

```

Fantastic!

On a side note, I also have a Celestron mount, a couple of iOptrons, and a SkyWatcher (Synta, like Orion).  I was only planning to operate the Meade mount as a satellite tracker with hamlib, but I can help with testing on the others if that would be useful to you.

Thanks so much for the quick turnaround.

Cheers
-Ryan, NG6F

rkinnett, 2020-05-29:
I seem to have jumped the gun.  Doesn't look the elevation is changing much at all.  In the log in my previous post, it went backward a few degrees.

I'm experimenting with direct commanding now to see if I can figure out why the target elevation seems to stay at zero even after the set-elevation command succeeds.

rkinnett, 2020-05-29:
Hey Mike,

Sorry to say this, I think the elevation command was correct before the change.  That little "s" shown in the protocol specification for the set-altitude command is not part of the opcode but rather a placeholder for the sign of the elevation angle.  I confirmed this by experimenting using a terminal app.  Further, I was finding that even manual commanding worked sometimes and not others.  It seems the mount software gets in a weird state and refuses valid slew commands after some trigger which I haven't found yet.  Power cycling clears that state.  I tried this with two firmware versions and observed the same behavior with both.  I don't know if this might be a systematic problem with many Meade mounts or is particular to my mount model or even my specific unit.  Anyways, after realizing that I can restore slew capability by rebooting the mount, I went back and tested the pre-release v4.0 which you pointed me to prior to sending me the newer point build after the adjustment to the elevation command, and it turns out that version works just fine.  You can revert that change.

Thanks again!

mdblack98, 2020-05-29:
Can you please test this again?
I'd like to see the debug where your elevation is working after you reboot it.
In the original debug you sent it was getting a bad return code which may be your clue that a reboot is needed.

It was the 110 - 112 error that was reported.
I want to ensure it's getting the correct return code when it's working correctly.

Mike

rkinnett, 2020-05-29:
Here's a test run using the w64-4.0~git version.  This is what I did:

1. Booted the mount
2. Started rotctl
3. Read position (0, 0)
4. Commanded position (10, 20) - failed
5. Rebooted the mount
6. Read position (0, 0)
7. Commanded position (10, 20) - succeeded (verified visually)
8. Read position (9.98, 19.98)
9. Commanded position (0, 0) - succeeded
10. Read position (0, 0)

````
 c:\Program Files (x86)\hamlib-w64-4.0~git\bin>rotctl -m 1801 -r COM3 -vvvvv -Z
2020-05-29:17:34:28.931221: rotctl Hamlib 4.0~git
Last commit was Wed May 27 10:56:40 2020 -0500
2020-05-29:17:34:28.932190: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-05-29:17:34:28.932190: rot_init called
2020-05-29:17:34:28.932190: initrots4_meade: _init called
2020-05-29:17:34:28.932190: rot_register (1801)
2020-05-29:17:34:28.933189: meade_init called version 0.2
2020-05-29:17:34:28.933189: set_conf: called
2020-05-29:17:34:28.933189: rot_open called
2020-05-29:17:34:28.933189: serial_open called
2020-05-29:17:34:28.933189: serial_open: OPEN before
2020-05-29:17:34:28.934197: serial_open: OPEN after
2020-05-29:17:34:28.935189: serial_open: serial_setup before
2020-05-29:17:34:28.935189: serial_setup called
2020-05-29:17:34:28.935189: serial_setup: tcgetattr
2020-05-29:17:34:28.935189: serial_setup: cfsetispeed
2020-05-29:17:34:28.935189: serial_setup: cfsetospeed
2020-05-29:17:34:28.935189: serial_setup: tcsetattr TCSANOW
2020-05-29:17:34:28.936180: serial_open: serial_setup after
2020-05-29:17:34:28.936180: serial_open: serial_flush before
2020-05-29:17:34:28.936180: serial_flush called
2020-05-29:17:34:28.936180: serial_flush: tcflush
2020-05-29:17:34:28.936180: serial_open: serial_flush before
2020-05-29:17:34:28.936180: meade_open called
2020-05-29:17:34:28.936180: serial_flush called
2020-05-29:17:34:28.936180: serial_flush: tcflush
2020-05-29:17:34:28.937178: write_block called
2020-05-29:17:34:29.137827: write_block(): TX 5 bytes
2020-05-29:17:34:29.137827: 0000    3a 47 56 50 23                                      :GVP#
2020-05-29:17:34:29.138643: read_string called
2020-05-29:17:34:29.569635: read_string(): RX 9 characters
2020-05-29:17:34:29.570527: 0000    41 75 74 6f 73 74 61 72 23                          Autostar#
2020-05-29:17:34:29.571488: meade_open product_name=Autostar
2020-05-29:17:34:29.571488: serial_flush called
2020-05-29:17:34:29.572479: serial_flush: tcflush
2020-05-29:17:34:29.572479: write_block called
2020-05-29:17:34:29.774228: write_block(): TX 16 bytes
2020-05-29:17:34:29.775184: 0000    3a 41 4c 23 3a 53 6f 30 30 23 3a 53 68 39 30 23     :AL#:So00#:Sh90#
Opened rot model 1801, 'LX200/Autostar'
2020-05-29:17:34:29.776225: Backend version: 0.2, Status: Stable

Rotator command: p
2020-05-29:17:34:34.854700: rot_get_position called
2020-05-29:17:34:34.855698: meade_get_position called
2020-05-29:17:34:34.855698: serial_flush called
2020-05-29:17:34:34.856695: serial_flush: tcflush
2020-05-29:17:34:34.856695: write_block called
2020-05-29:17:34:35.057159: write_block(): TX 8 bytes
2020-05-29:17:34:35.058156: 0000    3a 47 5a 23 3a 47 41 23                             :GZ#:GA#
2020-05-29:17:34:35.058156: read_string called
2020-05-29:17:34:35.492153: read_string(): RX 20 characters
2020-05-29:17:34:35.493004: 0000    30 30 30 df 30 30 3a 30 30 23 2b 30 30 df 30 30     000.00:00#+00.00
2020-05-29:17:34:35.493995: 0010    3a 30 30 23                                         :00#
2020-05-29:17:34:35.493995: meade_get_position: returned '000▀00:00#+00▀00:00#'
[... 129 lines cut ...]
2020-05-29:17:36:54.745992: 0010    3a 30 37 23                                         :07#
2020-05-29:17:36:54.746990: meade_get_position: returned '000▀00:09#+00▀00:07#'
2020-05-29:17:36:54.748987: meade_get_position: az=000:00:09, el=000:00:07
2020-05-29:17:36:54.749983: dmmm2dec called
2020-05-29:17:36:54.751985: dmmm2dec called
2020-05-29:17:36:54.755970: rot_get_position: got az=0.00, el=0.00
Azimuth: 0.00
Elevation: 0.00

Rotator command: q
2020-05-29:17:36:56.215299: rot_close called
2020-05-29:17:36:56.217279: meade_close called
2020-05-29:17:36:56.218285: serial_flush called
2020-05-29:17:36:56.219276: serial_flush: tcflush
2020-05-29:17:36:56.220272: write_block called
2020-05-29:17:36:56.420928: write_block(): TX 3 bytes
2020-05-29:17:36:56.421798: 0000    3a 51 23                                            :Q#
2020-05-29:17:36:56.423738: ser_close called
2020-05-29:17:36:56.424730: ser_close: restoring options
2020-05-29:17:36:56.427721: rot_cleanup called
2020-05-29:17:36:56.427721: meade_cleanup called

c:\Program Files (x86)\hamlib-w64-4.0~git\bin>

````

mdblack98, 2020-05-29:
Can you contact me off list please?  mdblack98@yahoo.com 
I'd like to do some testing.  I find it bothersome that a reboot of the rotator makes things work.  You can see the 112 error that happens and the MA# command does not show a return value of 2 in the manuals I have.  The only thing really set is the min/max of elevation s
It implies that one of the commands being sent during initialization may be messing up things.
Mike

 
 

   
   - Booted the mount
   - Started rotctl
   - Read position (0, 0)
   - Rebooted the mount
   - Read position (0, 0)
   - Read position (9.98, 19.98)

mdblack98, 2020-05-31:
All working now.

closed 2020-05-31 by mdblack98