Hamlib/Hamlib #268: Change dmmm2dec to allow for seconds argument

mdblack98, 2020-05-29:
double HAMLIB_API dmmm2dec(int degrees, double minutes, int sw)
double HAMLIB_API dmmm2dec(int degrees, double minutes, double seconds, int sw)

N0NB, 2020-05-29:
This is already provided by the dms2dec or D command in rotctl.

The dmmm2dec handles the format given as DEG MINS.mins which already includes the seconds in the fractional part.

mdblack98, 2020-05-29:
Hmmm...doesn't seem so....this got converted incorrectly as it is missing seconds in dmmm2dec
2020-05-29:05:25:33.438297: read_string(): RX 20 characters
2020-05-29:05:25:33.438297: 0000    30 30 31 df 32 32 3a 35 30 23 2b 31 31 df 31 34     001.22:50#+11.14
2020-05-29:05:25:33.438297: 0010    3a 35 38 23                                         :58#            
2020-05-29:05:25:33.438297: meade_get_position: returned '001�22:50#+11�14:58#'
2020-05-29:05:25:33.438297: meade_get_position: az=001:22:50, el=011:14:58
2020-05-29:05:25:33.438297: dmmm2dec called
2020-05-29:05:25:33.438297: dmmm2dec called
2020-05-29:05:25:33.438297: rot_get_position: got az=1.37, el=11.23
1.37
11.23

N0NB, 2020-05-30:
If I read that correctly, the device returned an azimuth of 1 deg 22.5 min and an elevation of 11 deg 14.58 min?

Using rotclt:

```
Rotator command: E
Degrees: 1
Dec Minutes: 22.5
S/W: 0
Dec Deg: 1.375000

Rotator command: E
Degrees: 11
Dec Minutes: 14.58
S/W: 0
Dec Deg: 11.243000
```

Or, going to separate values:

```
Rotator command: d
Dec Degrees: 1.375
Degrees: 1
Minutes: 22
Seconds: 30.000000
S/W: 0

Rotator command: d
Dec Degrees: 11.243
Degrees: 11
Minutes: 14
Seconds: 34.800000
S/W: 0

```

I would agree that there is a rounding error somewhere in the trace, or in the return from rotctl.  I did not calculate these by hand, but they look to be ballpark correct.

I'll admit that I erred in my quick reply this morning as I thought there was a dmmm2dms function and there is not.

Looking at your proposed API, if given something like 10 deg 10.8 min 10.3 sec, what is the integer value of the seconds?

mdblack98, 2020-05-30:
2020-05-29:05:25:33.438297: meade_get_position: returned '001�22:50#+11�14:58#'2020-05-29:05:25:33.438297: meade_get_position: az=001:22:50, el=011:14:582020-05-29:05:25:33.438297: dmmm2dec called2020-05-29:05:25:33.438297: dmmm2dec called2020-05-29:05:25:33.438297: rot_get_position: got az=1.37, el=11.231.3711.23

No, 1 deg, 22min, 50secAnd 11 deg, 14min, 58sec
But the seconds are dropped. So we get
1+22/60 = 1.3711+14/60 = 11.23
With seconds added1+22/60+50/3600 = 1.3811+14/160+58/3600 = 11.25
We should be representing whatever the rotor returns.
Mike

mdblack98, 2021-04-12:
This API will be revamped in 5.0 to be consistent along with the documentation
All arguments will be double precision to allow for future usage.

Mike

N0NB, 2025-06-14:
Digging into this further, the underlying bug is that `meade.c` has been calling the incorrect function.  It should have been calling `dms2dec` instead.  I believe I've corrected this and a PR is forthcoming to revert commit 17b404a6 and fix `meade.c`.

closed 2025-06-14 by N0NB