Hamlib/Hamlib #272: rotctl issuse with GS-232B

ve9gj, 2020-05-31:
I am submitting this report on behalf of another ham.  I don't have the hardware myself.  His issue looked similar to the issue I reported about rotctld returning Ev and Az reversed.  He built the latest hamlib and now  rotctl is reporting elevation on a Az only rotor.  We have confirmed it is happening with rotctl itself but it does behave the same with rotctld.
He has a Yaesu G1000 with the GS-232B interface.  When using the P command the rotor will turn to the correct bearing.
73
Glenn

rotctl -vvvvv -m 603 -r /dev/ttyUSB1
rotctl Hamlib 4.0~git
Last commit was Fri May 29 07:56:24 2020 -0500
Report bugs to <hamlib-developer@lists.sourceforge.net>
rot_init called
initrots4_gs232a called
rot_register (601)
rot_register (602)
rot_register (603)
rot_register (604)
rot_register (605)
rot_register (606)
rot_register (607)
rot_register (608)
set_conf: called
rot_open called
serial_open called
serial_open: OPEN before
serial_open: OPEN after
serial_open: serial_setup before
serial_setup called
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed
serial_setup: cfsetospeed
serial_setup: tcsetattr TCSANOW
serial_open: serial_setup after
serial_open: serial_flush before
serial_flush called
serial_flush: tcflush
serial_open: serial_flush before
Opened rot model 603, 'GS-232B'
Backend version: 20200517, Status: Stable
Rotator command:

Rotator command: p
rot_get_position called
gs232b_rot_get_position called
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.
read_string called
read_string(): RX 15 characters
0000    41 5a 3d 31 38 31 20 20 45 4c 3d 30 30 37 0d        AZ=181  EL=007.
gs232b_rot_get_position: (az, el) = (181, 7)
rot_get_position: got az=181.00, el=7.00
Azimuth: 181.00
Elevation: 7.00

Rotator command: P 240 0
Azimuth: Elevation: rot_set_position called az=240.00 el=0.00
rot_set_position: south_zero=0
gs232b_rot_set_position called: az=240.00 el=0.00
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 9 bytes
0000    57 32 34 30 20 30 30 30 0d                          W240 000.
read_string called
read_string(): Timed out 0.400428 seconds after 0 chars
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 9 bytes
0000    57 32 34 30 20 30 30 30 0d                          W240 000.
read_string called
read_string(): Timed out 0.400425 seconds after 0 chars
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 9 bytes
0000    57 32 34 30 20 30 30 30 0d                          W240 000.
read_string called
read_string(): Timed out 0.400425 seconds after 0 chars
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 9 bytes
0000    57 32 34 30 20 30 30 30 0d                          W240 000.
read_string called
read_string(): Timed out 0.400426 seconds after 0 chars
set_pos: error = Communication timed out

w4jkl, 2020-05-31:
Mike, Glenn,

If the other ham has a REAL GS-232B interface, rotct(d) should be made 
to work with it, unless it has the ability to reverse the axes. My 
interface can be reversed, and I set it up to work with hamlib 3.3.

John W4JKL

mdblack98, 2020-06-01:
The rotor is reporting elevation so elevation you will get.That's the GS232B protocol.  The fact that you have an azimuth-only is not detectable that I know of.
But I'm putting in a patch for fixing the timeout which hopefully doesn't mess up any other GS232B's.  I don't think we need to wait for the W command (apparently not in this case for sure).
Mike

closed 2020-06-02 by mdblack98