Hamlib/Hamlib #335: on tag 3.3 any rigctl command makes my ic-7300 go into tx and send a single dit

etamme, 2020-07-02:
I was having this issue on 3.1, so i updated to 3.3, but it persists.  

running ANY command, for example

./rigctl -m 373 -r /dev/ttyUSB0 f

will make my ic-7300 go into tx mode and send a single dit.  verbose output below.

`rigctl, Hamlib 3.3
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
icom: _init called
rig_register called
rig_register: rig_register (355)
rig_register called
rig_register: rig_register (309)
rig_register called
rig_register: rig_register (310)
rig_register called
rig_register: rig_register (311)
rig_register called
rig_register: rig_register (313)
rig_register called
rig_register: rig_register (314)
rig_register called
rig_register: rig_register (315)
rig_register called
rig_register: rig_register (319)
rig_register called
rig_register: rig_register (320)
rig_register called
rig_register: rig_register (321)
rig_register called
rig_register: rig_register (322)
rig_register called
rig_register: rig_register (367)
rig_register called
rig_register: rig_register (323)
rig_register called
rig_register: rig_register (346)
rig_register called
rig_register: rig_register (324)
rig_register called
rig_register: rig_register (328)
rig_register called
rig_register: rig_register (330)
rig_register called
rig_register: rig_register (326)
rig_register called
rig_register: rig_register (327)
rig_register called
rig_register: rig_register (347)
rig_register called
rig_register: rig_register (357)
rig_register called
rig_register: rig_register (363)
rig_register called
rig_register: rig_register (329)
rig_register called
rig_register: rig_register (362)
rig_register called
rig_register: rig_register (345)
rig_register called
rig_register: rig_register (356)
rig_register called
rig_register: rig_register (375)
rig_register called
rig_register: rig_register (360)
rig_register called
rig_register: rig_register (370)
rig_register called
rig_register: rig_register (361)
rig_register called
rig_register: rig_register (373)
rig_register called
rig_register: rig_register (378)
rig_register called
rig_register: rig_register (331)
rig_register called
rig_register: rig_register (312)
rig_register called
rig_register: rig_register (316)
rig_register called
rig_register: rig_register (332)
rig_register called
rig_register: rig_register (334)
rig_register called
rig_register: rig_register (344)
rig_register called
rig_register: rig_register (368)
rig_register called
rig_register: rig_register (335)
rig_register called
rig_register: rig_register (369)
rig_register called
rig_register: rig_register (377)
rig_register called
rig_register: rig_register (336)
rig_register called
rig_register: rig_register (358)
rig_register called
rig_register: rig_register (337)
rig_register called
rig_register: rig_register (338)
rig_register called
rig_register: rig_register (339)
rig_register called
rig_register: rig_register (340)
rig_register called
rig_register: rig_register (341)
rig_register called
rig_register: rig_register (342)
rig_register called
rig_register: rig_register (343)
rig_register called
rig_register: rig_register (366)
rig_register called
rig_register: rig_register (303)
rig_register called
rig_register: rig_register (304)
rig_register called
rig_register: rig_register (306)
rig_register called
rig_register: rig_register (307)
rig_register called
rig_register: rig_register (302)
rig_register called
rig_register: rig_register (352)
rig_register called
rig_register: rig_register (353)
rig_register called
rig_register: rig_register (351)
rig_register called
rig_register: rig_register (364)
rig_register called
rig_register: rig_register (365)
rig_register called
rig_register: rig_register (354)
rig_register called
rig_register: rig_register (371)
rig_register called
rig_register: rig_register (372)
rig_register called
rig_register: rig_register (374)
rig_register called
rig_register: rig_register (376)
icom_init called
icom_init: civ_version=1
rig_open called
port_open called
serial_open called
serial_setup called
rig_get_vfo called
Opened rig model 373, 'IC-7300'
rig_strstatus called
Backend version: 0.8.3, Status: Stable
rig_get_freq called
icom_get_freq called
serial_flush called
write_block called
write_block(): TX 6 bytes
0000    fe fe 94 e0 03 fd                                   ......          
read_string called
read_string(): RX 6 characters
0000    fe fe 94 e0 03 fd                                   ......          
read_string called
read_string(): RX 11 characters
0000    fe fe e0 94 03 00 20 05 14 00 fd                    ...... ....     
from_bcd called
14052000
rig_close called
port_close called
ser_close called
rig_cleanup called
icom_cleanup called
`

etamme, 2020-07-02:
I tried switching the rig from using RTS based keying to DTR but the issue still persists.

mdblack98, 2020-07-02:
Please upgrade to 4.0 and let me know if the issue still persists.
https://github.com/Hamlib/Hamlib

Mike W9MDB

etamme, 2020-07-02:
By 4.0 do you mean master? I dont see any tags or branches for 4.0

etamme, 2020-07-02:
I did the following: 

git checkout master
git pull origin master
git clean -fxd
./autoconfig && ./configure && make
tests/rigctl -m 3073 -r /dev/ttyUSB0 

My IC-7300 went into tx mode and sent a single dit still.  I compiled on commit 1ad029bfb3322b3ef34f67d732c7e29ba450e979

mdblack98, 2020-07-02:
Try setting the baud rate too...adjust the 9600 to match the rig.
tests/rigctl -m 3073 -r /dev/ttyUSB0 -s  9600

etamme, 2020-07-02:
Specifying the baud did not make any difference.  

I tried running 

rigctl -vvvvvvvv -m 3073 -s 9600 -r /dev/ttyUSB0

To see if interactive mode made any difference - the whole time i was in interactive mode my radio went into tx mode and was tx'ing.  So its just keying down ....

etamme, 2020-07-02:
I did some more testing.

I tried disabling USB CW keying, this unsurprisingly "fixed" the problem.  However, I also tested USB Keying using RTS and DTR, either causes the rig to key.  So rigctl is definitely keying down both of those pins.

mdblack98, 2020-07-02:
According to notes in the hamlib code Linux sets the DTR/RTS lines during the opening of a serial port and we have to  turn them back off again.
So it appears that's what you're seeing.
Mike W9MDB

etamme, 2020-07-03:
But that does not account for the perpetual keydown of running rigctl in interactive mode.

etamme, 2020-07-03:
Also, for what its worth, I have written a serial [serial keyer in python](https://github.com/etamme/cwkeyer), it does not experience this behavior when I open the serial device.

etamme, 2020-07-03:
more anecdotal information, this does not happen when I had a friend test on ununtu 20.04 with his ic-7100

mdblack98, 2020-07-03:
Maybe because he doesn't have any CW keying turned on?
Mike

mdblack98, 2020-07-03:
Is this on a built-in serial port or a USB serial port?
Mike W9MDB

mdblack98, 2020-07-03:
I just tested on a USB serial port and I don't see DTR or RTS being toggled at all.
Mike W9MDB

etamme, 2020-07-03:
I am testing over the built in USB port on the ic-7300.  If you are using the built in USB port for the 7300, can you save and share your ic-7300 config with me?

mdblack98, 2020-07-03:
One more thing to try...
Try these switches You may need ON instead of OFF.

--set-conf=dtr_state=OFF--set-conf=rts_state=OFFAlso try this one--set-conf=dcd_type=None

Mike W9MDB

etamme, 2020-07-03:
Thank you Mik!

specifying 

-set-conf=dtr_state=OFF --set-conf=rts_state=OFF

Solved the problem

My full test command for anyone who comes across this thread in the future:

rigctl --set-conf=dtr_state=OFF --set-conf=rts_state=OFF -m 3073  -r /dev/ttyUSB0 f

etamme, 2020-07-03:
So ... I have thought and experimented with this.  I am still considering this a bug.  Ideally rigctl should talk to the rig without doing ANYTHING to the rts and dtr pins.  

The example here is that I am using it with my logger to check the frequency, to get the band, and check for dupes while cw keying is happening.  If rigctl disables both rts and dtr, it messes up the CW being sent.

Using the command line args to turn off dtr and rts ... is a mostly workable workaround, but It seems counter intuitive to me that rigctl would change the state of the rts or dtr pins without being instructed to do so for a specific purpose.

mdblack98, 2020-07-03:
If you turn on verbose debug with -vvvvv you should NOT see any setting of DTR/RTS when you eliminate those switches.
Can you run ioninja on your system?  It will show you any line changes that occur.https://ioninja.com/downloads.html

I cannot duplicate any such behavior here.  
What OS are you running?
Mike W9MDB

etamme, 2020-07-03:
Here is a[gif screen cap of rigctl setting both rts and dtr](https://i.imgur.com/N9l8PU0.gif).  However, I discovered something else, which points back to the linux subsystem as you had mentioned previously.  

When opening statserial, it too sets  both rts and dtr bits.  So there must be something in the way a serial port is opened that causes the linux subsystem to set the pins high. 

So I will have to live with the work around till i can dig into the linux serial subsystem deeper.

Thanks for your patience in this Mike.

closed 2020-07-03 by etamme