Hamlib/Hamlib #338: FlDigi with hamlib 4.0 PTT fails on TS-480

mdblack98, 2020-07-06:
TS-480 PTT fails with FLDigi  4.1.13.20 with recent hamlib.
No PTT at all.
PTT does work with TS-2000 using TS-480 selection.

g4wjs, 2020-07-08:
Mike,

I'm confused by your commit comment. The RX; command returns the rig to receive, it is a set command not a query. Of course the TS-480 CAT manual is wrong.

73
Bill
G4WJS.

mdblack98, 2020-07-08:
The TS-480 was returning "?" when you did "RX;" and the rig was already in RX.So yes...the comment says the manual is wrong.
So this would flowTX1; << no responseRX;  << no-responseRX; << ?; So I removed the ID check for the RX command.
It showed up in FLDigi since it was doing RX at startup.
Mike W9MDB

g4wjs, 2020-07-08:
Mike,

to be clear, I am asking why RX has to be a special case. We send RX;ID: and either get ID020; back which shows the RX command worked and has finished, or we get back ?;ID020; which shows the RX command was rejected, probably because the rig was in RX mode already. Nothing execptional here.

73
Bill
G4WJS.

g4wjs, 2020-07-08:
Mike,

by jumping out you defeat the purpose of the verification and may end up sending another command before the rig is ready, which leads back to the whole mess of unnecessary delays.

73
Bill
G4WJS.

mdblack98, 2020-07-08:
Well it was returning RIG_ERJCTED.  FLDigi upgraded to 4.0 and that error was reported.  3.0 didn't have the error.
WSJT-X apparently has just never run into this...or at least haven't seen any reports of the rejected command.
So as of now it's a special case for the RX command for the TS-480 that I decided to just let the command go as no idea what other rigs it might affect.
Mike W9MDB

g4wjs, 2020-07-08:
Hi Mike,

WSJT-X always sends a PTT off request on exit as a safety measure, it ignores any errors there. All other invocations should be correctly sequenced as I don't think it is possible to cancel a CAT TX; request from the rig other than turning it off, which would kill the CAT connection anyway. I suppose there may be a race condition where you could turn the rig off an back on without losing CAT control, then an RX command might cause an error. I would not be too concerned about that happening!

I don't think it is reasonable to lie about the rig rejecting a command quite reasonably, just because a client doesn't like it, especially as it may have much deeper consequences.

73
Bill
G4WJS.

mdblack98, 2020-07-08:
The other way I'd prefer to do is ask for PTT status to ensure the RX turned it off and repeat if not.That would in lieu of doing the ID check.
Mike W9MDB

g4wjs, 2020-07-08:
Hi Mike,

that might get you into trouble, a good number of rigs emulate the 
TS-480 or other Kenwood models, I would be surprised if they all report 
SEND status accurately. You could end up in a loop.

73
Bill
G4WJS.

On 08/07/2020 18:19, Michael Black wrote:
> The other way I'd prefer to do is ask for PTT status to ensure the RX
> turned it off and repeat if not.That would in lieu of doing the ID check.
> Mike W9MDB
>
>
>
> On Wednesday, July 8, 2020, 12:14:18 PM [...]

mdblack98, 2020-07-08:
It was firing an ID; command immediately after the RX and if that failed would send RX again.  Not sure if that was ever getting hit.
So if a following command is too quick it should fail with "?" and do the repeat there instead so it seems to me that possibility is still handled.

Mike

closed 2020-07-08 by mdblack98