Hamlib/Hamlib #350: Binding configuration for python 3.8 fails in configure on OS/X.

billshine, 2020-07-23:
os x issue.
I built hamlib for python 2.7.  Trying to do the configure for 3.8, I got an error, because the extra linking flag is broken:
[config.log](https://github.com/Hamlib/Hamlib/files/4963830/config.log)
 Python.framework/Versions/3.8/Python as a framework :
checking python extra linking flags... -Wl,-stack_size,1000000  -framework CoreFoundation Python.framework/Versions/3.8/Python
checking consistency of all components of python development environment... no
configure: error: in `/Users/bill/work/hamlib-3.3':
configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LIBS environment variable.
  Example: ./configure LIBS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

See `config.log' for more details

billshine, 2020-07-23:
the problem is with the python m4 macros -- they are trying to get the ldflags from distutils.sysconfig, instead of calling python3_config.  the returned ldflags are not prefixed with  /Library/Frameworks ...  python3_config --ldflags gives different ldflags.  I am not sure if those are correct.

mdblack98, 2020-07-23:
You guys can't receive WWV?
Mike W9MDB

mdblack98, 2020-07-23:
Oops...this went to the wrong list...please ignore...
Mike

mdblack98, 2020-07-25:
Could you try the latest macro from here and see if it works...doesn't look like it will...
https://git.savannah.gnu.org/gitweb/?p=autoconf-archive.git;a=blob_plain;f=m4/ax_python_devel.m4

Mike W9MDB

mdblack98, 2020-07-29:
I see that you're using hamlib-3.3 -- can you please try the latest 4.0 ?

Mike W9MDB

mdblack98, 2021-02-27:
See
https://github.com/Hamlib/Hamlib/issues/477

closed 2021-02-27 by mdblack98