Hamlib/Hamlib #354: FT991-A SWR incorrect

iodigitalsec, 2020-07-29:
Whilst holding down the transmit on 433.500 with 5w power, l SWR shows 0.050980 whilst SWR is actually about 1.20

How do I turn the value from 'l SWR' into the actual SWR reading?

mdblack98, 2020-07-29:
Can you compile hamlib?
Nobody has ever implemented getting a calibrated SWR answer on Yaeus rigs.
We've done it for many Icom rigs.  
We can do this for Yaesu's too.
This is the default Icom table        {0, 1.0f},        {48, 1.5f},        {80, 2.0f},        {120, 3.0f},        {240, 6.0f}
You can see the 0-255 value returned by the Yaesu RM command with this rigctl commandw RM6;
If you can provide several SWR readings across the scale of 0-255 we can put in a default Yaesu swr calibration which will, of course, have to be tested against other rigs and maybe customized for other Yaesu's.
Mike W9MDB

iodigitalsec, 2020-07-29:
Dear Mike,

Yes. I can compile the latest version from scratch and do this.

What data would be most useful? 

a) The output of RM6 command (0 to 255) vs the actual SWR showing on
the radio? 

or

b) The output of RM6 vs the output of 'l SWR' ?

73s,

Adam M7OTP

mdblack98, 2020-07-29:
Yes...
Just like SWR 1,2,3,4,5,6,etc....
If you can.  Need at least two points and preferably more.
Mike

iodigitalsec, 2020-07-29:
SWR 1.0:
l SWR
0.050980
RM6
RPRT 0

SWR 1.1:
l SWR
0.050980
RM6
RPRT 0

SWR 1.35:
l SWR
0.152941
RM6
RPRT 0

SWR 2.0:
l SWR
0.349020
RM6
RPRT 0

SWR 5.0:
l SWR
0.949020
RM6
RPRT 0

At SWR > 5.0, l SWR returns 1.000000

iodigitalsec, 2020-07-29:
If it's any use, I've also compiled a list of the output of 'l RFPOWER'
vs actual RFPOWER on the radio, starting at the minimum of 5W and
finishing on 100W:

0.019608, 0.023529, 0.027451, 0.031373, 0.035294, 0.039216, 0.043137,
0.047059, 0.050980, 0.054902, 0.058824, 0.062745, 0.066667, 0.070588,
0.074510, 0.078431, 0.082353, 0.086275, 0.090196, 0.094118, 0.098039,
0.101961, 0.105882, 0.109804, 0.113725, 0.117647, 0.121569, 0.125490,
0.129412, 0.133333, 0.137255, 0.141176, 0.145098, 0.149020, 0.152941,
0.156863, 0.160784, 0.164706, 0.168627, 0.172549, 0.176471, 0.180392,
0.184314, 0.188235, 0.192157, 0.196078, 0.200000, 0.203922, 0.207843,
0.211765, 0.215686, 0.219608, 0.223529, 0.227451, 0.231373, 0.235294,
0.239216, 0.243137, 0.247059, 0.250980, 0.254902, 0.258824, 0.262745,
0.266667, 0.270588, 0.274510, 0.278431, 0.282353, 0.286275, 0.290196,
0.294118, 0.298039, 0.301961, 0.305882, 0.309804, 0.313726, 0.317647,
0.321569, 0.325490, 0.329412, 0.333333, 0.337255, 0.341176, 0.345098,
0.349020, 0.352941, 0.356863, 0.360784, 0.364706, 0.368627, 0.372549,
0.376471, 0.380392, 0.384314, 0.388235, 0.392157

Regards,

Adam

closed 2020-08-04 by mdblack98