Hamlib/Hamlib #355: Not compatible with the latest uhd

FFY00, 2020-07-30:
```
usrp_impl.cc:35:10: fatal error: usrp_standard.h: No such file or directory
   35 | #include <usrp_standard.h>
```

mdblack98, 2020-07-30:
What software package is uhd?
Mike W9MDB

FFY00, 2020-07-30:
https://github.com/EttusResearch/uhd

Up until now have been building with `sed -i 's|usrp|uhd|g' configure.ac`.

mdblack98, 2020-07-30:
I assume you have libusrp-dev installed?  That's where that include file is located.
Mike W9MDB

 
 

https://github.com/EttusResearch/uhd

FFY00, 2020-07-30:
I am using arch, but I do have UHD installed, all the header files should be present. I am using the latest release, 3.15.0.0, and the file doesn't seem to be shipped there.

mdblack98, 2020-07-30:
I can compile the latest UHD without USRP being enabled.  So it doesn't appear to be related to UHD.You have --enable-usrp in your config?  
That would be a different problem.
Mike

FFY00, 2020-07-30:
I do have it, yes. I want to build with support for it.

mdblack98, 2020-07-30:
Then you need to install libusrpThat's where the include file is located.Sounds like configure is finding the library but then the build is not finding the include file.Which sounds like you don't have a development package of libusrp.
Mike W9MDB

FFY00, 2020-07-30:
Oh, I didn't notice libusrp existed. Is there any reason for you to use it instead of the standard libuhd?

By using UHD you automatically get compatibility with the whole SoapySDR environment, enabling devices like the LimeSDR, bladeRF, PlutoSDR, etc.

mdblack98, 2020-07-30:
Not my code...don't know why that is there other than probably a different interface to the library.
You don't need to enable usrp.You apparently have some part of it installed if it configures with it.
Mike

mdblack98, 2020-08-03:
This was a user configuration problem.

closed 2020-08-03 by mdblack98