Hamlib/Hamlib #357: FT-817 RigFailure Hamlib error: Target VFO unaccessible...

ah1102, 2020-08-08:
WSJT-X 2.2.1 (WSJT-X 2.2.2) error:

`RigFailure Hamlib error: Target VFO unaccessible while getting current frequency`

**WSJT-X 2.2.0 works!!!** Thanks to the author https://www.mail-archive.com/wsjt-devel@lists.sourceforge.net/msg21159.html

My system:
- macOS (10.15.5)
- Yaesu FT-817
- Interface Yaesu SCU-17

CP210x USB to UART Bridge VCP Drivers:
 - Mac_OSX_VCP_Driver CP2105 version 5.0.10 (the same on updated 5.3.5) 
    https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers 

[image]
[image]
[image]

mdblack98, 2020-08-08:
Can you try the FT-897 and see if it works?
And can you compile hamlib if I post a patch?

Mike W9MDB

ah1102, 2020-08-08:
I tried using FT-897 profile on wsjtx-2.2.1 and it didn't fix the problem.
I can try to compile.

walzert, 2020-08-11:
Had the same issue by using rigctl/d  with my FT 818 and trying the model FT-897 instead is working.

Fedora 32 
Hamlib 4.0~git

mdblack98, 2020-08-11:
You should be able to run the FT-817 with Mode=None instead of Data/Pkt and just set the rig's mode yourself.
I would recommend that for the FT-817, 857, and 897.  The ability to set the mode was added but the eeprom has a limited lifetime of writes and setting mode takes an eeprom write.  Once the eeprom burns out you have a useless rig as the eeprom is no longer manufactured and can't be replaced.

Mike W9MDB

closed 2020-08-11 by mdblack98