Hamlib/Hamlib #364: IC-705 Support

mdblack98, 2020-09-06:
Implement IC-705

closed 2020-09-06 by mdblack98

mikaelnousiainen, 2020-09-07:
@mdblack98 The IC-7300 and IC-705 caps are now mixed here

mikaelnousiainen, 2020-09-07:
@mdblack98 I mean that they are in incorrectly set now, can you take a quick look

mdblack98, 2020-09-07:
> @mdblack98 I mean that they are incorrectly set now, can you take a quick look

Should be fixed now....please check.
Mike W9MDB

mikaelnousiainen, 2020-09-07:
@mdblack98 All good otherwise, but the highest band on IC-7300 is 4m. (so 2m + 70cm should be removed)

mikaelnousiainen, 2020-09-07:
@mdblack98 Also, couldn't we use RIG_PARM_BACKLIGHT for setting backlight brightness? Other rigs support it already. I think backlight settings should anyway be in the "parameters" category... There should be no need for a new level, right?

mdblack98, 2020-09-07:
I reviewed them with the specs so hopefully correct now.

mikaelnousiainen, 2020-09-07:
@mdblack98 Thanks, now it looks correct.

Still restating my previous comment: Couldn't we use RIG_PARM_BACKLIGHT for setting backlight brightness? Other rigs support it already. I think backlight settings should anyway be in the "parameters" category... There should be no need for a new level, right?