Hamlib/Hamlib #367: binding for perl, python and tcl fails in hamlib-4.0_rc1

dl1jbe, 2020-09-07:
bindings/hamlib.swg refers in line 250 to a file amplifier.swg which is not in the tar ball.

Digging deeper it looks as if the bindings/hamlib.swg file is not the same as in github repo.

mdblack98, 2020-09-07:
It was missing from Makefile.am but I think Nate may have to change his distribution script.
Mike W9MDB

mdblack98, 2020-09-07:
Which tarball did you download?
Mike W9MDB

N0NB, 2020-09-07:
If it was missing from the Makefile.am then it will be included when `make dist` is invoked.

closed 2020-09-07 by mdblack98

dl1jbe, 2020-09-07:
> Which tarball did you download? Mike W9MDB On Monday, September 7, 2020, 09:52:49 AM CDT, 

hamlib-4.0_rc1.tar.gz from sourceforge.net

N0NB, 2020-09-07:
Okay, I have cherry-picked all of Mike's commits up to this one into the Hamlib-4.0 branch.  Tom, can you pull from master and verify the fix?  If it's working I'll do an rc2.

dl1jbe, 2020-09-07:
Give some minutes. I will try.

dl1jbe, 2020-09-07:
Looks good now. I have cloned Hamlib/hamlib.git from github. Perl and Python bindings got build (the others I did not check atm).

N0NB, 2020-09-07:
Ok, I can quickly work up an rc2 and upload them.  Thanks, Tom.

dl1jbe, 2020-09-07:
Your welcome. Glad to help. Btw, first checks with tlf are looking good.

N0NB, 2020-09-07:
I've been using the master branch of Hamlib with the master branch of Tlf for some time.

Thanks, Tom.

dl1jbe, 2020-09-07:
Did just a full build from the rc2 tar ball on sourceforge. Works as expected.

N0NB, 2020-09-07:
* On 2020 07 Sep 13:03 -0500, Thomas Beierlein wrote:
> Did just a full build from the rc2 tar ball on sourceforge. Works as expected. 
> 

Thumbs up!

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819