Hamlib/Hamlib #371: Yaesu GS-232B protocol backend with K3NG rotator firmware

mikaelnousiainen, 2020-09-09:
This is mainly a question about the protocol implementation: the bug can be either in Hamlib or K3NG firmware.

I'm trying to use K3NG rotator firmware (https://github.com/k3ng/k3ng_rotator_controller) with GS-232B emulation (with latest Hamlib master), but I'm getting an error with `get_pos` right after doing `set_pos`.

This is because the K3NG firmware prints out an empty line feed after a `W` command that sets the rotator position. This confuses the following `get_pos` command even though the command is executed fine and there is no error response.

See: https://github.com/k3ng/k3ng_rotator_controller/blob/master/k3ng_rotator_controller/k3ng_rotator_controller.ino#L3605

Should these empty lines allowed by Hamlib -- and by the protocol? Is there some official documentation about the allowed responses?

Some Hamlib debug log:
```
gs232b_rot_get_position called
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.         
    
read_string called
read_string(): RX 13 characters
0000    41 5a 3d 33 32 32 45 4c 3d 30 30 30 0d              AZ=322EL=000
.   
gs232b_rot_get_position: (az, el) = (322, 0)
rot_get_position: got az=322.00, el=0.00
rotctl(d): P '319.76' '0.00' '' ''
rot_set_position called az=319.76 el=0.00
rot_set_position: south_zero=0 
gs232b_rot_set_position called: az=319.76 el=0.00
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 9 bytes
0000    57 33 32 30 20 30 30 30 0d                          W320 000.       
write_block called
write_block(): TX 1 bytes
0000    0d                                                  .               
rotctl(d): p '' '' '' ''
rot_get_position called
gs232b_rot_get_position called
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    43 32 0d                                            C2.             
read_string called
read_string(): RX 1 characters
0000    0a                                                  .               <- this is where it fails
```

mdblack98, 2020-09-09:
At this point I'm going to call the this a K3NG firmware bug.
The serial port has been flushed right before the C2 command is sent so should be seeing the current position.  The 0x0a response is bogus at that point.  Maybe it can't return the position while moving?
The GS232B is being used by quite a few other rotators so K3NG is a one-off from Yaesu's protocol.

mikaelnousiainen, 2020-09-13:
I'm actually running `\get_pos` queries in a tight loop (for polling) and doing `\set_pos` in between, so flushing may not happen early enough there. What I have noticed that if I simply type these commands (without polling quickly every 100ms), I don't get any errors, which would be an indication of serial port flushing working correctly. These commands do get executed sequentially in the same thread in my application, so I still wonder why the flushing doesn't work correctly in this case.

I'll have to check if I can find detatiled GS-232B protocol docs and also ask K3NG firmware developers about the empty lines... They don't make much sense to me if there response lines are empty!

mikaelnousiainen, 2020-09-13:
I'm seeing this being stated about the GS-232B and GS-232A protocols:

"Most commands have two versions: one for azimuth, and one for elevation. Commands are not echoed by the Control Interface, but a carriage return character (“0Dh”) is returned after every command, and also a line feed character (“0Ah”) if the command invoked returned data. Invalid commands cause “? >” to be returned and the input buffer cleared."

So basically what should be returned by every command is a carriage return (`0x0D`) and not a line-feed (`0x0A`), unless there is a response from the command.

This means that Hamlib should probably be accepting at least a carriage return(`0x0D`) as a response, but maybe also simply ignore any line feeds (`0x0A`) in order to be extra compatible with all emulations too? I can make the change if it sounds acceptable...

See these links for GS-232A and GS-232B references:

* https://www.yaesu.com/downloadFile.cfm?FileID=820&FileCatID=155&FileName=GS232A.pdf&FileContentType=application%2Fpdf
https://www.inradio.pl/images/stories/virtuemart/product/Yaesu_GS-232B_instrukcjaobslugi_manual.pdf
* https://github.com/k3ng/k3ng_rotator_controller/issues/76

mdblack98, 2020-09-13:
That println in k3ng does look like the culprit.
Let's see if they fix it.
Those rotators (other than k3ng) are used by quite a few people without any problems.
Your short term fix would be to do a short sleep on those commands that don't expect a response so the flush can catch the extra data.

mikaelnousiainen, 2020-12-01:
@mdblack98 It's been quite on K3NG firmware side. That said, I still think that we could make a change in the GS-232 protocols so that parsing would simply skip over any extra `0x0A` and `0x0D` characters, as they are used as command/response separators anyway. That would simply make Hamlib more compatible with all the GS-232 emulation out there. What do you think?

mdblack98, 2020-12-01:
Try a pull and see if the patch I just did fixes it.
Hopefully works will all gs232a/b rotors and emulators if they follow the 0x0d and 0x0a paradigms.

Mike W9MDB

mikaelnousiainen, 2020-12-01:
Wow, that was fast! Thanks, I'll test it right away.

mikaelnousiainen, 2020-12-01:
@mdblack98 It didn't fix the issue unfortunately.

```
write_block(): TX 9 bytes\n
0000    57 33 30 34 20 30 30 30 0d                          W304 000.       \n
write_block called\n
write_block(): TX 1 bytes\n
0000    0d                                                  .               \n
rotctl_parse: NETROTCTL_RET 0\n
rotctl(d): p '' '' '' ''\n
rot_get_position called\n
gs232b_rot_get_position called\n
rig_flush: called for serial device\n
serial_flush called\n
tcflush\n
write_block called\n
write_block(): TX 3 bytes\n
0000    43 32 0d                                            C2.             \n
read_string called, rxmax=32\n
read_string(): RX 2 characters\n
0000    0d 0a                                               ..              \n <-- That's where the error happens now and it's repeateable. Now we get both 0x0d and 0x0a.
rotctl_parse: NETROTCTL_RET -8\n

rotctl(d): p '' '' '' ''\n
rot_get_position called\n
gs232b_rot_get_position called\n
rig_flush: called for serial device\n
serial_flush called\n
tcflush\n
write_block called\n
write_block(): TX 3 bytes\n
0000    43 32 0d                                            C2.             \n
read_string called, rxmax=32\n
read_string(): RX 14 characters\n
0000    41 5a 3d 33 32 33 45 4c 3d 30 30 30 0d 0a           AZ=323EL=000..  \n
gs232b_rot_get_position: (az, el) = (323, 0)\n
rot_get_position: got az=323.00, el=0.00\n
```

mikaelnousiainen, 2020-12-01:
I wonder if there's something weird still going on in the K3NG firmware regarding how it handles fast polling. Hamlib should anyway be using a mutex for the commands and so is my application too (to have predictable parsing from the TCP connection to rotctld) -- so I'm not trying to execute commands simultaneously -- and even that should be handle correctly by Hamlib.

mdblack98, 2020-12-01:
A mutex?  There's only one process doing anything here...mutex not needed as there is no multi-threading going on.

mdblack98, 2020-12-01:
Please add -Z switch and let's try --set-conf=post_write_delay=100
Maybe bump up the 100ms to 200ms, 400ms and see if changes anything.
Then let's bang out the position requests
rotctl -m 603 -r /dev/ttyUSB0 -s 4800 -vvvvv -Z --set-conf=post_write_delay=100 p p p p p
See if any of those 5 p's work...they should ALL work.
So play with the post_write_delay
Mike

mikaelnousiainen, 2020-12-01:
@mdblack98 Oh, right, I wasn't aware that there's a global post_write_delay setting in Hamlib :dagger: 

The problem is actually when *setting* the position using set_pos command, which seems to always return extra line feeds and/or carriage returns, so when querying get_pos immediately it will read those extra characters. Setting post_write_delay to something over 25ms seems to fix this, so we have a workaround here. Thanks!

Here's a debug log with post_write_delay=25:
```
2020-12-01:22:45:05.114699: rot_get_position called
2020-12-01:22:45:05.114755: gs232b_rot_get_position called
2020-12-01:22:45:05.114802: rig_flush: called for serial device
2020-12-01:22:45:05.114860: serial_flush called
2020-12-01:22:45:05.114912: tcflush
2020-12-01:22:45:05.114996: write_block called
2020-12-01:22:45:05.140894: write_block(): TX 3 bytes
2020-12-01:22:45:05.141044: 0000    43 32 0d                                            C2.             
2020-12-01:22:45:05.141182: read_string called, rxmax=32
2020-12-01:22:45:05.145784: read_string(): RX 14 characters
2020-12-01:22:45:05.145913: 0000    41 5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a           AZ=330EL=000..  
2020-12-01:22:45:05.146063: gs232b_rot_get_position: (az, el) = (330, 0)
2020-12-01:22:45:05.146148: rot_get_position: got az=330.00, el=0.00
330.00
0.00
2020-12-01:22:45:05.146346: rot_set_position called az=330.00 el=0.00
2020-12-01:22:45:05.146511: rot_set_position: south_zero=0 
2020-12-01:22:45:05.146575: gs232b_rot_set_position called: az=330.00 el=0.00
2020-12-01:22:45:05.146819: rig_flush: called for serial device
2020-12-01:22:45:05.146886: serial_flush called
2020-12-01:22:45:05.146935: tcflush
2020-12-01:22:45:05.147007: write_block called
2020-12-01:22:45:05.172223: write_block(): TX 9 bytes
2020-12-01:22:45:05.172390: 0000    57 33 33 30 20 30 30 30 0d                          W330 000.       
2020-12-01:22:45:05.172497: write_block called
2020-12-01:22:45:05.197804: write_block(): TX 1 bytes
2020-12-01:22:45:05.198022: 0000    0d                                                  .               
2020-12-01:22:45:05.198153: rot_get_position called
2020-12-01:22:45:05.198226: gs232b_rot_get_position called
2020-12-01:22:45:05.198300: rig_flush: called for serial device
2020-12-01:22:45:05.198356: serial_flush called
2020-12-01:22:45:05.198497: tcflush
2020-12-01:22:45:05.198579: write_block called
2020-12-01:22:45:05.223817: write_block(): TX 3 bytes
2020-12-01:22:45:05.223937: 0000    43 32 0d                                            C2.             
2020-12-01:22:45:05.224013: read_string called, rxmax=32
2020-12-01:22:45:05.227052: read_string(): RX 14 characters
2020-12-01:22:45:05.227251: 0000    41 5a 3d 33 32 39 45 4c 3d 30 30 30 0d 0a           AZ=329EL=000..  
2020-12-01:22:45:05.227423: gs232b_rot_get_position: (az, el) = (329, 0)
2020-12-01:22:45:05.227680: rot_get_position: got az=329.00, el=0.00
329.00
0.00
```

mdblack98, 2020-12-01:
OK...I put two fixes in...one is the 25ms delay as default and one is a retry if \n\r is returned.
Give it a try.
Mike

mikaelnousiainen, 2020-12-02:
@mdblack98 Thanks. The delay definitely helps, but I think it may need to be a bit larger, 30 ms seems to be very stable.

However, I tested the retry mechanism by setting post_write_delay back to zero, so that I could trigger the error. Now I'm getting even more combinations of 0x0a, 0x0d and the `>` character. Additionally, sometimes Hamlib seems to miss parts of the actual response data -- probably because of serial port flush accompanied with retry?

Anyway, based on testing, we probably should be retrying in case the response has _any number and combination_ of these characters only:

* `>`
* 0x0a
* 0x0d

I think that according to GS-232 protocol, any response that contains only these characters will be invalid anyway, so this should be safe.

Test results:

```
rotctl -m 603 -r /dev/... -s 9600 -vvvvv -Z --set-conf=post_write_delay=0 p p p p P 330 0 p p p
```

```
2020-12-02:07:36:32.588598: rot_get_position called
2020-12-02:07:36:32.588656: gs232b_rot_get_position called
2020-12-02:07:36:32.588714: rig_flush: called for serial device
2020-12-02:07:36:32.588757: serial_flush called
2020-12-02:07:36:32.588808: tcflush
2020-12-02:07:36:32.588882: write_block called
2020-12-02:07:36:32.588970: write_block(): TX 3 bytes
2020-12-02:07:36:32.589028: 0000    43 32 0d                                            C2.             
2020-12-02:07:36:32.589085: read_string called, rxmax=32
2020-12-02:07:36:32.620129: read_string(): RX 14 characters
2020-12-02:07:36:32.620263: 0000    41 5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a           AZ=330EL=000..  
2020-12-02:07:36:32.620466: gs232b_rot_get_position: (az, el) = (330, 0)
2020-12-02:07:36:32.620660: rot_get_position: got az=330.00, el=0.00
330.00
0.00
2020-12-02:07:36:32.621041: rot_set_position called az=330.00 el=0.00
2020-12-02:07:36:32.621137: rot_set_position: south_zero=0 
2020-12-02:07:36:32.621210: gs232b_rot_set_position called: az=330.00 el=0.00
2020-12-02:07:36:32.621432: rig_flush: called for serial device
2020-12-02:07:36:32.621505: serial_flush called
2020-12-02:07:36:32.621554: tcflush
2020-12-02:07:36:32.621625: write_block called
2020-12-02:07:36:32.621819: write_block(): TX 9 bytes
2020-12-02:07:36:32.621967: 0000    57 33 33 30 20 30 30 30 0d                          W330 000.       
2020-12-02:07:36:32.622058: write_block called
2020-12-02:07:36:32.622229: write_block(): TX 1 bytes
2020-12-02:07:36:32.622300: 0000    0d                                                  .               
2020-12-02:07:36:32.622447: rot_get_position called
2020-12-02:07:36:32.622520: gs232b_rot_get_position called
2020-12-02:07:36:32.622580: rig_flush: called for serial device
2020-12-02:07:36:32.622654: serial_flush called
2020-12-02:07:36:32.622755: tcflush
2020-12-02:07:36:32.622899: write_block called
2020-12-02:07:36:32.623009: write_block(): TX 3 bytes
2020-12-02:07:36:32.623078: 0000    43 32 0d                                            C2.             
2020-12-02:07:36:32.623123: read_string called, rxmax=32
2020-12-02:07:36:32.636214: read_string(): RX 2 characters
2020-12-02:07:36:32.636459: 0000    0d 0a                                               ..              
2020-12-02:07:36:32.636538: rig_flush: called for serial device
2020-12-02:07:36:32.636608: serial_flush called
2020-12-02:07:36:32.636803: tcflush
2020-12-02:07:36:32.636899: write_block called
2020-12-02:07:36:32.637079: write_block(): TX 3 bytes
2020-12-02:07:36:32.637142: 0000    43 32 0d                                            C2.             
2020-12-02:07:36:32.637248: read_string called, rxmax=32
2020-12-02:07:36:32.652182: read_string(): RX 3 characters
2020-12-02:07:36:32.652304: 0000    3e 0d 0a                                            >..             
get_pos: error = Protocol error
2020-12-02:07:36:32.652440: rot_get_position called
2020-12-02:07:36:32.652495: gs232b_rot_get_position called
2020-12-02:07:36:32.652549: rig_flush: called for serial device
[... 15 lines cut ...]
2020-12-02:07:36:32.684266: read_string(): RX 14 characters
2020-12-02:07:36:32.684409: 0000    41 5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a           AZ=330EL=000..  
2020-12-02:07:36:32.684554: gs232b_rot_get_position: (az, el) = (330, 0)
2020-12-02:07:36:32.684663: rot_get_position: got az=330.00, el=0.00
330.00
0.00
2020-12-02:07:36:32.684869: rot_get_position called
2020-12-02:07:36:32.684956: gs232b_rot_get_position called
2020-12-02:07:36:32.685077: rig_flush: called for serial device
2020-12-02:07:36:32.685142: serial_flush called
2020-12-02:07:36:32.685191: tcflush
2020-12-02:07:36:32.685258: write_block called
2020-12-02:07:36:32.685359: write_block(): TX 3 bytes
2020-12-02:07:36:32.685422: 0000    43 32 0d                                            C2.             
2020-12-02:07:36:32.685488: read_string called, rxmax=32
2020-12-02:07:36:32.700236: read_string(): RX 13 characters
2020-12-02:07:36:32.700416: 0000    5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a              Z=330EL=000..   
2020-12-02:07:36:32.700496: gs232b_rot_get_position: wrong reply 'Z=330EL=000
', expected AZ=xxx EL=xxx
get_pos: error = Protocol error
2020-12-02:07:36:32.700592: rot_close called
2020-12-02:07:36:32.700667: ser_close called
2020-12-02:07:36:32.700780: ser_close: restoring options
2020-12-02:07:36:32.706168: rot_cleanup called
```

mdblack98, 2020-12-02:
Hmm....the first C2 call is correct...note the response characters...2020-12-02:07:36:32.589028: 0000    43 32 0d                                            C2.             
2020-12-02:07:36:32.589085: read_string called, rxmax=32
2020-12-02:07:36:32.620129: read_string(): RX 14 characters
2020-12-02:07:36:32.620263: 0000    41 5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a           AZ=330EL=000..  
Then it's gets a nonsense response that includes ">" which is part of a ?> response indicating an error2020-12-02:07:36:32.637142: 0000    43 32 0d                                            C2.             
2020-12-02:07:36:32.637248: read_string called, rxmax=32
2020-12-02:07:36:32.652182: read_string(): RX 3 characters
2020-12-02:07:36:32.652304: 0000    3e 0d 0a                                            >..             
get_pos: error = Protocol errorThe flush is meant to get rid of anything still outstanding from previous commands and works on all rigs unless they queue their own serial data (like MicroHam).
Please see if the --set-conf=flushx=1 switch helps and send me the debug...it will show what gets flushed and drains the device using another method\
If 25ms was unstable and 30ms stable I normally do a 2X factor of the unstable to make sure we're covered so I've changed the post_write_delay default to 50ms. 
Also added a retry when a ">" shows up which indicates an error on the rotor protocol

Mike

mikaelnousiainen, 2020-12-02:
The flushx conf parameter doesn't seem to be supported on rotctl (but only on rigctl).

However, the post_write_delay seems to be enough to have a realiable workaround for this issue. The K3NG firmware is indeed the reason and doesn't tolerate fast reads after writes -- and it seems to write extra terminator chars, but all of this is solved by the delay/flush combination.

I think it is safe to close this issue for now, as the post_write_delay of 50ms will be used now by default, requiring no additional configuration to use wthe workaround.

I ran some additional tests with delay disabled and sometimes the retries conflict with the responses this way -- but it never happens with the post_write_delay.

```
rotctl -m 603 -r /dev/... -s 9600 -vvvvv -Z --set-conf=post_write_delay=0 p p p p P 330 0 p p p
```

```
2020-12-02:19:12:08.320473: 0000    43 32 0d                                            C2.
2020-12-02:19:12:08.320524: read_string called, rxmax=32
2020-12-02:19:12:08.351586: read_string(): RX 14 characters
2020-12-02:19:12:08.351753: 0000    41 5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a           AZ=330EL=000..
2020-12-02:19:12:08.351905: gs232b_rot_get_position: (az, el) = (330, 0)
2020-12-02:19:12:08.352006: rot_get_position: got az=330.00, el=0.00
330.00
0.00
2020-12-02:19:12:08.352301: rot_set_position called az=330.00 el=0.00
2020-12-02:19:12:08.352429: rot_set_position: south_zero=0
2020-12-02:19:12:08.352937: gs232b_rot_set_position called: az=330.00 el=0.00
2020-12-02:19:12:08.353487: rig_flush: called for serial device
2020-12-02:19:12:08.354105: serial_flush called
2020-12-02:19:12:08.354199: tcflush
2020-12-02:19:12:08.354271: write_block called
2020-12-02:19:12:08.354462: write_block(): TX 9 bytes
2020-12-02:19:12:08.354571: 0000    57 33 33 30 20 30 30 30 0d                          W330 000.
2020-12-02:19:12:08.354644: write_block called
2020-12-02:19:12:08.354782: write_block(): TX 1 bytes
2020-12-02:19:12:08.354847: 0000    0d                                                  .               
2020-12-02:19:12:08.354949: rot_get_position called
2020-12-02:19:12:08.355030: gs232b_rot_get_position called
2020-12-02:19:12:08.355103: rig_flush: called for serial device
2020-12-02:19:12:08.355173: serial_flush called
2020-12-02:19:12:08.355226: tcflush
2020-12-02:19:12:08.355290: write_block called
2020-12-02:19:12:08.355384: write_block(): TX 3 bytes
2020-12-02:19:12:08.355439: 0000    43 32 0d                                            C2.             
2020-12-02:19:12:08.355486: read_string called, rxmax=32
2020-12-02:19:12:08.367548: read_string(): RX 2 characters
2020-12-02:19:12:08.367680: 0000    0d 0a                                               ..              
2020-12-02:19:12:08.367747: rig_flush: called for serial device
2020-12-02:19:12:08.367823: serial_flush called
2020-12-02:19:12:08.367881: tcflush
2020-12-02:19:12:08.367956: write_block called
2020-12-02:19:12:08.368052: write_block(): TX 3 bytes
2020-12-02:19:12:08.368113: 0000    43 32 0d                                            C2.             
2020-12-02:19:12:08.368155: read_string called, rxmax=32
2020-12-02:19:12:08.383599: read_string(): RX 4 characters
2020-12-02:19:12:08.383726: 0000    3f 3e 0d 0a                                         ?>..
': '?>2-02:19:12:08.383798: gs232b_transaction: Error for 'C2
'
get_pos: error = Protocol error
```

mdblack98, 2020-12-02:
It's like it's swallowing the C and or 2 of the C2 command and giving the >? error response.
I'm going to add the flushx option to rotctl this afternoon and would like you to test again.  That should show more info.I have to be out for a couple hours...
Mike

mikaelnousiainen, 2020-12-03:
Thanks. I also uncommented the printf-loop to actually print out the bytes when flushx is enabled.

Here's one run:

```
rotctl -m 603 -r /dev/... -s 9600 -vvvvv -Z --set-conf=flushx=1 --set-conf=post_write_delay=0 p p P 330 0 p p P 330 0 p p P 330 0 p p >gs232.log 2>&1
```

```
2020-12-03:19:59:03.198084: rotctl Hamlib 4.1~git
Last commit was Thu Dec 03 05:47:30 2020 +0000 SHA=c4e1de
2020-12-03:19:59:03.198776: Report bugs to <hamlib-developer@lists.sourceforge.net>

2020-12-03:19:59:03.198969: rot_init called
2020-12-03:19:59:03.199365: initrots4_gs232a called
2020-12-03:19:59:03.199562: rot_register (601)
2020-12-03:19:59:03.199714: rot_register (602)
2020-12-03:19:59:03.199795: rot_register (603)
2020-12-03:19:59:03.199858: rot_register (604)
2020-12-03:19:59:03.199915: rot_register (605)
2020-12-03:19:59:03.199974: rot_register (606)
2020-12-03:19:59:03.200033: rot_register (607)
2020-12-03:19:59:03.200089: rot_register (608)
2020-12-03:19:59:03.200309: set_conf: called
2020-12-03:19:59:03.200393: set_conf: token=flushx, val=1
2020-12-03:19:59:03.200480: rot_token_lookup called lookup flushx
2020-12-03:19:59:03.200569: rot_set_conf called
2020-12-03:19:59:03.200629: rot_set_conf: flushx='1'
2020-12-03:19:59:03.200684: frontrot_set_conf called
2020-12-03:19:59:03.200739: set_conf: token=post_write_delay, val=0
2020-12-03:19:59:03.200789: rot_token_lookup called lookup post_write_delay
2020-12-03:19:59:03.200838: rot_set_conf called
2020-12-03:19:59:03.200888: rot_set_conf: post_write_delay='0'
2020-12-03:19:59:03.200939: frontrot_set_conf called
2020-12-03:19:59:03.201020: rot_open called
2020-12-03:19:59:03.201084: serial_open called
2020-12-03:19:59:03.210447: serial_setup called
2020-12-03:19:59:03.210619: serial_setup: tcgetattr
2020-12-03:19:59:03.210714: serial_setup: cfmakeraw
2020-12-03:19:59:03.210784: serial_setup: cfsetispeed=9600,0x000d
2020-12-03:19:59:03.210849: serial_setup: cfsetospeed=9600,0x000d
2020-12-03:19:59:03.210913: serial_setup: data_bits=8
2020-12-03:19:59:03.210963: serial_setup: parity=0
2020-12-03:19:59:03.211036: serial_setup: tcsetattr TCSANOW
2020-12-03:19:59:03.212358: serial_flush called
2020-12-03:19:59:03.212473: serial_flush: flushing
2020-12-03:19:59:03.212575: read flushed 0 bytes
2020-12-03:19:59:03.263037: Backend version: 20201202.0, Status: Stable
2020-12-03:19:59:03.263208: rot_get_position called
2020-12-03:19:59:03.263266: gs232b_rot_get_position called
2020-12-03:19:59:03.263379: rig_flush: called for serial device
2020-12-03:19:59:03.263476: serial_flush called
2020-12-03:19:59:03.263539: serial_flush: flushing
2020-12-03:19:59:03.263601: read flushed 0 bytes
2020-12-03:19:59:03.263682: write_block called
2020-12-03:19:59:03.263796: write_block(): TX 3 bytes
2020-12-03:19:59:03.263921: 0000    43 32 0d                                            C2.             
2020-12-03:19:59:03.263992: read_string called, rxmax=32
2020-12-03:19:59:03.316485: read_string(): RX 14 characters
2020-12-03:19:59:03.316685: 0000    41 5a 3d 33 33 30 45 4c 3d 30 30 30 0d 0a           AZ=330EL=000..  
2020-12-03:19:59:03.316844: gs232b_rot_get_position: (az, el) = (330, 0)
2020-12-03:19:59:03.316932: rot_get_position: got az=330.00, el=0.00
Opened rot model 603, 'GS-232B'
330.00
0.00
2020-12-03:19:59:03.317037: rot_get_position called
2020-12-03:19:59:03.317096: gs232b_rot_get_position called
2020-12-03:19:59:03.317161: rig_flush: called for serial device
2020-12-03:19:59:03.317223: serial_flush called
2020-12-03:19:59:03.317280: serial_flush: flushing
[... 166 lines cut ...]
2020-12-03:19:59:03.478756: read_string called, rxmax=32
2020-12-03:19:59:03.492579: read_string(): RX 14 characters
2020-12-03:19:59:03.492777: 0000    41 5a 3d 33 33 31 45 4c 3d 30 30 30 0d 0a           AZ=331EL=000..  
2020-12-03:19:59:03.492900: gs232b_rot_get_position: (az, el) = (331, 0)
2020-12-03:19:59:03.492984: rot_get_position: got az=331.00, el=0.00
331.00
0.00
2020-12-03:19:59:03.493096: rot_get_position called
2020-12-03:19:59:03.493151: gs232b_rot_get_position called
2020-12-03:19:59:03.493210: rig_flush: called for serial device
2020-12-03:19:59:03.493264: serial_flush called
2020-12-03:19:59:03.493314: serial_flush: flushing
2020-12-03:19:59:03.493388: read flushed 1 bytes
2020-12-03:19:59:03.493444: write_block called
2020-12-03:19:59:03.493534: write_block(): TX 3 bytes
2020-12-03:19:59:03.493597: 0000    43 32 0d                                            C2.             
2020-12-03:19:59:03.493647: read_string called, rxmax=32
2020-12-03:19:59:03.508459: read_string(): RX 1 characters
2020-12-03:19:59:03.508652: 0000    0a                                                  .               
0x0d(~) get_pos: error = Protocol error
2020-12-03:19:59:03.508767: rot_close called
2020-12-03:19:59:03.508982: ser_close called
2020-12-03:19:59:03.509064: ser_close: restoring options
2020-12-03:19:59:03.514505: rot_cleanup called
```

mikaelnousiainen, 2020-12-03:
Anyway, I feel like it's not necessary to debug this much further -- the way K3NG firmware handles serial reads/writes is probably fragile and breaks easily, but since it can be worked around with the delay, it's enough for me at least for now.

closed 2020-12-03 by mikaelnousiainen

leadpima, 2021-01-29:
Hello folks,
I've been pulling off the hairs of my head trying to get something from the GS-232B and I need your help @mikaelnousiainen @mdblack98.

I am using a serial terminal software called Termite to communicate with the GS-232B. After I do all the connections and wait for 15 seconds, I receive the message "CONNECT OK" from the Gs-232B as mentioned in its manual. However, when I want to start calibration by sending the O command, there's no response from the GS232b. If I send one O command, press the enter key and then send O again as shown below:
O 
O
I receive the ?> which according to the manual means it's an invalid command. 
I would appreciate if you could help me with this. :)

mdblack98, 2021-01-29:
Should be a CTRL-M after the O I think.
Maybe your terminal is sending line feed instead?
Mike

leadpima, 2021-01-30:
Hello @mdblack98,
Thanks for replying. I took a look to what the terminal software is sending. The picture below shows the data being sent and received and you are right: the terminal software is sending line feed "0a". How could I solve this issue?
[image]

leadpima, 2021-01-30:
@mikaelnousiainen I just changed my terminal software and you were absolutely right: The new terminal software allows me to send the carriage return and now I'm communicating with the GS232b properly!

mdblack98, 2021-01-30:
Press CTRL-M instead of CR
Mike W9MDB