Hamlib/Hamlib #372: TY returns TYK from Kenwood TS-590SG

bhorton, 2020-09-09:
I've seen some references to a patch but I'm getting the following error on my Mac using rigctl v3.3:

kenwood_transaction called
kenwood_transaction: cmdstr = TY
serial_flush called
write_block called
write_block(): TX 3 bytes
0000    54 59 3b                                            TY;
read_string called
read_string(): RX 7 characters
0000    54 59 4b 20 30 30 3b                                TYK 00;
kenwood_safe_transaction: wrong answer; len for cmd TY: expected = 5, got 6
client lock disengaged

bhorton, 2020-09-09:
Actually, control works fine when I'm using wsjtx and I believe they're using hamlib. However, I cannot get rigctl to work directly and this seems to be the issue. I've tried to emulate settings like this:

rigctld -m 237 -r /dev/cu.SLAB_USBtoUART -C serial_handshake=None,rts_state=ON,data_bits=8,stop_bits=1,dtr_state=Unset -s 115200 -vvvvvv

mdblack98, 2020-09-09:
That's because WSJTX is using hamlib 4.0 where it's been fixed.  No patches are being applied to 3.3.
Download hamlib-4.0 from here and it should work fine.
https://sourceforge.net/projects/hamlib/

closed 2020-09-09 by mdblack98

bhorton, 2020-09-09:
That did it. Thank you!