Hamlib/Hamlib #373: IC-7300 some levels wrong

mdblack98, 2020-09-09:
IC-7300 get/set level enumeration

Get level: 
PREAMP(0..0/0)		ok 
ATT(0..0/0)		ok 
VOXDELAY(0..20/1) 	1a 05 00 81 (LCD Background brightness) s/b 1a 05 01 91
AF(0..0/0)		ok 
RF(0..0/0)		1a 05 00 23 (Confirmation Beep) s/b 14 02
SQL(0..0/0) 		1a 05 00 95 (Send/read time) s/b 14 03
NR(0..0/0)		ok 
PBT_IN(0..0/0)		ok
PBT_OUT(0..0/0)		ok 
CWPITCH(300..900/1)	ok
RFPOWER(0..0/0)		ok 
MICGAIN(0..0/0)		ok		 
KEYSPD(6..48/1)		ok		 
COMP(0..0/0)		ok		 
AGC(0..0/0)		ok		 
BKINDL(0..0/0)		ok
VOXGAIN(0..0/0)		ok
ANTIVOX(0..0/0)		ok		 
RAWSTR(0..255/0)	ok (Read S-meter)	 
SWR(0..0/0) ALC(0..0/0)	ok 
STRENGTH(0..0/0)	ok 
RFPOWER_METER(0..0/0)	ok	 
COMP_METER(0..0/0)	ok	 
VD_METER(0..0/0)	ok 
ID_METER(0..0/0)	ok 
NOTCHF_RAW(0..0/0)	ok	 
MONITOR_GAIN(0..0/0)	ok	 
NB(0..0/0)		ok

Extra Get levels:
ANN			Invalid parameter s/b 1a 05 00 91 ???
BACKLIGHT		Invalid parameter s/b 1a 05 00 81
BEEP			Invalid parameter s/b 1a 05 00 23
TIME			Imvalid parameter s/b 1a 05 00 95

Set Level:
VOXDELAY		see above s/b 1a 05 01 91 nn (00 to 20)
RF			see above s/b 14 02 nn mm (0000 to 0255)
SQL			see above s/b 14 03 nn mm
ANN			see above s/b 1a 05 00 91 tt tt tt tt tt tt tt tt tt tt (up to 10 chars)
BACKLIGHT		see above s/b 1a 05 00 81 nn mm (0000 to 0255)
BEEP			see above s/b 1a 05 00 23 00/01
TIME			see above s/b 1a 05 00 95 hh mm

mdblack98, 2020-09-10:
icom.c extlevels is colliding PARM and LEVEL values.
Have to separate them.

mdblack98, 2020-09-14:
P TIME is broken

p ANN is broken (seg fault)

P ANN allows one param to be entered

L RF is sending the wrong command.

All else OK.

 

See below for details.

 

P TIME 68700 is broken

 

p TIME

rigctl_parse: cmd=p(70)

rigctl_parse: cmd==0x70

rigctl_parse: vfo_opt=0

rigctl_parse: debug1

rigctl_parse: debug3

rigctl_parse: debug4 c=20

rigctl_parse: debug5

rigctl_parse: debug10

rigctl_parse: vfo_opt=0

rig_parse_parm called

rig_has_get_parm called

rig_get_parm called

rig_has_get_parm called

icom_get_parm called

icom_get_parm: i=0

icom_get_parm: i=1

icom_get_parm: i=2

icom_get_cmd called

icom_get_raw_buf called

rig_flush: called for serial device

serial_flush called

serial_flush: tcflush

write_block called

write_block(): TX 9 bytes

[... 1232 lines cut ...]
rigctl_parse: debug3

rigctl_parse: debug4 c=20

rigctl_parse: debug5

rigctl_parse: debug10

rigctl_parse: vfo_opt=0

rig_parse_level called

rig_ext_lookup called

rigctl_parse: vfo_opt=0

get_level: error = Invalid parameter

rigctl_parse: retcode=-1

rigctl_parse: called, interactive=1

 

Rig command: rigctl_parse: cmd==0x0a

mikaelnousiainen, 2020-09-14:
@mdblack98 Thanks for working on fixing these.

I've noticed some issues as well, even after applying your latest fixes regarding the collisions:

* RF GAIN level (get or set) doesn't work on any Icom rig I've tested (7300, 7600, 756Pro3)
* Squelch level (get or set) doesn't work on any Icom rig I've tested (7300, 7600, 756Pro3)
* DRIVE GAIN level (get and set) is broken on 7600 -- this is a custom level, but nevertheless it was working (I added it last year)

All of these used to work late last year.

mikaelnousiainen, 2020-09-14:
@mdblack98 One more thing on IC-7300: it doesn't support setting or getting repeater offset -- something probably got mixed in the caps, as 3.3 had these settings set to NULL (which is correct):

```
.set_rptr_shift =  NULL,
.get_rptr_shift =  NULL,
.set_rptr_offs =  NULL,
.get_rptr_offs =  NULL,
```

And yes, this is true for only 7300, as 9700 and 705 should support them for repeater access.

mdblack98, 2020-09-14:
RF and SQL should be fixed.
Can you test DRIVE GAIN?
Mike

mikaelnousiainen, 2020-09-15:
@mdblack98 RF and SQL work now, but DRIVE GAIN is still failing (both get and set):

```
Hamlib (:20001) command '+\set_level drive_gain 105' failed: Invalid parameter (error code -1)
```

mikaelnousiainen, 2020-09-15:
@mdblack98 I suspect there is still something wrong with `icom_set|get_ext_level` or `icom_set|get_ext_cmd` regarding matching of the tokens... Couldn't immediately see what yet, though...

mdblack98, 2020-09-15:
Please send debug info.
Mike W9MDB

mikaelnousiainen, 2020-09-15:
@mdblack98  `icom_set_ext_cmd` and `icom_get_ext_cmd` cannot find `drive_gain` in IC-7600 extcmds -- which is true. So just confirming: the new extcmds list should have a command for the existing extlevels list?

```
Rig command: \set_level drive_gain 130
rigctl_parse: input_line: \set_level drive_gain 130
rigctl_parse: vfo_opt=0
rig_parse_level called
rig_ext_lookup called
rig_set_ext_level called
icom_set_ext_level called
icom_set_ext_cmd called
rigctl_parse: vfo_opt=0
set_level: error = Invalid parameter
rigctl_parse: retcode=-1
rigctl_parse: called, interactive=1

Rig command: \get_level drive_gain
rigctl_parse: input_line: \get_level drive_gain
rigctl_parse: vfo_opt=0
rig_parse_level called
rig_ext_lookup called
rig_get_ext_level called
icom_get_ext_level called
icom_get_ext_cmd called
rigctl_parse: vfo_opt=0
get_level: error = Invalid parameter
rigctl_parse: retcode=-1
rigctl_parse: called, interactive=1
```

```
Rig command: \get_level drive_gain
rigctl_parse: input_line: \get_level drive_gain
rigctl_parse: vfo_opt=0
rig_parse_level called
rig_ext_lookup called
rig_get_ext_level called
icom_get_ext_level called
icom_get_ext_cmd called
rigctl_parse: vfo_opt=0
get_level: error = Invalid parameter
rigctl_parse: retcode=-1
rigctl_parse: called, interactive=1
```

mikaelnousiainen, 2020-09-16:
@mdblack98 I can try to fix the extcmds implementation, there are some bugs there still.

One question though: now that it's possible to set the cmdparam type in the Icom extcmds, is there any need to have a **separate** list of tokens in `ext_tokens` in `icom_priv_caps`? It seems very redundant to me, as the support for an ext level token can be detected from the extcmds list too.

I think I'll end up removing `ext_tokens` as it's not even widely used anywhere.

mikaelnousiainen, 2020-09-16:
@mdblack98 Ok, I get now why `ext_tokens` was added. Some of the rigs use the common list of extlevels / extcmds that is shared, so list then determines which ones are actually supported by the rig, right?

This gets a bit complicated. I'll see if there could be a way to simplify it a bit.

mikaelnousiainen, 2020-09-16:
@mdblack98 And it would be nice to be able to share some parts of the extcmds between the rigs and make some of them rig-specific -- just to deal with the Icom madness of always introducing rig-specific commands with every model.

mdblack98, 2020-09-16:
Please don't touch this section of code without a LOT of thought.  There are 3 levels of such things....it's just that there was a collision between LEVEL and PARM values going on that had to be fixed and I figured that change would break something...which it did.    There are global levels/parms, Icom levels/parms, and rig levels/parms.  
The RigPi testing is pointing out some of these things now as Howard is actively improving the interface.
If you can think of some way to simplify this we can discuss.

Mike W9MDB

mikaelnousiainen, 2020-09-16:
@mdblack98 Agreed, I won't be removing anything -- but only fixing issues. I've got a PR coming soon (once I get to test it) for you to review, which will hopefully fix a bunch of things with the extcmds handling! So no radical changes, only improvements.

mdblack98, 2020-09-16:
I was just finishing up fixing the drive_gain bug which would fix all the other ext_cmds.
Sounds like it might collide with what you're doing.
Mike

mikaelnousiainen, 2020-09-16:
@mdblack98 Possibly. I'll try to fix conflicts. There was an issue with `ext_tokens` not being defined and `icom_set_ext_cmd` searching for something else than tokens in the list. I'll push out a PR later today. I've also cleaned up Icom code to use more extcmds and some other minor changes.

closed 2020-09-16 by mdblack98