Hamlib/Hamlib #377: Improving release quality / Review process for commits?

mikaelnousiainen, 2020-09-14:
As an active user of Hamlib (and also having developed some features in it), I'm wondering if we could implement some process and/or automation to improve the release quality -- especially to cut the number of regressions between releases. I know this is a particularly difficult software project from the aspect of testing, because it deals with real hardware, which is in most cases not available for the maintainers.

However, I think that we could still do _something_ for this matter.

My thoughts:

1. Would the maintainers be willing to start using a simple PR review process, where another maintainer would review and approve the PR? That said, I can volunteer here to be one person of the reviewers if the maintainers accept me :) The exact details of approval process need to be agreed, of course. Having "no merges without approval by someone else" may be too strict for this project, but for most non-urgent work, this could be a really good solution.

2. Implement a way to test current functionality with real hardware, e.g. something along the lines of #263 

3. Implement "mock ports" for serial/parallel/ethernet ports to capture input/output from the backends to test for regressions. This could be an extension to #263 where the code is run against a "mock rig". I know this would a lot of work, but it could save a lot of time fixing errors later on.

mdblack98, 2020-09-14:
Mind you 4.0 has not been released yet so not sure what release process you are referring to.  4.0 has undergone some major changes which has sped up things and revealed new behavior in different rigs and required other changes as result...all part of the alpha/beta/release cycle.  4.0RC2 is the latest beta and has about 100 downloads so far.
There already is a rig test program but #263 is meant to improve it....although I doubt it will improve anything other than having some controlled stress testing in lieu of running against WSJT-X or such to make bugs easier to find.
Just like any software project bugs and regressions get introduced.  We've got a few testers out there but nowhere near the 256 rigs in Hamlib to cover all bases.  The biggest test we get is when a new WSJT-X or JTDX gets released and we get thousands of operators and lots of different rigs.
Having a "mock rig" really won't do much as there's no way to emulate what these rigs really do with all the different timing and delays and firmware bugs they have.    "Writing to the test" for hardware isn't really very helpful and would be a herculean effort to do anyways (and get very little use actually so just not worth the effort).  

Mike W9MDB

mikaelnousiainen, 2020-09-14:
Indeed, 4.0 hasn't been released, which is why I asked these questions, as it would be nice to be able to have a really well-working 4.0 release without too many regressions. Also, I'm not implying anything with my comments, Hamlib is a great library, which is why I enjoy using it, but I'm rather trying to figure out ways to make maintenance easier in the long-term.

I guess my main point regarding regressions was to have a way to catch _simple_ regressions: incorrect values being sent to the rig or values incorrectly decoded from rig, which seem to cover the majority of issues that at least I have experienced with Hamlib.

There are lots of more complicated regression, like the timing ones and firmware bugs you mentioned, but I have a feeling these cases are not that common -- and they require access to the real hardware anyway!

Yes, regressions sometimes still happen, but I'm wondering if we could introduce some ways to catch _the most common ones_ and _the easy ones_, which usually happen when the code is being refactored, such as the changes with the Icom extcmds (which was a really welcome change!). I also remember introducing some regressions too with my larger changes
to Icom backends last year, but luckily noticed these myself -- though only after testing on real rigs.
  
Additionally, I see that many of the recent fixes simply cover cases where incorrect data gets sent or it is interpreted incorrectly
(for features that worked previously), so I'm simply seeing some way of handling these types of regressions could potentially help you as a maintainer too!

What do you think about the suggestion to have PR reviews for Hamlib commits?

mdblack98, 2020-09-14:
You're free to review all patches.When I'm making bigger changes I'm doing it to a fork and then merging it like I'm doing now with all the PARM/LEVEL changes.  Turns out what I thought was working broke other things.  This is a complicated section of code and has all sorts of side effects.
And a rig emulator would have caught those.
Mike W9MDB

closed 2020-10-08 by mdblack98