Hamlib/Hamlib #384: FT-897 - Hamlib error: Target VFO unaccessible while getting current frequency

worsthorse, 2020-09-21:
Running WJST V2.2.2 I get this error when trying to connect to an FT-897.  Switching the rig type to FT-857 results in a successful connection.  CAT control also works as expected in other applications not using hamlib.

mdblack98, 2020-09-21:
Fixed several months ago
https://github.com/Hamlib/Hamlib/issues/285

worsthorse, 2020-11-22:
tested again. whatever the #285 fix, error persists.

mdblack98, 2020-11-22:
Could you please send debug?
rigctl -m 1023 -r /dev/ttyUSB0 -s 9600 -vvvvv >&debug.txt
Mike W9MDB

mdblack98, 2020-11-28:
Waiting for debug info on this one....

FabC, 2020-11-29:
I suppose you meant :

rigctl -m **123** -r /dev/ttyUSB0 -s 9600 -vvvvv >&debug.txt

Tnx, 73 Fabrizio IU5GEZ
[debug.txt](https://github.com/Hamlib/Hamlib/files/5612004/debug.txt)

FabC, 2020-11-29:
Workaround:
1) run 
rigctld -m123 ft897 -r /dev/ttyUSB0 -s38400 &
2) start wsjtx with the following configuration
Radio > Rig > "Hamlib NET rigctl"
Network server: "localhost"

73, Fabrizio IU5GEZ

mdblack98, 2020-11-29:
The debug you sent was from Hamlib 3.3
I assume when you are running rigctld you are getting Hamlib 4.0 or 4.1 -- that would explain why it's working.
You can find out which libhamlib WSJT-X will use with this "ldd wsjtx"

Mike W9MDB

 
 

Workaround:

mdblack98, 2020-11-29:
And...if you uninstall hamlib 3.3 you may fix the WSJT-X behavior.
Mike W9MDB

FabC, 2020-11-29:
Starting from the debug:

$ ldd /usr/bin/rigctl
	linux-vdso.so.1 (0x00007fff64096000)
	**libhamlib.so.2 => /usr/local/lib/libhamlib.so.2 (0x00007f9680887000)**
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007f9680837000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f9680814000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9680622000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f968061c000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f96804cd000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f96804af000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f968047f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f9680e3b000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f9680452000)

The (working) daemon:

$ ldd /usr/bin/rigctld
	linux-vdso.so.1 (0x00007ffe908f3000)
	**libhamlib.so.2 => /usr/local/lib/libhamlib.so.2 (0x00007f1c27353000)**
	libreadline.so.8 => /lib/x86_64-linux-gnu/libreadline.so.8 (0x00007f1c27303000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1c272e0000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1c270ee000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1c270e8000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1c26f99000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f1c26f7b000)
	libtinfo.so.6 => /lib/x86_64-linux-gnu/libtinfo.so.6 (0x00007f1c26f4b000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1c27908000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f1c26f1e000)

The installed libhamlib:

 $ dpkg -l | grep -i hamlib
ii  libhamlib-dev:amd64                                         3.3-10build1                                        amd64        Development library to control radio transceivers and receivers
ii  libhamlib-utils                                             3.3-10build1                                        amd64        Utilities to support the hamlib radio control library
**ii  libhamlib2:amd64                                            3.3-10build1                                        amd64        Run-time library to control radio transceivers and receivers**

Are rigctl and rigtld using the same run-time library ?

mdblack98, 2020-11-29:
rigctl and rigctld are using the same library...but wsjtx is probably not.

Uninstall the 3.3 hamlib package
Mike

mdblack98, 2020-11-29:
Also try "ldd wsjtx"....it's probably pointing to the 3.3 libarry.
Mike

FabC, 2020-11-29:
> Also try "ldd wsjtx"....it's probably pointing to the 3.3 libarry. Mike On Sunday, November 29, 2020, 09:23:11 AM CST,

$ ldd /usr/bin/wsjtx  | grep -i ham

No reference(s) to any "libhamlib" reference.

mdblack98, 2020-11-29:
Hmmm...something doesn't make send unless you statically linked wsjtx.
ldd /usr/bin/wsjtx|grep ham libhamlib.so.2 => /lib/x86_64-linux-gnu/libhamlib.so.2 (0x00007f92aa53d000)
What do you get without the grep?  Should be a boatload of libraries..
The other thing you can do is build the version 4 hamlib to install over the /usr path.
 ./configure --prefix=/usr
Mike

FabC, 2020-11-29:
> Hmmm...something doesn't make send unless you statically linked wsjtx. ldd /usr/bin/wsjtx|grep ham libhamlib.so.2 => /lib/x86_64-linux-gnu/libhamlib.so.2 (0x00007f92aa53d000) What do you get without the grep?  

Here it is:

$ ldd /usr/bin/wsjtx
	linux-vdso.so.1 (0x00007ffddd993000)
	libQt5SerialPort.so.5 => /usr/lib/x86_64-linux-gnu/libQt5SerialPort.so.5 (0x00007f2f3460b000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f2f34605000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f2f345e9000)
	libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007f2f343d9000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007f2f34360000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f2f33cc5000)
	libQt5Sql.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Sql.so.5 (0x00007f2f33c75000)
	libQt5Multimedia.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Multimedia.so.5 (0x00007f2f33b5e000)
	libQt5Network.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Network.so.5 (0x00007f2f3399d000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f2f333b2000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f2f32e69000)
	libgfortran.so.4 => /usr/lib/x86_64-linux-gnu/libgfortran.so.4 (0x00007f2f32c88000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2f32aa5000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2f32956000)
	libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f2f32914000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f2f328f9000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2f328d6000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2f326e4000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2f380fa000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f2f326b5000)
	libpulse.so.0 => /usr/lib/x86_64-linux-gnu/libpulse.so.0 (0x00007f2f32660000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f2f32644000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f2f322f8000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f2f322c0000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f2f321b9000)
	libicui18n.so.66 => /usr/lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007f2f31eba000)
	libicuuc.so.66 => /usr/lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007f2f31cd4000)
	libpcre2-16.so.0 => /usr/lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007f2f31c51000)
	libdouble-conversion.so.3 => /usr/lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007f2f31c3b000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f2f31b12000)
	libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f2f31ac6000)
	libpulsecommon-13.99.so => /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-13.99.so (0x00007f2f31a44000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f2f319f3000)
	libnvidia-tls.so.340.108 => /usr/lib/x86_64-linux-gnu/tls/libnvidia-tls.so.340.108 (0x00007f2f317f0000)
	libnvidia-glcore.so.340.108 => /usr/lib/x86_64-linux-gnu/libnvidia-glcore.so.340.108 (0x00007f2f2ebdc000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f2f2ea9d000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f2f2ea88000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f2f2e9c9000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f2f2e99c000)
	libicudata.so.66 => /usr/lib/x86_64-linux-gnu/libicudata.so.66 (0x00007f2f2cedb000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f2f2ce66000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f2f2ce3c000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f2f2cd8d000)
	libwrap.so.0 => /usr/lib/x86_64-linux-gnu/libwrap.so.0 (0x00007f2f2cd81000)
	libsndfile.so.1 => /usr/lib/x86_64-linux-gnu/libsndfile.so.1 (0x00007f2f2cd04000)
	libasyncns.so.0 => /usr/lib/x86_64-linux-gnu/libasyncns.so.0 (0x00007f2f2cafe000)
	libapparmor.so.1 => /usr/lib/x86_64-linux-gnu/libapparmor.so.1 (0x00007f2f2cae7000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f2f2cadc000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f2f2cad6000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f2f2cace000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f2f2caa5000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f2f2ca82000)
	libgcrypt.so.20 => /usr/lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f2f2c964000)
	libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007f2f2c947000)
	libFLAC.so.8 => /usr/lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007f2f2c909000)
	libogg.so.0 => /usr/lib/x86_64-linux-gnu/libogg.so.0 (0x00007f2f2c8fc000)
	libvorbis.so.0 => /usr/lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f2f2c8ce000)
	libvorbisenc.so.2 => /usr/lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f2f2c821000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f2f2c805000)
	libbsd.so.0 => /usr/lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f2f2c7eb000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f2f2c7c8000)

mdblack98, 2020-11-29:
Hamlib must be statically linked since it doesn't show as a library.
You will have to compile it yourself or use rigctld and "Hamlib NET rigctl" in WSJT-X to get the new/fixed version.
Mike

FabC, 2020-11-29:
> Hamlib must be statically linked since it doesn't show as a library. You will have to compile it yourself or use rigctld and "Hamlib NET rigctl" in WSJT-X to get the new/fixed version. Mike 

Thanks Mike, understand.
For the sake of completeness, here a message that is shown after starting wsjtx/rigctld:

Hamlib: netrigctl_open:  unknown return from netrigctl_transaction=9
Hamlib: rig_set_vfo: set_vfo VFOB failed with 'Feature not available'

It looks strange to me: the FT-897 has got VFO A/B capabilities and why wsjtx is trying to use the VFO capability. The spit operations (Settings/Radio) is set to "none".

Ciao
Fabrizio IU5GEZ

mdblack98, 2020-11-30:
Try a pull and see if it works now.
What serial# is your rig?  It appears older FT897's won't support the VFO queries.
Mike W9MDB

FabC, 2020-11-30:
> Try a pull and see if it works now. What serial# is your rig?  It appears older FT897's won't support the VFO queries. Mike W9MDB 

It is 220123, don't know how much old it is, but indeed the VFO commands are not working...

Fabrizio IU5GEZ

mdblack98, 2020-12-01:
set_vfo and get_vfo have been removed from the FT-897 -- they don't work on all FT-897's apparently.
The only VFO operation available is toggle.
From rigctl it will be the "G TOGGLE".
Are you saying you are have compiled WSJT-X with the new library or are using rigctld?
WSJT-X is not working?
Can you send debug?
Mike W9MDB

mdblack98, 2020-12-31:
FT-897D gets get_vfo error on MicroHam device.
Need to add 897D explicitly and re-add the get/set vfo function for the 897D that were taken out of the 897 which apparently does not support get/set vfo.

closed 2021-02-03 by mdblack98