Hamlib/Hamlib #386: TH-D7A - Command rejected by the rig

baalansellers, 2020-09-29:
Hello,
I know TH-D7A is marked as alpha, but I'm interested in helping improve the integration.

Right now I'm attempting to use GPredict+Hamlib to control the HT for satellite operation, but when I engage it quickly loses connection, though it does set the initial frequency. So I went to the command line to run `rigctl` in verbose mode to set the frequency manually, which does change the frequency, but I get the following output with an error at the end stating the rig rejected the command (seems that maybe the message is terminated incorrectly - also, this _is_ on a RPi)

`
Rig command: F
Frequency: 145818000
rig_set_freq called
th_set_freq: called
kenwood_transaction called
kenwood_transaction: cmdstr = FQ 00145818750,1
serial_flush called
write_block called
write_block(): TX 17 bytes
0000    46 51 20 30 30 31 34 35 38 31 38 37 35 30 2c 31     FQ 00145818750,1
0010    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 15 characters
0000    46 51 20 30 30 31 34 35 38 31 38 37 35 30 2c        FQ 00145818750, 
kenwood_transaction: Command is not correctly terminated 'FQ 00145818750,'
kenwood_transaction: cmdstr = FQ 00145818750,1
serial_flush called
write_block called
write_block(): TX 17 bytes
0000    46 51 20 30 30 31 34 35 38 31 38 37 35 30 2c 31     FQ 00145818750,1
0010    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
kenwood_transaction: Unknown command or rig busy 'FQ 00145818750,1'
kenwood_transaction: Retrying shortly
read_string called
read_string(): Timed out 0.200231 seconds after 0 chars
kenwood_transaction: cmdstr = FQ 00145818750,1
serial_flush called
write_block called
write_block(): TX 17 bytes
0000    46 51 20 30 30 31 34 35 38 31 38 37 35 30 2c 31     FQ 00145818750,1
0010    0d                                                  .               
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;             
read_string called
read_string(): RX 2 characters
0000    3f 0d                                               ?.              
kenwood_transaction: Unknown command or rig busy 'FQ 00145818750,1'
set_freq: error = Command rejected by the rig
`

If I try to feed a raw command, the frequency changes on the radio, but to something a few Hz above what I set (also noticed this in the attempt above too). It tries to read a second time and gets a timeout. Maybe I need to adjust the timing between reads? Is that even possible for me to configure?

`
Rig command: w       
Cmd: FQ 00147820000,1
serial_flush called
write_block called
write_block(): TX 17 bytes
0000    46 51 20 30 30 31 34 37 38 32 30 30 30 30 2c 31     FQ 00147820000,1
0010    0d                                                  .               
read_string called
read_string(): RX 17 characters
0000    46 51 20 30 30 31 34 37 38 32 35 30 30 30 2c 31     FQ 00147825000,1
0010    0d                                                  .               
Reply: FQ 00147825000,1
read_string called
read_string(): Timed out 0.200232 seconds after 0 chars
`

mdblack98, 2020-09-29:
Actually it's marked as beta.  What version hamlib are you running?
Do you happen to have a manual that shows the CAT protocol for the rig?
Please contact me at mdblack98@yahoo.com and we can hook up and see about getting this working.
Mike W9MDB

baalansellers, 2020-09-30:
Oh, well then yeah, maybe I'm on an older version somehow. I installed hamlib version 3.3. 

I have the PDF version of the full manual for the HT, but it doesn't include any information on CAT commands. I'll do some googling and see if I can find such a document, though.

And sure, I'll email you directly and see if we can figure it out. Very likely user error - ha.

mdblack98, 2020-09-30:
If you can't compile try 4.0RC2 herehttps://sourceforge.net/projects/hamlib/files/

If you can compile try the git repository.
https://github.com/Hamlib/Hamlib

Mike W9MDB

baalansellers, 2020-09-30:
roger roger, I'll try compiling that candidate and let you know how it turns out. Thanks!

baalansellers, 2020-09-30:
Wonderful, the RC seems to work a lot better with the Kenwood TH-D7A now. I don't get any timeouts or other errors when test running `rigctl` in verbose mode, and I even got GPredict to stay engaged on a satellite that has only a downlink. Woohoo! 
However, it runs into a problem when trying to control the secondary VFO (the B VFO). It may be a GPredict issue now, I'm gonna do some more testing and log reviewing to try and narrow down, but I think it is safe to say the issue I started this thread on is solved, so I'll mark it so.

closed 2020-09-30 by baalansellers