Hamlib/Hamlib #394: FTDX101D IF and VOXDELAY not working

mdblack98, 2020-10-03:
Hi Mike

Trying to set up IF width and IF shift sliders

Using the scope decode:

\get_level Main IF     - this reads IF shift *CAT manual:  IS...

rig IF shift set to +980Hz

return on scope is: IS00+0980;    - good!

return from command line rigctl is:  0

rig IF shift set to -1160Hz

return on scope is:    IS00-1160;

return from command line rigctl is:   0

You are sending the correct command - but not interpreting the result

I then send:   \set_level Main IF -346   -    No effect on the rig

Scope decode of rigctl send is:  IS0-0346;

rigctl should have responded:   IS00-0346;  
(ie: I  S  P1   P2  P3  P4(4 chars) 
P1 = 0 = Main    P2 = 0  (fixed)   P3 - or +    P4 4 data chars 0-1200Hz

//////////////////////////////

VOX delay   - there seems to be an error in the Yaesu CAT manual here

From the manual, Vox delay 00:  30msec      01: 50msec     02 100ms   ie non linear ... I use a lookup table
So I set 100ms  (data = 02 )
I send \get_level Main VOXDELAY       rigctl returns: 0

Scope decode:  VD; (correct)     response VD02;  
My piWebCAT config for the response is 5 bytes:    VDtu;   (mapped as tens and units)

BUT .. the manual says V   D    P1   P1   P1   P1    ;   =   7 bytes  !!!!

....is this the problem???

In my initial comments on checking out these commands with rigctl,
I was only commenting on failed responses  .... not incorrect responses!!

The persistent zero responses were VOXDELAY, ANTIVOX and IF

The rest are I think ok  -.... I haven' tset them all up yet.

///////////////////

I thought IF width was absent - but of course it is returned by \get_mode   - which is already working here.

I will use   \get_mode    AND  \set_mode   separately for the IF width slider
The result will have two separate parsing results - but they will be Hamlib-specific and
apply to all rigs  - so can be hard coded for Hamlib operation.

/////////////////////

Thanks be to the Siglent SDS2202X-E scope
I dont think i could have  developed piWebCAT without it
  .... particularly for FT920  FT1000MkV where the documentation in incomplete.
(I have a FT920 here)

Kind regards

Ian

mdblack98, 2020-10-03:
VOXDELAY should be working

Ian-G3VPX, 2020-10-05:
FTdx101D RPi terminal rigctl
\get_level Main VOXDELAY   - returns correct values in ms
\set_level Main VOXDELAY 1600   no effect   scope decode: VD0033;  (for all bw attempts)\set_level Main
\set_level Main VOXDELAY 160    - same

The CAT manual is wrong:   should only send 5 bytes
\send_cmd_rx VD05; 0    correctly sets  index 5 as 250ms

Ian-G3VPX, 2020-10-05:
FTdx101D RPi terminal rigctl    IF shift   'IF'
   IF shift set to +840Hz   \get_level Main IF returns 0
    scope decode is correct, ie:  IS0;  >   IS00+0840;    so data parsing error

   \set_level Main IF 1200   has no effect
    scope decode;   IS0+1000;   >  ?;       should have sent:  IS00+1200;  (fixed 0 missign and wrong value)

mdblack98, 2020-10-05:
Try this one again...just made it 5 bytes
Mike

closed 2020-10-06 by mdblack98