Hamlib/Hamlib #407: FTdx101D CWPITCH control

Ian-G3VPX, 2020-10-09:
FTdx101D  KP command  K  P  P1  P1  ;  P1 is 00 to 75  for  300Hz to 1050Hz

Using command line rigctl:  \get_level Main CWPITCH
300Hz   returns   300
1050Hz   returns   4000

\set_level Main CWPITCH n
n = 300   sets 300Hz
n = 400   sets 320Hz
n = 600  sets  360Hz  scope decode KP06;
n = 1000 sets 440Hz   scope secode KP14;      300 + 10*14 = 440
n = 2000 sets 460Hz
n = 3000 sets 460Hz
n= 4000 sets 460Hz     scope decode KP16;

mdblack98, 2020-10-09:
Try again...scale was wrong for that.
Mike

Ian-G3VPX, 2020-10-10:
CWPITCH now working perfectly 300 - 1050Hz
- Sets  correctly from slider control.   Sync reads sets slider and  text freq display matches rig
Ian G3VPX

closed 2020-10-11 by mdblack98