Hamlib/Hamlib #408: FTdx101D  \set_vfo   \get_vfo not working

Ian-G3VPX, 2020-10-10:
**FTdx101D VFO selection problem**

\get_vfo

Main selected on rig:
v    >   result   =  Main      scope decode VS;    >   VS0;    = correct

Sub Selected on rig 
v    >   result   =  VFOB      scope decode:  VS;  >   VS1;  = correct   *********

\set_vfo    ....    scope decode:   no CAT command sent

V Main
set_vfo: error = Feature not impemented

V VFOA
set_vfo:  error = Invalid parameter

V Sub
set_vfo:   error = Feature not impemented

V VFOB
set_vfo:  error = Invalid parameter

\set_vfo Sub
set_vfo:   error = Feature not impemented

Ian-G3VPX, 2020-10-10:
\set_vfo Main    and  \set_vfo Sub   ( and \get_vfo )
- working fine at rigctl command line
- Not working in piWebCAT with rigctld   -    nothing on scope on Tx line
I echoed the string (ie \set_vfo Main) from the PHP code line immediately
before the socketwrte  and it looks ok
Is this possible?
Ian   G3VPX

mdblack98, 2020-10-10:
Seems to work for me talking through rigctld....
pi@piWebCAT:~ $ rigctl -m 2
Rig command: \get_vfoVFO: Sub
Rig command: \set_vfo Main
Rig command: \get_vfoVFO: Main
Rig command: \set_vfo Sub
Rig command: \get_vfoVFO: Sub
Rig command:

Ian-G3VPX, 2020-10-11:
This is working fine now   - (rig > piWebCAT   and  piWebCAT > rig)  
... rigctl command line ok
... piWebCAT rigctld ok  
(having to tmeporarily use:  if(cat == 'Sub\n')  ... need to tidy this up and remove '\n' for all returns
   ... most returns are numeric and parseInt ignores '\n'  )

closed 2020-10-11 by mdblack98