Hamlib/Hamlib #410: FTdx101D  \set_func XIT not working

Ian-G3VPX, 2020-10-10:
u RIT   - ok

U RIT n   - ok

u XIT   - ok

U XIT 0    fails     scope decode XT09;   should be XT0;

U XIT 1  fails       scope decode XT09;     should be XT0;

Ian-G3VPX, 2020-10-10:
All fine on Main receiver 
BUT
With Main    I set RIT -99Hz  (illuminated display CLAR RX  -99Hz
Switch to Sub  (on rig)
display persists unchamned for Main   ok
Set RIT on Sub to -5794Hz using slider     Sub display illuminated:  CLAR RX -5794Hz
2 seconds later slider flips back to -99Hz  (but rig still shows -5795 for Sub)
- so I am reading Main RIT on sync refresh

RIT commands are \get_rit Main   \set_rit Main irrespective of VFO setting
... maybe should be \get_rit Sub etc    I can make it VFO specific 
(piWebCAT structured to use A/B (current vfo)   or  X  (not vfo dependent)
BUT in the above example, when switched to Sub
     \set_rit  Main -5794  sets Sub RIT  to -5794    and \get_rit Main reads back from Main RIT
Ian

mdblack98, 2020-10-10:
Are you saying there is a bug there?  I can't tell from what you're saying.

Ian-G3VPX, 2020-10-12:
With Sub receiver selected, get_rit and get_xit   still reading Main

rig has no means to target a specified VFO  Only works on current VFO
 - so use either Main or Sub in command is ok  - (But woud confuse a user unless clarified!)

With Sub selected:
\get_rit Sub   returns Main value (0 in video)
\get_xit Main returns Main  value

See: http://www.g3vpx.net/ritxit.mp4

mdblack98, 2020-10-12:
RIT/XIT applies to RX/TX VFO -- not to Main/Sub or A/B.
Just depends on which way split is going.

closed 2020-10-12 by mdblack98