Hamlib/Hamlib #411: FTDX101D TUNE not working

mdblack98, 2020-10-10:
U Main TUNER 1 and U Main TUNER 2 both send  AC001; =   tuner  on
BUT
if you change rigctl to send :

U Main TUNER 1    >   AC002;

U Main TUNER 0    >  AC000;

that should be very useful.

Because AC002; switches on the tuner AND then starts a tuner  (about 10 max)

The tuner is then left ON.

Resending AC002; does another tune.

Sending AC000; switches the tuner off.

Ian-G3VPX, 2020-10-10:
This is still sending AC001;
I thought you were going to change ti to send AC002;
Ian

mdblack98, 2020-10-10:
Are you opening/closing rigctld every command?  I see it cycling through sockets like crazy.Best if you just keep it open and reconnect if it fails.

mdblack98, 2020-10-10:
You don't have the latest version installed.
Mike

Ian-G3VPX, 2020-10-10:
Tuner and Tune buttons now working exactly as planned
- Very effective and fit for purpose
Ian

closed 2020-10-10 by mdblack98