Hamlib/Hamlib #417: FTdx101d func COMP not working

Ian-G3VPX, 2020-10-13:
Must have missed this one ( fixed COMP level and Tx COMP)

u Main COMP
get_func: error = Invalid Paraameter

U Main COMP 1 
set_func: error = Communication bus collision

debug:
[COMP.txt](https://github.com/Hamlib/Hamlib/files/5372399/COMP.txt)

mdblack98, 2020-10-13:
Try again...101DX has  new format for the command.
Mike

Ian-G3VPX, 2020-10-15:
U Main COMP 1  / 0    ok

u Main COMP always returns 1

if rig comp on:  u Main COMP   decode:  send:  PR0;   receive: PR01;  correct -  returns 1 = correct

if rig comp off:  u Main COMP   decode:  send:  PR0;   receive: PR00;  correct  BUT  -  returns 1 = incorrect

mdblack98, 2020-10-15:
Should be working now.

Ian-G3VPX, 2020-10-15:
Fixed in command line rigctl

? not fixed in rigctld
Ian

mdblack98, 2020-10-15:
You must be running the rigctld in /usr/bin which is an older version.
Mike

Ian-G3VPX, 2020-10-15:
This is now working correctly with rigctld

Thanks

Ian
G3VPX

closed 2020-10-16 by mdblack98