Hamlib/Hamlib #422: End of CW indication

OH1KH, 2020-10-17:
Lately I have been thinking a way to change CW speed on the fly. It seems to be impossible at least with IC7300 and hamlib as there is no proper "end of buffer" indication.

With Winkeyer(USB) it is easy to create a CW macro   "tu +++5nn--- 123" A contest message where report is sent faster than other text. + increases and - decreases speed (this case programmed 5WPM/mark).

When IC7300 is sending CW via hamlib speed can be changed up and down but timing does not work. With a short  macro string above IC7300 will send all text, then raise and lower speed when text is already sent.

We could split string to "tu ", "+++5nn" and "---123" and it would work if we just could get indication that last letter of string is sent.
I even tried to send the string letter by letter but even that did not succeed. It should, but when letter is still going out it reports RPRT0 and that is the only feedback and if it is used to release next letter, or speed change, the timing is wrong.

I even  considered to make WPM calculated delay to top level program for feeding letters but not yet tried it. How ever it would be more or less blind way to find buffer end.

All this might be impossible because of rig's CAT way controlling CW but I have not studied it at serial connection binary level, so this should be checked, too.

mdblack98, 2020-10-17:
I think we could add the + and - doing the same thing as the winkeyer.  Nobody uses those chars in normal CW, right?
So a 2-step process.
#1 Add a morse_wait command that will work with breakin or semibreakin and will wait for PTT to go quiet for at least 1 inter-word time period.
Then if that works I can add the + to do the keyspd increase.

With just the morse_wait you could do this
L KEYSPD 18
bTU
\morse_wait
L KEYSPD 25
b5NN
\morse_wait
L KEYSPD 18
b123
Note that you will automatically get an interword space with morse_wait so would not have to use the space after the TU or 5NN
L KEY
Then...if that works OK I can intercept the + and - and to the KEYSPD change automatically.
How much does + change the key speed on the Winkeyer?

OH1KH, 2020-10-18:
Hi!
Sounds promising.

Actually cqrlog does it in CQ macro handling. If + or - is found in text the step is 5WPM, but can be programmed (hard coded in source, see below).
Winkeyer does not do it, but allows that because it can handle speed stepping in the middle of macro string that hamlib CW doesn't.
So the top level program can handle + and - but we just need to know that things happen in order. At the moment sending letter by letter produces nice output but if we want to send speed command in the middle it must be done after letter is actually sent, not when we get RPRT0 because letter is not finished and sending then it just causes rig's buffer to fill and somehow at least ic7300 sends first all text and then places speed commands.
When macro text is long and sending out keyboard step up and down keys will change speed in middle but it is not controlled I.E. happens, but nobody knows exactly after what letter the step is handled.

I have not tested to use PTT info as end indicator. It might even work! I have to make a test program for that. But on the other hand hamlib could could that. Perhaps returning RPRT0 after PTT when (just one) letter is sent. In case of error report it should not be tied to PTT but returned immediately.
 
procedure TCWWinKeyerUSB.SendText(text : String); 

...

if (Pos('+',text) > 0) or (Pos('-',text) > 0) then
  begin
    for i:=1 to Length(text) do
    begin
      if text[i] = '+' then
      begin
        spd := spd+5;
        ser.SendByte($1C);
        ser.SendByte(spd);
        Continue
      end
      else begin
        if text[i] = '-' then
        begin
          spd := spd-5;
          ser.SendByte($1C);
          ser.SendByte(spd);
          Continue
        end
      end;
      case text[i] of
        -special char cases...
      else 
        -send just letter...

mdblack98, 2020-10-18:
I did some testing on a friend's 7300 and on BREAKIN ptt is a good indicator.  On SEMI BREAKIN PTT drops between chars so would have to have logic to wait for the space which is a bit more complicated.
I think I'll code up the PTT wait for BREAKIN first.  If that works I can look at also handling SEMI BREAKIN.
That will be adding a "morse_wait" command that will wait until the current string is done.
Mike W9MDB

mdblack98, 2020-10-18:
Give the wait_morse try now.  It's implemented for the Icom rigs.
Have to add the other rigs next.

OH1KH, 2020-10-19:
Hi!
I did a new pull an compile and can see wait_morse with git show. So I think I'm at right version.

How ever if I paste (ic7300 is  BKIN mode):
L KEYSPD 18
bTU
\wait_morse
L KEYSPD 25
b5NN
\wait_morse
L KEYSPD 19
bDE OH1KH

to telnet console the response is only one "T" with CW and from console:
bDE OH1KHRPRT 0
RPRT 0
RPRT 0
RPRT 0
RPRT 0
RPRT 0
RPRT 0

And rig remains to 19WPM (so at least speed changes but can not see if it visits 25wpm also)

If I manually enter a long string to send from telnet and while sending give \wait_morse command sending stops immediately.
Am I now on wrong track?

mdblack98, 2020-10-19:
Can you turn on debug in rigctld please and send me the debug log.txt from this?
rigctld -m 3073 -r com1 -s 57600 -vvvvv -Z >&log.txt
Mike W9MDB

OH1KH, 2020-10-19:
Sent PM to yahoo

closed 2020-10-20 by mdblack98