Hamlib/Hamlib #424: FTDX101D band change no memory

mdblack98, 2020-10-20:
Band changes using rig remember band settings but frequency changes from CAT do not.
FTDX101D has access to band stacking register so we can restore the latest band stack when changing frequencies.
Should we be be setting the band stack too before we change freqs?

mdblack98, 2020-10-25:
Yaesu BS command apparently returns all the commands necessary to restore band settings.
So it appears we can query and restore.

Ian-G3VPX, 2020-11-03:
Before testing, I set the three 20m registers to defined frequencies.
All the testing, the rig had only changed/used one of these three.
(So you need to hit the 20m button whilst on 20m to change the others)

Start on 7.130.000
F Main 14200000    > 14.200.000
F Main 14300000    > 14.300.000
F Main 7150000     >  7.150.000
20m band button    > 14.300.000     as your expectation

Start on 7.150.000
F Main 14200100    	> 14.200.100
Tune rig to 14.200.200  > 14.200.200
F Main 7160000     	>  7.160.000
20m band button    	> 14.200.200	 

Start on 7.110.000
F Main 14200700    > 14.200.700
F Main 14200800    > 14.200.800
40m button     	   >  7.110.000
20m button    	   > 14.200.800  

So the use of BS for the new band (OR band button for the new band)
appears to store the current band frequency before actually changing.

///////////////
Pressing 80m button in range 3.5 - 4.0 stores the frequency as one of the threee 80m frequencies
Pressing 80m button  at 3.4 or 4.1 does not store the frequency

So although the rig is only Tx enabled for UK 3.5 - 3.8, it will store 3.8 - 4.0... which isn't good idea.

I investigated tuning above and below 3.500.000 with piWebCAT > rigctld   - seems ok.
.... something is different   relay click going out of band with piWebCAT (= F Main)
      no click with tuning dial.

I investigated effect of crossing band mrgin with negatve clrifier on  - seemed ok 
BUT   in fact, I noticed the clrifier going off.

.. staying in band:

Set RIT to -1000Hz    - tune with dial - no problem   - RIT offset maintained

Set RIT to -1000Hz    - tune with piWebCAT (= F Main fff)  -   RIT offset zeroed  

Set RIT to -1000Hz    - send F Main 3666666 from command line   -   RIT offset zeroed  

Set RIT to -1000Hz    - send  w BS01;      RIT offset zeroed

So BS appears to zero the clarifier offset ...  
... are you send it with every frequency change.

I thought you were sending it just on entering the band.

Even if if you do that then there may be an issue at the band margin because a CAT 
application needs to be allowed tune tune slightly beyond the band margin and back again (or block such)

In piWebCAT, I only change tuning scale etc etc on entering a new band .... to allow the above.
..... which is long way on the dial!!!!

However, you need to send BS on leaving the band to effect the memory   ... so we have to live
with the clirifier zeroed feature ??

Ian
G3VPX

mdblack98, 2020-11-03:
You should only be seeing the BS command on band change.
And if the rig memory doesn't remember the clarifier setting not much we can do about that.  Kind of makes sense that they would zero it on band change actually.  
It seems I need to restrict the band memory recall to just the max ITU region bands at least....
Mike W9MDB

closed 2021-03-09 by mdblack98