Hamlib/Hamlib #428: IC7000 \set_mode    passband incorrect

Ian-G3VPX, 2020-10-29:
IC7000  - \set_mode LSB 2000    etc   not working correctly for passband

in PiWebCAT direct CI-V mode, I use ... 0x1A 0x03  0xnn.. for passband
- work accurately on bith get and set.   
CAT section of manual says: nn = 0 -40   >  bw = 50Hz > 3600Hz
-- Manual doesn't give details os mapping
Actually is:  0 - 9    >   bw 50Hz to  500Hz in 50Hz steps
              10 - 40  >   bw 600Hz to 3600Hz in 100Hz steps
So I have a lookup table.

I carefully checked out rigctl at command line against the rig:

\get_mode  no problem
  ...  returns eg:  LSB 2000 etc.  Passband is correct for whole range

\set_mode LSB nnnn  is very incorrect
Examples  ( I am not using --vfo mode here)

M LSB 3600    >   3600 on rig
M LSB 3000    >   3600
M LSB 2700    >   3600
M LSB 2400    >   3000
M LSB 2300    >   1300
M LSB 2100    >   1300
M LSB 1000    >   1300
M LSB 600     >   1300
M LSB 100     >   1300
M LSB 50      >   1300

debug attached
Ian
G3VPX
[IC7000_set_mode.txt](https://github.com/Hamlib/Hamlib/files/5457089/IC7000_set_mode.txt)

mdblack98, 2020-10-29:
Can you send me debug from a few get_mode commands?The logic in this is rather convoluted.

mdblack98, 2020-10-29:
Can you please check a few things.The USB/LSB widths should working OK now...but there are two conditions need testing.#1 Probably RTTY mode is limited to 2700Hz value 31...so what happens when > 2700 is set with RTTY mode...error...or does it clip it?

#2 AM mode has it's own mapping.  Can you please test to see what it is?
Mike W9MDB

Ian-G3VPX, 2020-11-02:
Testing on IC7000

Using rigctl -m   3060 -s 19200 -r 

Rig command:  M LSB 2000
set_mode: error = command rejected by the rig

\get_mode    is fine

F 3777666   is fine

using --vfo mode
M VFOA LSB 2000M   - same response

IC7000 is on another RPi on local IP 192.168.1.117   /dev/ttyAMA0
Use port 5901

Router forwards ******.dyndns.org:5901 to 192.168.1.117:5900 for VNC (tested here = OK !)
same user and pw

Ian
G3VPX

[MLSB2000.txt](https://github.com/Hamlib/Hamlib/files/5475388/MLSB2000.txt)

mdblack98, 2020-11-03:
get_mode working but set_mode needs new code to implement bandwidth for the IC-7000.
Deferred for 4.1 developmenrt.l

Ian-G3VPX, 2021-02-17:
BW setting worse than originally reported
All commands  M LSB bw     set bw to 344

Ian
G3VPX

[IC7000_BW_log.txt](https://github.com/Hamlib/Hamlib/files/5998744/IC7000_BW_log.txt)
[IC7000_LSB_BW_setting.txt](https://github.com/Hamlib/Hamlib/files/5998745/IC7000_LSB_BW_setting.txt)

mdblack98, 2021-02-18:
Try again -- should be working now.
Mike W9MDB

closed 2021-02-22 by mdblack98