Hamlib/Hamlib #432: Yaesu break-in delay scale

mikaelnousiainen, 2020-11-04:
@mdblack98 I'm trying to fix more bugs in Yaesu CAT commands (will be able to test on FTDX 5000 and FT-991 now) and ran into some issues, as I couldn't understand how the conversion for break-in delay from tenths-of-dots per second to milliseconds goes.

I'm seeing two conversions here, which don't match at all:

When setting the delay, conversion is a simple multiplication by 100: https://github.com/Hamlib/Hamlib/blob/master/rigs/yaesu/newcat.c#L3178

But when getting the delay, conversion is 5000 / value: https://github.com/Hamlib/Hamlib/blob/master/rigs/yaesu/newcat.c#L3931

Neither of those make sense to me... what does the "tenths-of-dots" per second mean? At what speed?

mikaelnousiainen, 2020-11-04:
The rest of Yaesu rigs (excluding FTDX101) all seem to use milliseconds for the break-in delay, so this could be easily fixed.

mdblack98, 2020-11-04:
Not quite so easily fixed....
It appears the BKINDL implementation was based on Icom commands which talk in dot length -- so Icom takes into account the CW speed setting and the delay will adjust based on the CW speed setting.
Yaesu thinks in ms and this was apparently not implemented correctly in hamlib.
dot=dit and the setting is 10ths of a dot so a minimum BKINDL to avoid relay clicking would be 3 dots = 3*10 = 30 for Icom rigs regardless of CW speed.
For Yaesu and other rigs we'll need to query the CW keying speed to compute the ms for the dit length.
P = di da da di = 1 1 3 1 3 1 1 (3) = 14 elements
A = di da = 1 1 3 (3) = 8 elements
R = di da di = 1 1 3 1 1 (3) = 10 elements
I = di di = 1 1 1 (3) = 6 elements
S = di di di = 1 1 1 1 1 [7] = 12 elements
Total = 50 elements
() = intercharacter
[] = interword
http://kent-engineers.com/codespeed.htm

So for rigs that need ms we need to read KEYSPD whenever breakin is set or get and compute ms and then to/from dot tenths depending on set or get
I think this will work -- get the keyspd and compute ms-per-dot.    Then we can convert to/from dot tenths.
To read BKINDLY
msperdot = ceil(240*(5.0/KEYSPD)
dot10ths = floor(msfromrig/msperdot)*10
To set BKINDLY
msperdot = ceil(240*(5.0/KEYSPD)
mstorig = msperdot*(dot10ths/10.0)

mdblack98, 2020-11-04:
It should convert like this

DotTenths | KeySpd | ms | ms2tenths
10 | 5 | 240 | 10
10 | 10 | 120 | 10
10 | 15 | 80 | 10
10 | 20 | 60 | 10
20 | 5 | 480 | 20
20 | 10 | 240 | 20
20 | 15 | 160 | 20
20 | 20 | 120 | 20

mikaelnousiainen, 2020-11-04:
Right, yes, so the API contract is simply following Icom's definition, which is difficult to replicate on Yaesu rigs. I'll see what I can do to fix this. I just did some testing on FT-991 and many levels/funcs are broken plus some weirdness in mode handling... most of these should be quick fixes, though :crossed_fingers:

mikaelnousiainen, 2020-11-06:
@mdblack98 We can do it this way (through conversion), but I was also thinking of using a separate level with the argument in milliseconds. Then I found `RIG_LEVEL_BKIN_DLYMS` in Hamlib code, though that level is only referenced by 2 rare backends and not really implemented anywhere. Thoughts? Or should we deprecate/remove `RIG_LEVEL_BKIN_DLYMS`?

mdblack98, 2020-11-06:
That flag isn't implemented anywhere that I see...it's referenced in trc80.c but it's not implemented at all.  It was added in 2009.
I would like to get rid of it since it's not used at all and add a conversion routine maybe in misc.c to convert to/from tenths of dots to ms.   
That way BKIN works the same for all rigs and is internally adapted as needed.
Mike

mikaelnousiainen, 2020-11-06:
Ok, that's probably the best option -- to keep things consistent. I'll add the conversion routines to Yaesu newcat code and remove support for `RIG_LEVEL_BKIN_DLYMS` in a separate PR.

mdblack98, 2020-11-06:
Let's put the conversion in misc.c as it may well apply to other rigs too.

mikaelnousiainen, 2020-11-07:
@mdblack98 Agreed. I've added the conversion routines to misc.c/misc.h. I'll need to do a couple of small fixes and testing with FTDX 5000 and FT-991 first before creating a PR with lots of Yaesu fixes.

The Yaesu rigs seem to be picky about which commands they accept in which mode. For example, querying for IF shift in AM/FM modes simply returns the ambiguous `?` response, which should not trigger retries in this case. How hard can it be to implement proper error codes... oh well :)

mikaelnousiainen, 2020-11-07:
Ok, break-in delay seems to work, though it's a bit confusing to see that changing key speed "magically" changes break-in delay too. In a way it would make more sense to provide a separate level for rigs that support only absolute time for break-in delay... but I'm fine keeping it like this for now.

mikaelnousiainen, 2020-11-08:
PR #435 fixes this.

closed 2020-11-08 by mikaelnousiainen