Hamlib/Hamlib #434: Yaesu error response `?` interpretation

mikaelnousiainen, 2020-11-07:
@mdblack98 I've got a question regarding the `?` (question mark) responses from Yaesu rigs: It seems the error responses for Yaesu rigs are not really documented anywhere and the interpretation that a question mark would indicate the rig being "busy" originates from some Kenwood code -- am I correct here?

Based on my tests, FTDX101, FTDX 5000 and FT-991 use the `?` response to indicate that the command parameter isn't valid (e.g. invalid memory channel number) or that the command is not supported in the current mode (e.g. IF shift or NR level in FM mode on a FT-991). Retrying the command never helps in these cases and the "command timed out" error code used currently is misleading.

I was wondering if we should simply change the `?` to indicate `RIG_ERJCTED`, which IMO would be the correct interpretation as the use of this response seems to be (at least somewhat) consistent across Yaesu rigs.

mdblack98, 2020-11-07:
Theh code is already there -- so you have to set this value on, call, then off again.
                if (priv->question_mark_response_means_rejected)                {                    /* Some commands, like MR and MC return "?;" when choosing a channel that doesn't exist */                    rig_debug(RIG_DEBUG_ERR, "%s: Command rejected: '%s'\n", __func__,                              priv->cmd_str);                    return -RIG_ERJCTED;                }

Like this one    /* Set Memory Channel Number ************** */    rig_debug(RIG_DEBUG_TRACE, "channel_num = %d, vfo = %s\n", ch, rig_strvfo(vfo));
    snprintf(priv->cmd_str, sizeof(priv->cmd_str), "MC%03d%c", ch, cat_term);
    rig_debug(RIG_DEBUG_TRACE, "%s: cmd_str = %s\n", __func__, priv->cmd_str);
    priv->question_mark_response_means_rejected = 1;    err = newcat_set_cmd(rig);    priv->question_mark_response_means_rejected = 0;

mikaelnousiainen, 2020-11-07:
@mdblack98 Yes, I added that flag in my last commit, but according to my tests having a separate flag doesn't make sense, as it would have to be enabled basically for all commands. That's why I was thinking of removing the flag altogether and simply returning `RIG_ERJCTED` always when encountering `?` as a response for newcat commands. So if there is no known reason to have the retry logic for `?` response, I'd like to change that.

mdblack98, 2020-11-07:
The ? can also mean busy.  Happens during band changes and such on Yaesu rigs.
It's unfortunate that Yaesu doesn't distinguish between the two.  Basically we should never send a bad command so the ? response meaning bad command should be very rare to nonexistent.

mikaelnousiainen, 2020-11-07:
Right, ok. The problem is that especially with FT-991 one cannot get status of many levels in FM or RTTY modes as we get a ? response. That leads to rig communication being stalled until retries are complete. Querying e.g. MIC GAIN in RTTY mode or NR level in FM mode trigger this and waiting for the retries to complete (and fail) makes the problem really annoying.

Soit's not about the command itself being bad, but a certain context (e.g. mode) makes the command bad and rejected. There are many of these commands and the vary between the rigs.

Would it make more sense to have the retry/busy logic as a flag, enabled for the specific case of band switching? Which one would be more common -- the error indicating rig is busy (and retries are needed) or that the command was rejected? I'd make the default behavior for the more common case...

mdblack98, 2020-11-08:
I've seen busy status on newcat_set_command but don't think I've ever seen it on newcat_get_command.
4.0rc3 is going out soon and I don't want to put in a change on this until 4.1 development so it can be tested longer.
But...I would think we could get rid of the retry in newcat_get_command.
As far as bad channels though the backend should be returning an error and never call newcat_get_cmd.  That's what the chan_list is for to do error checking.
Mike W9MDB

mikaelnousiainen, 2020-11-08:
Ok, I'll keep the flag there, but enable it for newcat_get_cmd, as that's where the problem is more severe -- if that's what you meant. I'll also add the flag to a couple of set levels I've identified as problematic.

Also, with invalid memory channels I simply meant channel numbers that are in the valid range, but do not have any saved data yet. E.g. if the user has not yet saved anything in any of the channels, the rig will give this response for selecting any channel. So the meaning of ? indicates an empty channel too. The flag helps with this currently, of course.

mikaelnousiainen, 2020-11-08:
PR #435 improves handling of the `?` response as discussed above.

closed 2020-11-08 by mikaelnousiainen