Hamlib/Hamlib #443: Extending rotator control API

mikaelnousiainen, 2020-11-15:
I've developed a custom network-based rotator backend for use at our local club house. The controller is Arduino Due based and the code is open-source, available in GitHub: https://github.com/mikaelnousiainen/oh3aarot-controller

The controller offers precise speed control, which is currently not easily possible to implement with the Hamlib rotator API, as one can specify rotation speed only with the move command and not with set az/el commands at all. There's also no possibility to use the "previous/default" speed with the move command.

Before I make any changes, would the following improvements be accepted in the rotator API:

1. Add a `ROT_SPEED_NOCHANGE` (possibly with a value of `0`) for the `move` command to indicate that the previous or a default speed should be used. The speed range should anyway be 1-100 (though not all rotator backends obey this and I could fix it too), as documented in the man pages.

2. Add rotator levels with a standard level for speed, so that it can be controlled without rotation.

3. Add read-only indicators (possibly via funcs?) to provide better visibility in sensors like CW/CCW limit reached, or CW/CCW threshold passed (for over 360 degree rotation support).

The existing backends would most likely need minimal changes and the level/func commands could simply follow the implementaitons of their rigctl equivalents.

mikaelnousiainen, 2020-11-15:
Questions again mostly directed to @mdblack98 and/or @N0NB :)

mdblack98, 2020-11-15:
I would recommend using ROT_SPEED_NOCHANGE -10 might be used by some rotators to denote slowest or such.
A read only of status or error could be of use.
It's not really a FUNC....more of a LEVEL since those are read/write values so read-only fits there.
So ROT_LEVEL_STATUS with an enum list of possible values that it returns.
Mike W9MDB

mikaelnousiainen, 2020-11-15:
Right, well -- I think the range of the speed values should be well-defined and the backends should obey the definition. After all, there are only a handful of backends using the speed parameter for anything. Currently, as the speed is defined as 1-100, that's how it should be implemented in each backend -- and that's how I was planning to fix the backends to do it in the upcoming changes. If we wish to use zero as "slowest possible speed" then we should document it in the API. Using 0 as NOCHANGE value in the API would be logical as it is used in rigctl side already in the filters. It's ok to use -1 as well, but I think we should be strict in how the API is implemented and documented -- and being consistent in implementing similar behaviour is something I think would benefit Hamlib users too.

OTOH we could even add a new "status" function to the rot caps, similar to the `get_dcd` in rigctl, but one which could return several status bit flags according to an enum. These status bits would then be standardized, as it's common to have CW/CCW status indicators/sensors of different sorts in rotators.

mdblack98, 2020-11-15:
The reason to avoid 0 is that 0 is a valid numeric for speed -- it could have multiple meanings.  Also true of other CAT values too.I've been moving away from all the 0-values as it has caused numerous conflicts.
So we will be migrating away from 0 being some macro value for convenience sake....probably in hamlib 5.0 even though the pure API won't be broken...just those apps that use hard coded values.
Mike

mikaelnousiainen, 2020-11-15:
Ok, makes sense -- will use value `-1` for no change. Also, I was wrong about the consistency with rigctl, as RIG_FILTER_NOCHANGE is indeed defined as -1 too :)

mikaelnousiainen, 2020-11-24:
Just a note on progress, I'm working on this and will be pushing out a PR probably next week. The rotator move command will have the option not to change the speed and the rotator backends have been adjusted for this (+ for the speed scale of 1-100). Rotators will be getting a full set of funcs/levels/parms + the ext counterparts (yeah, backends don't use them yet, but many controllers offer advanced functions that have not been possible to implement without these!). It's mostly working already, but I'll need to add caps dump and documentation.

mikaelnousiainen, 2020-12-03:
Implemented in #453

closed 2020-12-03 by mikaelnousiainen