Hamlib/Hamlib #445: Hamlib error: IO error while opening connection to rig (Icom IC-705)

rmay001, 2020-11-21:
I see the "Hamlib error: IO error while opening connection to rig (Icom IC-705)" when attempting to transmit using WSJT-X (2.3.0 rc-2 and  the earlier version 2.2.2) using my Icom IC-705 (Dell laptop running Windows 10 with latest updates). I have no problem with receiving. I can transmit and receive using my Icom 7300 with WSJT-X on the same laptop. I have tried both the default A4h CIV address and the 94h CIV address on the IC-705 to emulate the IC-7300. My IC-705 rig setup matches all online examples. I have uninstalled and reinstalled all of the relevant Icom USB drivers. Any suggestions? Thank you.

mdblack98, 2020-11-21:
Would you please contact me off list?
Mike W9MDBmdblack98@yahoo.com

mdblack98, 2020-11-23:
This was due to RFI...no action to be taken.

closed 2020-11-23 by mdblack98