Hamlib/Hamlib #449: python installation

F4HTB, 2020-11-23:
Hello!

when i use my usual procedure it seems something is wrong:

```
cd /tmp
git clone https://github.com/Hamlib/Hamlib
cd Hamlib
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install
```

no error in installation and:

```
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib/python3.7/site-packages

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
```

When i run my software with this pythonpath complétion :

```
PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
```

I get:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

```

Maybe new issue?

F4HTB, 2020-11-23:
this has appeared in the last two weeks since i haven't updated from git since

F4HTB, 2020-11-23:
**./bootstrap**
```
./bootstrap
Running 'autoreconf -i' to process configure.ac
and generate the configure script.
aclocal: installing 'macros/libtool.m4' from '/usr/share/aclocal/libtool.m4'
aclocal: installing 'macros/ltoptions.m4' from '/usr/share/aclocal/ltoptions.m4'
aclocal: installing 'macros/ltsugar.m4' from '/usr/share/aclocal/ltsugar.m4'
aclocal: installing 'macros/ltversion.m4' from '/usr/share/aclocal/ltversion.m4'
aclocal: installing 'macros/lt~obsolete.m4' from '/usr/share/aclocal/lt~obsolete.m4'
aclocal: installing 'macros/pkg.m4' from '/usr/share/aclocal/pkg.m4'
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
configure.ac:99: installing 'build-aux/ar-lib'
configure.ac:30: installing 'build-aux/compile'
configure.ac:130: installing 'build-aux/config.guess'
configure.ac:130: installing 'build-aux/config.sub'
configure.ac:39: installing 'build-aux/install-sh'
configure.ac:39: installing 'build-aux/missing'
amplifiers/elecraft/Makefile.am: installing 'build-aux/depcomp'
bindings/Makefile.am:106: installing 'build-aux/py-compile'
parallel-tests: installing 'build-aux/test-driver'
```

F4HTB, 2020-11-23:
**./configure --with-python-binding PYTHON=$(which python3)**
```
./configure --with-python-binding PYTHON=$(which python3)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of gcc... gcc3
checking whether make supports nested variables... (cached) yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C preprocessor... gcc -E
checking for gawk... (cached) mawk
checking whether ln -s works... yes
checking for g++... yes
checking for ar... ar
checking the archiver (ar) interface... ar
checking for inline... inline
checking AM_CFLAGS for maximum warnings... -Wall
checking AM_CXXFLAGS for maximum warnings... -Wall
checking errno.h usability... yes
checking errno.h presence... yes
checking for errno.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking getopt.h usability... yes
checking getopt.h presence... yes
checking for getopt.h... yes
checking limits.h usability... yes
checking limits.h presence... yes
checking for limits.h... yes
[... 323 lines cut ...]

 Prefix         /usr/local
 Preprocessor   gcc -E
 C Compiler     gcc -g -O2
 C++ Compiler   g++ -g -O2

 Package features:
    With C++ binding                yes
    With Perl binding               no
    With Python binding             yes
    With TCL binding                no
    With Lua binding                no
    With rigmem XML support         no
    With Readline support           no
    With INDI support               no

    Enable HTML rig feature matrix  no
    Enable WinRadio                 yes
    Enable USRP                     no
    Enable USB backends             yes
    Enable shared libs              yes
    Enable static libs              yes

-----------------------------------------------------------------------
```

F4HTB, 2020-11-23:
**make**
```
Making all in macros
make[1]: Entering directory '/home/pi/Hamlib/macros'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/pi/Hamlib/macros'
Making all in include
make[1]: Entering directory '/home/pi/Hamlib/include'
make  all-am
make[2]: Entering directory '/home/pi/Hamlib/include'
make[2]: Leaving directory '/home/pi/Hamlib/include'
make[1]: Leaving directory '/home/pi/Hamlib/include'
Making all in lib
make[1]: Entering directory '/home/pi/Hamlib/lib'
  CC       termios.lo
  CC       dummy.lo
  CCLD     libmisc.la
make[1]: Leaving directory '/home/pi/Hamlib/lib'
Making all in rigs/adat
make[1]: Entering directory '/home/pi/Hamlib/rigs/adat'
  CC       adt_200a.lo
  CC       adat.lo
  CCLD     libhamlib-adat.la
make[1]: Leaving directory '/home/pi/Hamlib/rigs/adat'
Making all in rigs/alinco
make[1]: Entering directory '/home/pi/Hamlib/rigs/alinco'
  CC       dx77.lo
  CC       dxsr8.lo
  CC       alinco.lo
  CCLD     libhamlib-alinco.la
make[1]: Leaving directory '/home/pi/Hamlib/rigs/alinco'
Making all in rigs/aor
make[1]: Entering directory '/home/pi/Hamlib/rigs/aor'
  CC       ar8200.lo
  CC       ar8000.lo
  CC       ar5000.lo
  CC       ar3000.lo
  CC       ar7030.lo
  CC       ar3030.lo
  CC       ar2700.lo
  CC       ar8600.lo
  CC       ar7030p.lo
  CC       ar7030p_utils.lo
  CC       sr2200.lo
  CC       aor.lo
  CCLD     libhamlib-aor.la
make[1]: Leaving directory '/home/pi/Hamlib/rigs/aor'
Making all in rigs/barrett
make[1]: Entering directory '/home/pi/Hamlib/rigs/barrett'
  CC       barrett.lo
  CCLD     libhamlib-barrett.la
make[1]: Leaving directory '/home/pi/Hamlib/rigs/barrett'
Making all in rigs/dorji
make[1]: Entering directory '/home/pi/Hamlib/rigs/dorji'
  CC       dorji.lo
  CC       dra818.lo
  CCLD     libhamlib-dorji.la
make[1]: Leaving directory '/home/pi/Hamlib/rigs/dorji'
Making all in rigs/drake
make[1]: Entering directory '/home/pi/Hamlib/rigs/drake'
  CC       r8a.lo
  CC       r8b.lo
  CC       drake.lo
  CCLD     libhamlib-drake.la
make[1]: Leaving directory '/home/pi/Hamlib/rigs/drake'
Making all in rigs/dummy
make[1]: Entering directory '/home/pi/Hamlib/rigs/dummy'
  CC       dummy.lo
  CC       rot_dummy.lo
  CC       netrigctl.lo
[... 523 lines cut ...]
  CCLD     ampctld
make[2]: Leaving directory '/home/pi/Hamlib/tests'
make[1]: Leaving directory '/home/pi/Hamlib/tests'
Making all in doc
make[1]: Entering directory '/home/pi/Hamlib/doc'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/pi/Hamlib/doc'
make[1]: Entering directory '/home/pi/Hamlib'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/home/pi/Hamlib'

```

only warning:

```
hamlibpy_wrap.c: In function ‘SWIG_Python_addvarlink’:
hamlibpy_wrap.c:40928:9: warning: ‘strncpy’ specified bound depends on the length of the source argument [-Wstringop-overflow=]
         strncpy(gv->name,name,size);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
hamlibpy_wrap.c:40925:21: note: length computed here
       size_t size = strlen(name)+1;
                     ^~~~~~~~~~~~

```

F4HTB, 2020-11-23:
sudo make install

```
sudo make install
Making install in macros
make[1]: Entering directory '/home/pi/Hamlib/macros'
make[2]: Entering directory '/home/pi/Hamlib/macros'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/macros'
make[1]: Leaving directory '/home/pi/Hamlib/macros'
Making install in include
make[1]: Entering directory '/home/pi/Hamlib/include'
make[2]: Entering directory '/home/pi/Hamlib/include'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/include'
 /bin/mkdir -p '/usr/local/include/hamlib'
 /usr/bin/install -c -m 644  hamlib/rig.h hamlib/riglist.h hamlib/rig_dll.h hamlib/rotator.h hamlib/rotlist.h hamlib/rigclass.h hamlib/rotclass.h hamlib/amplifier.h hamlib/amplist.h hamlib/ampclass.h '/usr/local/include/hamlib'
make[2]: Leaving directory '/home/pi/Hamlib/include'
make[1]: Leaving directory '/home/pi/Hamlib/include'
Making install in lib
make[1]: Entering directory '/home/pi/Hamlib/lib'
make[2]: Entering directory '/home/pi/Hamlib/lib'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/lib'
make[1]: Leaving directory '/home/pi/Hamlib/lib'
Making install in rigs/adat
make[1]: Entering directory '/home/pi/Hamlib/rigs/adat'
make[2]: Entering directory '/home/pi/Hamlib/rigs/adat'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/rigs/adat'
make[1]: Leaving directory '/home/pi/Hamlib/rigs/adat'
Making install in rigs/alinco
make[1]: Entering directory '/home/pi/Hamlib/rigs/alinco'
make[2]: Entering directory '/home/pi/Hamlib/rigs/alinco'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/rigs/alinco'
make[1]: Leaving directory '/home/pi/Hamlib/rigs/alinco'
Making install in rigs/aor
make[1]: Entering directory '/home/pi/Hamlib/rigs/aor'
make[2]: Entering directory '/home/pi/Hamlib/rigs/aor'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/rigs/aor'
make[1]: Leaving directory '/home/pi/Hamlib/rigs/aor'
Making install in rigs/barrett
make[1]: Entering directory '/home/pi/Hamlib/rigs/barrett'
make[2]: Entering directory '/home/pi/Hamlib/rigs/barrett'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/rigs/barrett'
make[1]: Leaving directory '/home/pi/Hamlib/rigs/barrett'
Making install in rigs/dorji
make[1]: Entering directory '/home/pi/Hamlib/rigs/dorji'
make[2]: Entering directory '/home/pi/Hamlib/rigs/dorji'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/rigs/dorji'
make[1]: Leaving directory '/home/pi/Hamlib/rigs/dorji'
Making install in rigs/drake
make[1]: Entering directory '/home/pi/Hamlib/rigs/drake'
make[2]: Entering directory '/home/pi/Hamlib/rigs/drake'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/pi/Hamlib/rigs/drake'
make[1]: Leaving directory '/home/pi/Hamlib/rigs/drake'
Making install in rigs/dummy
[... 420 lines cut ...]
make[2]: Leaving directory '/home/pi/Hamlib/tests'
make[1]: Leaving directory '/home/pi/Hamlib/tests'
Making install in doc
make[1]: Entering directory '/home/pi/Hamlib/doc'
make[2]: Entering directory '/home/pi/Hamlib/doc'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/man/man1'
 /usr/bin/install -c -m 644 man1/ampctl.1 man1/ampctld.1 man1/rigctl.1 man1/rigctld.1 man1/rigmem.1 man1/rigsmtr.1 man1/rigswr.1 man1/rotctl.1 man1/rotctld.1 man1/rigctlcom.1 '/usr/local/share/man/man1'
 /bin/mkdir -p '/usr/local/share/man/man7'
 /usr/bin/install -c -m 644 man7/hamlib.7 man7/hamlib-primer.7 man7/hamlib-utilities.7 '/usr/local/share/man/man7'
make[2]: Leaving directory '/home/pi/Hamlib/doc'
make[1]: Leaving directory '/home/pi/Hamlib/doc'
make[1]: Entering directory '/home/pi/Hamlib'
make[2]: Entering directory '/home/pi/Hamlib'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/aclocal'
 /usr/bin/install -c -m 644 hamlib.m4 '/usr/local/share/aclocal'
 /bin/mkdir -p '/usr/local/share/doc/hamlib'
 /usr/bin/install -c -m 644 ChangeLog COPYING COPYING.LIB LICENSE README README.betatester README.developer '/usr/local/share/doc/hamlib'
 /bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 hamlib.pc '/usr/local/lib/pkgconfig'
make[2]: Leaving directory '/home/pi/Hamlib'
make[1]: Leaving directory '/home/pi/Hamlib'

```

F4HTB, 2020-11-23:
cd bindings/
make

```
make  all-am
make[1]: Entering directory '/home/pi/Hamlib/bindings'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/home/pi/Hamlib/bindings'

```

F4HTB, 2020-11-23:
**sudo make install**
```
make  install-am
make[1]: Entering directory '/home/pi/Hamlib/bindings'
make[2]: Entering directory '/home/pi/Hamlib/bindings'
 /bin/mkdir -p '/usr/local/lib/python3.7/site-packages'
 /usr/bin/install -c -m 644 Hamlib.py '/usr/local/lib/python3.7/site-packages'
-c:2: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Byte-compiling python modules...
Hamlib.py
Byte-compiling python modules (optimized versions) ...
Hamlib.py
 /bin/mkdir -p '/usr/local/lib/python3.7/site-packages'
 /bin/bash ../libtool   --mode=install /usr/bin/install -c   _Hamlib.la '/usr/local/lib/python3.7/site-packages'
libtool: warning: relinking '_Hamlib.la'
libtool: install: (cd /home/pi/Hamlib/bindings; /bin/bash "/home/pi/Hamlib/libtool"  --silent --tag CC --mode=relink gcc -fno-strict-aliasing -g -O2 -no-undefined -module -avoid-version -o _Hamlib.la -rpath /usr/local/lib/python3.7/site-packages hamlibpy_wrap.lo ../src/libhamlib.la -L/usr/lib -lpython3.7m )
libtool: install: /usr/bin/install -c .libs/_Hamlib.soT /usr/local/lib/python3.7/site-packages/_Hamlib.so
libtool: install: /usr/bin/install -c .libs/_Hamlib.lai /usr/local/lib/python3.7/site-packages/_Hamlib.la
libtool: install: /usr/bin/install -c .libs/_Hamlib.a /usr/local/lib/python3.7/site-packages/_Hamlib.a
libtool: install: chmod 644 /usr/local/lib/python3.7/site-packages/_Hamlib.a
libtool: install: ranlib /usr/local/lib/python3.7/site-packages/_Hamlib.a
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/local/lib/python3.7/site-packages
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib/python3.7/site-packages

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /bin/mkdir -p '/usr/local/share/doc/hamlib/examples'
 /usr/bin/install -c -m 644 py3test.py '/usr/local/share/doc/hamlib/examples'
make[2]: Leaving directory '/home/pi/Hamlib/bindings'
make[1]: Leaving directory '/home/pi/Hamlib/bindings'

```

F4HTB, 2020-11-23:
**PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH python3**

```
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> **import Hamlib**
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
**ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory**

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
**ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory**
>>>
```

F4HTB, 2020-11-23:
**ls -alR /usr/local/lib/python3.7/site-packages**

```
/usr/local/lib/python3.7/site-packages:
total 6040
drwxr-sr-x 3 root staff    4096 Nov 23 19:12 .
drwxrwsr-x 4 root staff    4096 Oct 12 16:17 ..
-rw-r--r-- 1 root staff 3345492 Nov 23 19:12 _Hamlib.a
-rwxr-xr-x 1 root staff    1016 Nov 23 19:12 _Hamlib.la
-rw-r--r-- 1 root staff  199672 Nov 23 19:12 Hamlib.py
-rwxr-xr-x 1 root staff 2619732 Nov 23 19:12 _Hamlib.so
drwxr-sr-x 2 root staff    4096 Nov 23 19:12 __pycache__

/usr/local/lib/python3.7/site-packages/__pycache__:
total 280
drwxr-sr-x 2 root staff   4096 Nov 23 19:12 .
drwxr-sr-x 3 root staff   4096 Nov 23 19:12 ..
-rw-r--r-- 1 root staff 139230 Nov 23 19:12 Hamlib.cpython-37.opt-1.pyc
-rw-r--r-- 1 root staff 139230 Nov 23 19:12 Hamlib.cpython-37.pyc

```

F4HTB, 2020-11-23:
**ls -alR Hamlib | grep libhamlib.so.4**

```
lrwxrwxrwx 1 pi pi       18 Nov 23 19:01 libhamlib.so -> libhamlib.so.4.0.1
lrwxrwxrwx 1 pi pi       18 Nov 23 19:01 libhamlib.so.4 -> libhamlib.so.4.0.1
-rwxr-xr-x 1 pi pi 14684420 Nov 23 19:01 libhamlib.so.4.0.1
```

**grep -rnw './Hamlib ' -e 'libhamlib.so.4'**
```
grep -rnw './' -e 'libhamlib.so.4'
Binary file ./tests/.libs/rigswr matches
Binary file ./tests/.libs/rigmem matches
Binary file ./tests/.libs/rigctl matches
Binary file ./tests/.libs/rotctl matches
Binary file ./tests/.libs/rigctld matches
Binary file ./tests/.libs/ampctl matches
Binary file ./tests/.libs/ampctld matches
Binary file ./tests/.libs/rigsmtr matches
Binary file ./tests/.libs/rigctlcom matches
Binary file ./tests/.libs/rotctld matches
Binary file ./bindings/.libs/_Hamlib.so matches
Binary file ./bindings/.libs/_Hamlib.soT matches
./src/libhamlib.la:8:dlname='libhamlib.so.4'
./src/libhamlib.la:11:library_names='libhamlib.so.4.0.1 libhamlib.so.4 libhamlib.so'
Binary file ./src/.libs/libhamlib.so.4.0.1 matches
./src/.libs/libhamlib.lai:8:dlname='libhamlib.so.4'
./src/.libs/libhamlib.lai:11:library_names='libhamlib.so.4.0.1 libhamlib.so.4 libhamlib.so'
Binary file ./c++/.libs/libhamlib++.so.4.0.1T matches
Binary file ./c++/.libs/libhamlib++.so.4.0.1 matches

```

F4HTB, 2020-11-23:
**find ./ -type f -path '*libhamlib.*'**

```
find ./ -type f -path '*libhamlib.*'
./src/libhamlib.la
./src/.libs/libhamlib.so.4.0.1
./src/.libs/libhamlib.a
./src/.libs/libhamlib.lai
```

F4HTB, 2020-11-23:
**ls /usr/local/lib/**

```
drwxr-xr-x  5 root root      4096 Nov 23 19:05 .
drwxr-xr-x 10 root root      4096 Aug 20 11:26 ..
-rw-r--r--  1 root root    278410 Nov 23 19:05 libhamlib++.a
-rw-r--r--  1 root root  20847866 Nov 23 19:05 libhamlib.a
-rwxr-xr-x  1 root root      1001 Nov 23 19:05 libhamlib++.la
-rwxr-xr-x  1 root root       963 Nov 23 19:05 libhamlib.la
lrwxrwxrwx  1 root root        20 Nov 23 19:05 libhamlib++.so -> libhamlib++.so.4.0.1
lrwxrwxrwx  1 root root        18 Nov 23 19:05 libhamlib.so -> libhamlib.so.4.0.1
lrwxrwxrwx  1 root root        20 Nov 23 19:05 libhamlib++.so.4 -> libhamlib++.so.4.0.1
lrwxrwxrwx  1 root root        18 Nov 23 19:05 libhamlib.so.4 -> libhamlib.so.4.0.1
-rwxr-xr-x  1 root root    195236 Nov 23 19:05 libhamlib++.so.4.0.1
-rwxr-xr-x  1 root root  14684420 Nov 23 19:05 libhamlib.so.4.0.1
drwxr-xr-x  2 root root      4096 Nov 23 19:05 pkgconfig
drwxrwsr-x  4 root staff     4096 Aug 20 11:32 python2.7
drwxrwsr-x  4 root staff     4096 Oct 12 16:17 python3.7

```

F4HTB, 2020-11-23:
**I have test some things and it's work:**

```
sudo apt-get install autoconf automake libtool swig
Reading package lists... Done
Building dependency tree
Reading state information... Done
autoconf is already the newest version (2.69-11).
automake is already the newest version (1:1.16.1-4).
automake set to manually installed.
libtool is already the newest version (2.4.6-9).
swig is already the newest version (3.0.12-2).
The following packages were automatically installed and are no longer required:
  javascript-common libhamlib2 libjs-jquery libjs-underscore
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

```
sudo apt autoremove
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  javascript-common libhamlib2 libjs-jquery libjs-underscore
0 upgraded, 0 newly installed, 4 to remove and 0 not upgraded.
After this operation, 3,944 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 43974 files and directories currently installed.)
Removing javascript-common (11) ...
Removing libhamlib2:armhf (3.3-5) ...
Removing libjs-jquery (3.3.1~dfsg-3) ...
Removing libjs-underscore (1.9.1~dfsg-1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
```

**sudo make uninstall and sudo make install in Hamlib root dir and binding and it's work yet.**

closed 2020-11-23 by F4HTB

mstumbo, 2020-12-14:
I installed hamlib using the steps you outlined at the top of the post and I also encountered the issue where rigtctl couldn't find libhamlib.so.4:
```
 ldd /usr/local/bin/rigctl
	linux-vdso.so.1 (0x00007fff403e6000)
	libhamlib.so.4 => not found
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f98082cc000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f98080da000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f980832b000)
```

I resolved it with a `sudo ldconfig`

```
ldd /usr/local/bin/rigctl
	linux-vdso.so.1 (0x00007fff7d9f3000)
	libhamlib.so.4 => /usr/local/lib/libhamlib.so.4 (0x00007fa86a67a000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa86a657000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa86a465000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fa86a45f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa86a310000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fa86afb5000)
```