Hamlib/Hamlib #451: rotator M2 RC2800 error

mdblack98, 2020-11-27:
Here are the logs for the M2 rotor 1001.  I tried “p” and “w a;”, the latter the native rotor command.

Thanks!

--Howard
hlnurse@cmmsft.com

 
pi@rigpi2:~ $ rotctl -m 1001 -r /dev/ttyUSB0 -s 9600 -vvvvv
rotctl Hamlib 4.1~git
Last commit was from indeterminate source revision.
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_m2 called
rot_register (1001)
set_conf: called
rot_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=9600,0x000d
serial_setup: cfsetospeed=9600,0x000d
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
serial_flush: tcflush
Opened rot model 1001, 'RC2800'
Backend version: 20200107.0, Status: Beta
 
Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
rc2800_rot_get_position called
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 2 bytes
0000    41 0d                                               A.             
read_string called, rxmax=32
read_string(): RX 18 characters
0000    41 20 50 3d 2d 33 35 39 2e 39 20 53 3d 39 20 53     A P=-359.9 S=9 S
0010    54 0a                                               T.             
read_string called, rxmax=32
read_string(): RX 1 characters
0000    0d                                                  .              

get_pos: error = Protocol error

Rotator command: w a;
rotctl_parse: input_line: w a;
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 3 bytes
0000    61 3b 0d                                            a;.            
read_string called, rxmax=128
read_string(): RX 29 characters
0000    41 20 45 72 72 3d 30 33 20 22 55 72 65 78 70 65     A Err=03 "Urexpe
0010    63 74 65 64 20 63 68 61 72 22 20 20 0a              cted char"  .  
Reply: A Err=03 "Urexpected char" 
 read_string called, rxmax=128
read_string(): RX 1 characters
0000    0d                                                  .              
 read_string called, rxmax=128
0000    41 20 3d                                            A =            
read_string(): read() failed - Resource temporarily unavailable
send_cmd: error = IO error
 
Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
rc2800_rot_get_position called
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 2 bytes
0000    41 0d                                               A.             
read_string called, rxmax=32
read_string(): RX 18 characters
0000    41 20 50 3d 2d 33 35 39 2e 39 20 53 3d 39 20 53     A P=-359.9 S=9 S
0010    54 0a                                               T.             
read_string called, rxmax=32
0000    0d                                                  .               
read_string(): read() failed - Resource temporarily unavailable
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 2 bytes
0000    41 0d                                               A.              
read_string called, rxmax=32
0000    20 50 3d 2d                                          P=-           
read_string(): read() failed - Resource temporarily unavailable
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 2 bytes
0000    41 0d                                               A.             
read_string called, rxmax=32
0000    0d 41 20 50 3d 2d                                   .A P=-         
read_string(): read() failed - Resource temporarily unavailable
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 2 bytes
0000    41 0d                                               A.             
read_string called, rxmax=32
read_string(): RX 4 characters
0000    53 20 53 0a                                         S S.           
read_string called, rxmax=32
read_string(): read() failed - Resource temporarily unavailable
get_pos: error = IO error

mdblack98, 2020-11-27:
Not there yet.

 

--h

 

pi@rigpi2:~/Downloads/Hamlib-master $ rotctl -m 1001 -r /dev/ttyUSB0 -s 9600 -vvvvv

rotctl Hamlib 4.1~git

Last commit was from indeterminate source revision.

Report bugs to <hamlib-developer@lists.sourceforge.net>

 

rot_init called

initrots4_m2 called

rot_register (1001)

set_conf: called

rot_open called

serial_open called

serial_setup called

serial_setup: tcgetattr

serial_setup: cfmakeraw

serial_setup: cfsetispeed=9600,0x000d

serial_setup: cfsetospeed=9600,0x000d

serial_setup: data_bits=8

serial_setup: parity=0

serial_setup: tcsetattr TCSANOW

serial_flush called

tcflush

Opened rot model 1001, 'RC2800'

Backend version: 20201127.0, Status: Beta

 

Rotator command: p

rotctl_parse: input_line: p

rot_get_position called

rc2800_rot_get_position called

rig_flush: called for serial device

serial_flush called

tcflush

write_block called

write_block(): TX 2 bytes

0000    41 0d                                               A.             

read_string called, rxmax=32

read_string(): RX 7 characters

0000    41 20 50 3d 20 31 39                                A P= 19        

read_string called, rxmax=32

read_string(): Timed out 1.1082 seconds after 0 chars

rig_flush: called for serial device

serial_flush called

tcflush

write_block called

write_block(): TX 2 bytes

0000    41 0d                                               A.             

read_string called, rxmax=32

0000    41 20 50 3d 20                                      A P=           

read_string(): read() failed - Resource temporarily unavailable

rig_flush: called for serial device

serial_flush called

tcflush

write_block called

write_block(): TX 2 bytes

0000    41 0d                                               A.             

read_string called, rxmax=32

read_string(): RX 4 characters

0000    39 37 2e 33                                         97.3           

read_string called, rxmax=32

read_string(): Timed out 1.1076 seconds after 0 chars

rig_flush: called for serial device

serial_flush called

tcflush

write_block called

write_block(): TX 2 bytes

0000    41 0d                                               A.             

read_string called, rxmax=32

read_string(): RX 19 characters

0000    41 20 50 3d 20 31 39 37 2e 33 20 53 3d 35 20 53     A P= 197.3 S=5 S

0010    54 0a 0d                                            T..            

read_string called, rxmax=32

0000    41 20                                               A              

read_string(): read() failed - Resource temporarily unavailable

get_pos: error = IO error

 

Rotator command: w a

rotctl_parse: input_line: w a

rig_flush: called for serial device

serial_flush called

tcflush

write_block called

write_block(): TX 2 bytes

0000    61 0d                                               a.             

read_string called, rxmax=128

read_string(): RX 18 characters

0000    41 20 50 3d 20 31 39 37 2e 33 20 53 3d 35 20 53     A P= 197.3 S=5 S

0010    54 0a                                               T.             

Reply: A P= 197.3 S=5 ST

 

read_string called, rxmax=128

read_string(): RX 1 characters

0000    0d                                                  .              

 

read_string called, rxmax=128

read_string(): RX 5 characters

0000    35 20 53 54 0a                                      5 ST.          

5 ST

 

read_string called, rxmax=128

read_string(): RX 1 characters

0000    0d                                                  .              

 

read_string called, rxmax=128

read_string(): Timed out 1.1086 seconds after 0 chars

mdblack98, 2020-11-30:
Still not working
Rotator command: P
2020-11-30:00:48:56.564254: rotctl_parse: input_line: P
Azimuth: 140
Elevation: 0
2020-11-30:00:49:08.082946: rot_set_position called az=140.00 el=0.00
2020-11-30:00:49:08.083004: rot_set_position: south_zero=0
2020-11-30:00:49:08.083053: rc2800_rot_set_position called: 140.000000 0.000000
2020-11-30:00:49:08.083107: rig_flush: called for serial device
2020-11-30:00:49:08.083151: serial_flush called
2020-11-30:00:49:08.083186: tcflush
2020-11-30:00:49:08.083232: write_block called
2020-11-30:00:49:08.083330: write_block(): TX 5 bytes
2020-11-30:00:49:08.083374: 0000    41 31 34 30 0d                                      A140.          
2020-11-30:00:49:08.083419: read_string called, rxmax=128
2020-11-30:00:49:08.133454: read_string(): RX 30 characters
2020-11-30:00:49:08.133538: 0000    41 20 45 72 72 3d 30 34 20 22 55 6e 65 78 70 65     A Err=04 "Unexpe
2020-11-30:00:49:08.133592: 0010    63 74 65 64 20 62 79 74 65 22 20 20 0a 0d           cted byte"  .. a

skypodolsky, 2021-01-29:
@mdblack98 @N0NB sorry for raising this topic again, but could someone please clarify why did you comment on the echo suppression? Indeed, I see that manual says 'no echoes are returned'. But the reality for my M2 RC2800 is that echo _actually exists_ by default. Any thoughts on that?

mdblack98, 2021-01-29:
This debug info here shows no echo.If you run this same command you see "A20" being echoed?

Rotator command: w A20

2020-11-29:00:49:57.219816: rotctl_parse: input_line: w A20

2020-11-29:00:49:57.219941: rig_flush: called for serial device

2020-11-29:00:49:57.219995: serial_flush called

2020-11-29:00:49:57.220043: tcflush

2020-11-29:00:49:57.220102: write_block called

2020-11-29:00:49:57.220196: write_block(): TX 4 bytes

2020-11-29:00:49:57.220246: 0000    41 32 30 0d                                         A20.           

2020-11-29:00:49:57.220295: read_string called, rxmax=128

2020-11-29:00:49:57.251563: read_string(): RX 18 characters

2020-11-29:00:49:57.251660: 0000    41 20 50 3d 20 20 20 30 2e 31 20 53 3d 39 20 53     A P=   0.1 S=9 S

2020-11-29:00:49:57.251705: 0010    54 0a                                               T.

skypodolsky, 2021-01-29:
Suspicious. Here's my log:

```
rotctl Hamlib 4.1~git
Last commit was Mon Jan 25 18:23:05 2021 +0000 SHA=54fc0e
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_m2 called
rot_register (1001)
rot_register (1002)
rot_register (1003)
set_conf: called
rot_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=9600,0x000d
serial_setup: cfsetospeed=9600,0x000d
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
tcflush
Opened rot model 1003, 'RC2800_EARLY_AZEL'
Backend version: 20201130, Status: Stable

Rotator command: w A90
Cmd: rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 4 bytes
0000    41 39 30 0d                                         A90.
read_string called, rxmax=128
read_string(): RX 4 characters
0000    41 39 30 0d                                         A90.
Reply: A90
read_string called, rxmax=128
read_string(): RX 1 characters
0000    0a                                                  .

read_string called, rxmax=128
read_string(): RX 13 characters
0000    41 3d 38 39 2e 36 20 53 3d 39 20 53 0d              A=89.6 S=9 S.
A=89.6 S=9 S
read_string called, rxmax=128
read_string(): RX 1 characters
0000    0a
```

And it works because I've uncommented a workaround of the echo handling. Checked with all three protocols (1001, 1002, 1003)

closed 2021-02-03 by mdblack98