Hamlib/Hamlib #452: Comm failure, TS-870S

ehidle, 2020-11-29:
I recently lost the ability to control my TS-870 after an ubuntu update (which updated hamlib to 3.1). I observed that I could talk to the rig with minicom and it would respond to CAT commands. But, fldigi and flrig would not communicate with the radio. I invoked rigcat from the command line and kept getting a Protocol Error message:

ehidle@**redacted**:~/software/hamlib-3.3$ rigctl -r /dev/ttyUSB2 -m 210 -D 19200
rig_open: error = Protocol error 

One other thing I noticed was that sometimes when I started minicom, I would see a string of commands load up the terminal as soon as it was opened, as if hamlib had tried to send them to the serial port but for some reason the serial port was not accepting them. That doesn't happen every time but I noticed it once or twice while trying to debug this.

Hamlib is working fine for all of my other radios (TS-590S, TS-590SG, TS-790A, TS-950SDX).

mdblack98, 2020-11-29:
Please upgrade to the current hamlib.  3.3 is no longer receiving patches.
https://github.com/Hamlib/Hamlib

Mike W9MDB

ehidle, 2020-11-29:
> Please upgrade to the current hamlib.  3.3 is no longer receiving patches. https://github.com/Hamlib/Hamlib Mike W9MDB 

Done. Pulled from git and rebuilt, and the problem persists:

ehidle@**redacted**:~/software/Hamlib$ rigctl -r /dev/ttyUSB2 -m 2010 -s 19200
rig_open: error = Protocol error

mdblack98, 2020-11-29:
rigctl -r /dev/ttyUSB2 -m 2010 -s 19200 -vvvvv -Z >&debug.txt

And send me the debug.txt
Mike

ehidle, 2020-11-29:
Ooohhhhh this is interesting. I figured it out: 

ehidle@**redacted**:~$ rigctl -m 2010 -r /dev/ttyUSB2 -s 19200 -vvvvv
rigctl Hamlib 4.1~git
Last commit was Sun Nov 29 04:34:15 2020 +0000 SHA=d782fd
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
[... 130 lines cut ...]
I: [11:34:59] rigcontrol/rigsupport.cxx : 251 : readFreqList
    18100000 USB 1000 BPSK31 
I: [11:34:59] rigcontrol/rigsupport.cxx : 251 : readFreqList
    21005000 USB 800 CW 
I: [11:34:59] rigcontrol/rigsupport.cxx : 251 : readFreqList
    21070000 USB 1000 BPSK31 
I: [11:34:59] rigcontrol/rigsupport.cxx : 251 : readFreqList
    24920000 USB 1000 BPSK31 
I: [11:34:59] rigcontrol/rigsupport.cxx : 251 : readFreqList
    28005000 USB 800 CW 
I: [11:34:59] rigcontrol/rigsupport.cxx : 251 : readFreqList
    28120000 USB 1000 BPSK31 
V: [11:34:59] rigcontrol/ptt.cxx : 132 : set
    PTT via NONE : OFF
D: [11:34:59] rigcontrol/serial.cxx : 213 : SetPTT
    PTT fd < 0
V: [11:34:59] rigcontrol/ptt.cxx : 132 : set
    PTT via HAMLIB : OFF
ehidle@**redacted**:~/.fldigi$ rigctl -r /dev/ttyUSB2 -m 2010 -s 19200
rig_open: error = Protocol error 
ehidle@**redacted**:~/.fldigi$ rigctl -r /dev/ttyUSB2 -m 2010 -s 19200

Rig command: 

Thanks for the prodding. At least I got hamlib updated to git out of it :)

mdblack98, 2020-11-29:
So nothing to fix.

closed 2020-11-29 by mdblack98