Hamlib/Hamlib #454: TS-2000 VFO and Timeout issues

mld4p, 2020-12-04:
I have been trying to setup our clubs TS-2000 for satellite working using GPredict and hamlib, more specifically rigctld. I am currently working with Hamlib 4.1~git, and unfortunately it not only times out, but when using other programs (such as flrig, due to limitations I have not been able to use fldigi) the sub vfo (or vfoB) cannot be accessed. Below is the debug codes acquired from the cli. I can't make any sense of it and have tried all the other suggestions I have been able to find on it.

```
rigctl -m 2014 -s 9600 -r COM1 -c serial_handshake=Hardware -vvvvv

rigctl Hamlib 4.1~git
Last commit was Thu Dec 03 05:47:30 2020 +0000 SHA=c4e1de
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
rig_register: rig_register (2042)
rig_register called
rig_register: rig_register (2020)
rig_register called
rig_register: rig_register (2023)
rig_register called
rig_register: rig_register (2027)
rig_register called
rig_register: rig_register (2034)
rig_register called
rig_register: rig_register (2031)
rig_register called
rig_register: rig_register (2039)
rig_register called
rig_register: rig_register (2037)
rig_register called
rig_register: rig_register (2028)
rig_register called
rig_register: rig_register (2019)
rig_register called
rig_register: rig_register (2032)
rig_register called
rig_register: rig_register (2036)
rig_register called
rig_register: rig_register (2048)
rig_register called
rig_register: rig_register (2040)
rig_register called
rig_register: rig_register (2041)
rig_register called
rig_register: rig_register (2046)
kenwood_init called, version 20201126/20201126.0
kenwood_init: if_len = 37
rig_token_lookup called
rig_confparam_lookup called
rig_set_conf called
rig_confparam_lookup called
rig_open called
port_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfsetispeed=9600,0x000d
serial_setup: cfsetospeed=9600,0x000d
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
tcflush
kenwood_open called
kenwood_get_id called
kenwood_transaction called
kenwood_transaction: cmdstr = ID
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string called, rxmax=128
read_string(): Timed out 0.318851 seconds after 0 chars
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: returning retval=-5
kenwood_open: no response to get_id from rig...continuing anyways.
kenwood_open: rig has_rit2
kenwood_transaction called
kenwood_transaction: cmdstr = FA
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 3 bytes
0000    46 41 3b                                            FA;
read_string called, rxmax=128
read_string(): Timed out 0.316963 seconds after 0 chars
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: returning retval=-5
kenwood_open: no response from rig
rig_open: error = Communication timed out

```

mdblack98, 2020-12-04:
Turn off the handshake -- TS-2000's don't use hardware handshake
Mike W9MDB

mld4p, 2020-12-04:
Still no dice, even increased the baud rate on the radio and computer.
```

rigctl -m 2014 -s 57600 -r COM1 -vvvvv
rigctl Hamlib 4.1~git
Last commit was Thu Dec 03 05:47:30 2020 +0000 SHA=c4e1de
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_kenwood called
rig_register called
rig_register: rig_register (2012)
rig_register called
rig_register: rig_register (2013)
rig_register called
rig_register: rig_register (2001)
rig_register called
rig_register: rig_register (2025)
rig_register called
rig_register: rig_register (2003)
rig_register called
rig_register: rig_register (2004)
rig_register called
rig_register: rig_register (2016)
rig_register called
rig_register: rig_register (2024)
rig_register called
rig_register: rig_register (2005)
rig_register called
rig_register: rig_register (2007)
rig_register called
rig_register: rig_register (2009)
rig_register called
rig_register: rig_register (2010)
rig_register called
rig_register: rig_register (2022)
rig_register called
rig_register: rig_register (2014)
rig_register called
rig_register: rig_register (2030)
rig_register called
rig_register: rig_register (2021)
rig_register called
rig_register: rig_register (2029)
rig_register called
rig_register: rig_register (2043)
rig_register called
rig_register: rig_register (2044)
rig_register called
rig_register: rig_register (2045)
rig_register called
rig_register: rig_register (2047)
rig_register called
rig_register: rig_register (2038)
rig_register called
rig_register: rig_register (2002)
rig_register called
rig_register: rig_register (2011)
rig_register called
rig_register: rig_register (2006)
rig_register called
rig_register: rig_register (2008)
rig_register called
rig_register: rig_register (2015)
rig_register called
rig_register: rig_register (2026)
rig_register called
rig_register: rig_register (2017)
rig_register called
rig_register: rig_register (2033)
rig_register called
rig_register: rig_register (2042)
rig_register called
rig_register: rig_register (2020)
rig_register called
rig_register: rig_register (2023)
rig_register called
rig_register: rig_register (2027)
rig_register called
rig_register: rig_register (2034)
rig_register called
rig_register: rig_register (2031)
rig_register called
rig_register: rig_register (2039)
rig_register called
rig_register: rig_register (2037)
rig_register called
rig_register: rig_register (2028)
rig_register called
rig_register: rig_register (2019)
rig_register called
rig_register: rig_register (2032)
rig_register called
rig_register: rig_register (2036)
rig_register called
rig_register: rig_register (2048)
rig_register called
rig_register: rig_register (2040)
rig_register called
rig_register: rig_register (2041)
rig_register called
rig_register: rig_register (2046)
kenwood_init called, version 20201126/20201126.0
kenwood_init: if_len = 37
rig_open called
port_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfsetispeed=57600,0x1001
serial_setup: cfsetospeed=57600,0x1001
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
tcflush
kenwood_open called
kenwood_get_id called
kenwood_transaction called
kenwood_transaction: cmdstr = ID
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 3 bytes
0000    49 44 3b                                            ID;
read_string called, rxmax=128
read_string(): Timed out 0.317729 seconds after 0 chars
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: returning retval=-5
kenwood_open: no response to get_id from rig...continuing anyways.
kenwood_open: rig has_rit2
kenwood_transaction called
kenwood_transaction: cmdstr = FA
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 3 bytes
0000    46 41 3b                                            FA;
read_string called, rxmax=128
read_string(): Timed out 0.316970 seconds after 0 chars
kenwood_transaction: read_string(len=0)=''
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=0, rs->rigport.retry=0
kenwood_transaction: returning retval=-5
kenwood_open: no response from rig
rig_open: error = Communication timed out
```

mdblack98, 2020-12-04:
So the rig isn't responding at all.
I suspect you have a serial cable that requires DTR or RTS to wake up.
--set-conf=rts_state=ON
or--set-conf=dtr_state=ON

Mike W9MDB

mld4p, 2020-12-04:
--set-conf=rts_state=ON did it. VFO interfacing via gpredict requires the rig to NOT be placed in Sat mode. Something to do with how it puts it into the MEM VFO and just screws it all up. This issue is fixed.

closed 2020-12-04 by mld4p