Hamlib/Hamlib #458: `bindings/luatest.lua` returns error

N0NB, 2020-12-06:
I'm seeing this when running the installed `luatest.lua` script:

```
$ lua /usr/local/share/doc/hamlib/examples/luatest.lua 
Lua 5.2 test, Hamlib 4.0~rc3

status(str):		Command completed successfully
get_conf:		path = /dev/Rig, retry = 5
freq:			5700000000
mode:			FM
bandwidth:		15000
mode:			CW
bandwidth:		15000
Backend copyright:	LGPL
Model:			Dummy
Manufacturer:		Hamlib
Backend version:	20200606.0
Backend license:	LGPL
Rig info:		Nothing much (dummy)
status:			0 - Command completed successfully
VOX delay:		1
status:			0 - Command completed successfully
VOX delay:		5
Setting AF to 12.340000...
status:			0 - Command completed successfully
AF level:		12.340000
strength:		-38
status:			0
status(str):		Command completed successfully
get_channel status:	0
VFO:			VFOB, 145000000
Attenuators:		10, 20, 30, 0, 0, 0, 0, 0

Sending Morse, '73'

Some static functions:
Loc1:		IN98XC ->   -0.0417,   48.1042 -> IN98XC
Loc2:		DM33DX -> -113.7083,   33.9792 -> DM33DX
Distance:	8765.814 km, azimuth 309.00, long path:	31266.186 km
Longitude:	-0.0417,    0° 2' 30" W	recoded:   -0.0417
Latitude:	48.1042,   48° 6' 15" N	recoded:   48.1042
The next two lines should show 0x8000000000000000
lua: /usr/local/share/doc/hamlib/examples/luatest.lua:111: bad argument #2 to 'format' (not a non-negative number in proper range)
stack traceback:
	[C]: in function 'format'
	/usr/local/share/doc/hamlib/examples/luatest.lua:111: in function 'doStartup'
	/usr/local/share/doc/hamlib/examples/luatest.lua:116: in main chunk
	[C]: in ?
```
Hopefully, someone familiar with Lua can take a look.  This is on a Debian Testing system with Lua 5.2.

mdblack98, 2020-12-06:
What version of lua?   There was a problem with 64-bit values in an older version.Maybe need a version check in the test script?
Lua 5.4.0 here....

Some static functions:Loc1: IN98XC ->   -0.0417,   48.1042 -> IN98XCLoc2: DM33DX -> -113.7083,   33.9792 -> DM33DXDistance: 8765.814 km, azimuth 309.00, long path: 31266.186 kmLongitude: -0.0417,    0° 2' 30" W recoded:   -0.0417Latitude: 48.1042,   48° 6' 15" N recoded:   48.1042The next two lines should show 0x8000000000000000RIG_MODE_TESTS_MAX: 0x8000000000000000RIG_FUNC_BIT63: 0x8000000000000000
Mike W9MDB

N0NB, 2020-12-06:
Seems like 5.2 is the "default" in Debian Testing, but perhaps not.  I see that 5.3 and 5.4 are also available.  Each version has a list of packages dependent on it though 5.4 has the shortest list.

If the script is not compatible with anything older than 5.4, then a version test is probably warranted.

N0NB, 2020-12-07:
You patch has quelled the error with Lua 5.2 on Debian Testing.  Thanks, Mike.

closed 2020-12-07 by N0NB