Hamlib/Hamlib #466: Prosistel D backend not working

mikaelnousiainen, 2020-12-14:
@mdblack98 I'm wondering what hardware this particular commit is based on:

https://github.com/Hamlib/Hamlib/commit/5a6a58f4783a5e59576a56cd36081d8afdb4137f

I've got a Prosistel D controller box for a PST61D rotator available for testing and the official documentation from Prosistel, which both indicate that the angles are degrees (not 1/10ths of degrees). Additionally, it is an error to query for elevation on an azimuth-only rotator controller.

I can fix both of these issues and make separate models for az / el / az+el rotators, which use the correct commands.

However, I wanted first to confirm if there is some other version / model of the "D box" controller, which actually works the way the backend is now implemented?

mdblack98, 2020-12-14:
Was on  PST2051D-PRO + PST2051E COMBO
Mike

 
 

5a6a58f

mdblack98, 2020-12-14:
I added some info to that commit that showed the response the patch was based on.
It does return "tenths" but they aren't used in the code.  Are you getting wrong answers?
What happens if you ask for elevation?  What does it return?  I'd rather allow for that then make a new entry unless we can't figure it out.
Mike

mikaelnousiainen, 2020-12-14:
@mdblack98 This is an example of trying to get position on the PST61D controller.

* Angles are returned as degrees (as documented in the protocol specs by Prosistel)
* Asking `B?` returns an error (also as documented)

I think we are dealing with two slightly different versions of the Prosistel protocol here and should make PST2051 az-el combo a separate model from the azimuth-only ones (PST 641/2051/61/71) ... Based on photos on the Prosistel website, the az-el combo controller also looks different from the az-only or el-only controllers.

Also, the document that I got from Prosistel states the following about firmware versions:

```
Controller CBOX 2003
A Control Box that can be fully operated via RS232.
(Firmware versions 3.22)
```

Maybe the newer combo controllers have a newer firmware with an update protocol?

Debug log:
```
# rotctl -vvvvv --model=1701 --serial-speed=9600 --rot-file=/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_D-if00-port0
rotctl Hamlib 4.1~git
Last commit was Tue Oct 06 14:27:23 2020 +0000 SHA=2e1a4e
Report bugs to <hamlib-developer@lists.sourceforge.net>

rot_init called
initrots4_prosistel: _init called
rot_register (1701)
set_conf: called
rot_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=9600,0x000d
serial_setup: cfsetospeed=9600,0x000d
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
serial_flush: tcflush
prosistel_rot_open called
rot_open called
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 4 bytes
0000    02 41 53 0d                                         .AS.
read_string called, rxmax=20
read_string(): RX 7 characters
0000    02 41 2c 53 2c 52 0d                                .A,S,R.
prosistel_transaction Command S reply received
Opened rot model 1701, 'Prosistel D'
Backend version: 20191219.0, Status: Stable

Rotator command: p
rotctl_parse: input_line: p
rot_get_position called
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 4 bytes
0000    02 41 3f 0d                                         .A?.
read_string called, rxmax=20
read_string(): RX 11 characters
0000    02 41 2c 3f 2c 32 39 30 2c 52 0d                    .A,?,290,R.
prosistel_transaction Command ? reply received
' converted to 29.000000on got position from 'A,?,290,R
rig_flush: called for serial device
serial_flush called
serial_flush: tcflush
write_block called
write_block(): TX 4 bytes
0000    02 42 3f 0d                                         .B?.
read_string called, rxmax=20
read_string(): RX 13 characters
0000    02 42 2c 3f 2c 45 2c 30 30 30 30 33 0d              .B,?,E,00003.
prosistel_transaction Command ? reply received
'rosistel_rot_get_position failed to parse elevation 'B,?,E,00003
get_pos: error = Protocol error
```

mikaelnousiainen, 2020-12-15:
I contacted Prosistel and they actually have all the protocol documentation available here: https://www.prosistel.it/download/download.htm

The documentation revealed that there are three "product families", namely the "D" series, the PRO series and the Combi-Track (az+el) combos.

Of these three the D and the PRO series seem to use the same protocol with angle in degrees and A/E identifiers for azimuth and elevation. These rotator controllers support either azimuth or elevation, but not both.

The Combi-Track controllers are combined azimuth+elevation controllers and these represent angles in 1/10ths of degrees *and* use A/B identifiers for unit A and B (which can be thought of as azimuth and elevation).

I've rewritten the backends for Prosistel allowing user to choose between the az, el or az+er models. I'll be able to test the azimuth-only model, so the rest need further testing. I also added support for the stop command.

Will push out a PR shortly.

closed 2020-12-15 by mdblack98

mikaelnousiainen, 2020-12-15:
Implemented in #468