Hamlib/Hamlib #471: Icom packet + filter combinations mode not working correctly

mikaelnousiainen, 2020-12-19:
I noticed the following behavior in the laster master for Icom backends (tested on IC-7300 and IC-7600):

* `\set_mode` (or `M`) with PKTUSB and PKTLSB modes will result into regular USB and LSB modes if the filter bandwidth is exatly 2400 Hz  (resulting into FIL2 being selected in Icom).
* If the filter bandwidth is not 2400, FIL1 or FIL3 is selected and PKTUSB / PKTLSB mode applied correctly
* If the bandwidth is `0` (default) or `-1` (no change), a regular USB or LSB mode is also selected, even though bandwidth selection may still be FIL1

In addition to that, choices leading to FIL2 / default filter being selected result in "Command rejected by rig" error, so there's something going wrong with the filter and/or data mode selection.

Error when setting filter width 2400:
```
Rig command: M PKTUSB 2400
rigctl_parse: input_line: M PKTUSB 2400
rigctl_parse: vfo_opt=0
rig_parse_mode called
rig_set_mode called
icom_set_mode_with_data called
icom_set_mode_with_data mode=4, width=2400
icom_set_mode called vfo=currVFO, mode=USB, width=2400
rig_passband_normal called
icom_set_mode: icmode=1, icmode_ext=-1
icom_set_mode: #2 icmode=1, icmode_ext=-1
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 94 e0 06 01 fd                                .......         
read_string called, rxmax=80
read_string(): RX 7 characters
0000    fe fe 94 e0 06 01 fd                                .......         
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe e0 94 fb fd                                   ......          
rig_passband_normal called
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 9 bytes
0000    fe fe 94 e0 1a 06 01 ff fd                          .........       
read_string called, rxmax=80
read_string(): RX 9 characters
0000    fe fe 94 e0 1a 06 01 ff fd                          .........       
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe e0 94 fa fd                                   ......          
icom_transaction: failed: Command rejected by the rig
icom_set_mode_with_data: protocol error (00), len=80
rig_set_mode: retcode after set_mode=-9
elapsed_ms: start = 0,0
elapsed_ms: after gettime, start = 1608370508,883801760
rigctl_parse: vfo_opt=0
set_mode: error = Command rejected by the rig
rigctl_parse: retcode=-9
rigctl_parse: called, interactive=1
```

Error when selecting default filter:
```
Rig command: M PKTUSB 0
rigctl_parse: input_line: M PKTUSB 0
rigctl_parse: vfo_opt=0
rig_parse_mode called
rig_set_mode called
icom_set_mode_with_data called
icom_set_mode_with_data mode=4, width=0
icom_set_mode called vfo=currVFO, mode=USB, width=0
rig_passband_normal called
icom_set_mode: icmode=1, icmode_ext=-1
icom_set_mode: #2 icmode=1, icmode_ext=-1
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 94 e0 06 01 fd                                .......         
read_string called, rxmax=80
read_string(): RX 7 characters
0000    fe fe 94 e0 06 01 fd                                .......         
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe e0 94 fb fd                                   ......          
rig_passband_normal called
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 9 bytes
0000    fe fe 94 e0 1a 06 01 ff fd                          .........       
read_string called, rxmax=80
read_string(): RX 9 characters
0000    fe fe 94 e0 1a 06 01 ff fd                          .........       
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe e0 94 fa fd                                   ......          
icom_transaction: failed: Command rejected by the rig
icom_set_mode_with_data: protocol error (00), len=80
rig_set_mode: retcode after set_mode=-9
elapsed_ms: start = 0,0
elapsed_ms: after gettime, start = 1608370585,29131008
rigctl_parse: vfo_opt=0
set_mode: error = Command rejected by the rig
rigctl_parse: retcode=-9
rigctl_parse: called, interactive=1
```

No error when FIL1 selected before and using -1 for no change in filter, but this leads to regular USB mode:
```
Rig command: M PKTUSB -1
rigctl_parse: input_line: M PKTUSB -1
rigctl_parse: vfo_opt=0
rig_parse_mode called
rig_set_mode called
icom_set_mode_with_data called
icom_set_mode_with_data mode=4, width=-1
icom_set_mode called vfo=currVFO, mode=USB, width=-1
rig_get_mode called
elapsed_ms: start = 1608370601,444920664
elapsed_ms: elapsed_msecs=3904
rig_get_mode: cache check age=3904ms
rig_get_mode: cache miss age=3904ms
icom_get_mode_with_data called
icom_get_mode called vfo=currVFO
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 6 bytes
0000    fe fe 94 e0 04 fd                                   ......          
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe 94 e0 04 fd                                   ......          
read_string called, rxmax=80
read_string(): RX 8 characters
0000    fe fe e0 94 04 01 01 fd                             ........        
icom_get_mode: modebuf[0]=0x04, modebuf[1]=0x01, modebuf[2]=0x01, mode_len=3, filter=1
icom2rig_mode: mode=0x01, pd=1
rig_passband_wide called
icom_get_dsp_flt called
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 94 e0 1a 03 fd                                .......         
read_string called, rxmax=80
read_string(): RX 7 characters
0000    fe fe 94 e0 1a 03 fd                                .......         
read_string called, rxmax=80
read_string(): RX 8 characters
0000    fe fe e0 94 1a 03 34 fd                             ......4.        
from_bcd called
icom_get_mode: vfo=currVFO returning mode=USB, width=3000
icom_get_mode_with_data mode=4
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 94 e0 1a 06 fd                                .......         
read_string called, rxmax=80
read_string(): RX 7 characters
0000    fe fe 94 e0 1a 06 fd                                .......         
read_string called, rxmax=80
read_string(): RX 9 characters
0000    fe fe e0 94 1a 06 01 01 fd                          .........       
icom_get_mode_with_data databuf[2]=1, mode=4
rig_get_mode: retcode after get_mode=0
rig_get_mode(1894): debug
elapsed_ms: start = 0,0
elapsed_ms: after gettime, start = 1608370605,408980112
icom_set_mode: icmode=1, icmode_ext=1
icom_set_mode: #2 icmode=1, icmode_ext=1
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 8 bytes
0000    fe fe 94 e0 06 01 01 fd                             ........        
read_string called, rxmax=80
read_string(): RX 8 characters
0000    fe fe 94 e0 06 01 01 fd                             ........        
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe e0 94 fb fd                                   ......          
icom_set_mode_with_data RIG_PASSBAND_NOCHANGE
rig_set_mode: retcode after set_mode=0
elapsed_ms: start = 0,0
elapsed_ms: after gettime, start = 1608370605,428907723
rigctl_parse: vfo_opt=0
rigctl_parse: retcode=0
rigctl_parse: called, interactive=1
```

mdblack98, 2020-12-19:
Did this used to work?Do you know how to use "git bisect" to find the offending patch?
This is a tricky section of code.
Mike W9MDB

mikaelnousiainen, 2020-12-19:
This definitely used to work, as I noticed this while using WSJT-X with Packet mode enabled and it didn't set the PKTUSB mode, but went with normal USB instead. I'll test the change.

mikaelnousiainen, 2020-12-21:
I've fixed these issues in #474

closed 2020-12-21 by mikaelnousiainen