Hamlib/Hamlib #479: g313_caps undefined reference

mdblack98, 2020-12-26:
Compiling under cygwin

  CCLD     libhamlib.la
/usr/lib/gcc/i686-pc-cygwin/10/../../../../i686-pc-cygwin/bin/ld: ../rigs/winradio/.libs/libhamlib-winradio.a(winradio.o): in function `initrigs4_winradio':
/home/pvros/dev/hamlib-4.0/rigs/winradio/winradio.c:362: undefined reference to `g313_caps'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:529: libhamlib.la] Error 1
make[1]: Leaving directory '/home/pvros/dev/hamlib-4.0/src'
make: *** [Makefile:602: all-recursive] Error 1
 

I notice the relevant code (in winradio.c) is somewhat OS dependent, and there are two versions of g313_win.c (this one is used) and g313_posix.c:

     /* Receivers with DLL only available under Windows */

#ifdef _WIN32
#ifdef __CYGWIN__
    rig_register(&g303_caps);
    rig_register(&g305_caps);
#endif
#endif

 

    /* Available on Linux and MS Windows */
#ifndef OTHER_POSIX
    rig_register(&g313_caps);
#endif
gm3zza@btinternet.com

jj1bdx, 2020-12-26:
Please take a look at this: https://github.com/Hamlib/Hamlib/commit/35c1d9c3bf07bdaa19450a7a271ca32e8979cc77#commitcomment-45436140

jj1bdx, 2020-12-27:
Issue on Linux fixed by https://github.com/Hamlib/Hamlib/commit/b9d79641a6eb1b1ecd01de314048446397665d5f. Thanks!

mdblack98, 2020-12-27:
after much toing and froing and many restarts of the laptop, I managed to get back to where I was before. I’ve downloaded the latest master and the make falls over in the same way. I’ve checked that the version of winradio.h has your latest check-in OK.

 

  CCLD     libhamlib.la

/usr/lib/gcc/i686-pc-cygwin/10/../../../../i686-pc-cygwin/bin/ld: ../rigs/winradio/.libs/libhamlib-winradio.a(winradio.o): in function `initrigs4_winradio':

/home/pvros/dev/Hamlib-master/rigs/winradio/winradio.c:362: undefined reference to `g313_caps'

collect2: error: ld returned 1 exit status

make[1]: *** [Makefile:529: libhamlib.la] Error 1

make[1]: Leaving directory '/home/pvros/dev/Hamlib-master/src'

make: *** [Makefile:602: all-recursive] Error 1

 

73 Phil.

closed 2020-12-27 by mdblack98