Hamlib/Hamlib #481: freq callback in python

mdblack98, 2020-12-29:
I would like to use the "freq_event" Callback with python. However, I
struggle getting this to work. That's how far I get:

def f():
    print("test")
    return 1

my_rig = Hamlib.Rig(Hamlib.RIG_MODEL_DUMMY)
my_rig.rig.callbacks.freq_event = f

However, I tried many things for f with ctypes but I'm not able to pass
this line (I have also tried f without ctypes, different set of
parameters, and so on) . That's the error message:

TypeError: in method 'rig_callbacks_freq_event_set', argument 2 of type
'freq_cb_t'

I have digged into the files Hamlib.py and hamlibpy_wrap.c and some SWIG
files but I can't get it working.

Any help would be appreciated.

Regards
Alex

mdblack98, 2020-12-29:
What's needed is to write a C function to pass in the Python rig object and Python function name for the callback.
The C function will do the callback setup and it's own freq event which then would call the Python function.

N0NB, 2025-07-30:
@dforsi Is this issue still relevant?

dforsi, 2025-07-30:
This isn't implemented yet.

I see that there are more callback functions that can be set by the C library:
```
int 	rig_set_dcd_callback (RIG *rig, dcd_cb_t cb, rig_ptr_t arg)
int 	rig_set_freq_callback (RIG *rig, freq_cb_t cb, rig_ptr_t arg)
int 	rig_set_mode_callback (RIG *rig, mode_cb_t cb, rig_ptr_t arg)
int 	rig_set_pltune_callback (RIG *rig, pltune_cb_t cb, rig_ptr_t arg)
int 	rig_set_ptt_callback (RIG *rig, ptt_cb_t cb, rig_ptr_t arg)
int 	rig_set_spectrum_callback (RIG *rig, spectrum_cb_t cb, rig_ptr_t arg)
int 	rig_set_vfo_callback (RIG *rig, vfo_cb_t cb, rig_ptr_t arg)
```
and I guess that the binding will be implemented the same, so the effort might be worthwhile.

I can look into this in the next days.

commit 8324c854 (2025-08-09): Implement the Python callback for freq_event

Fixes #481.
PR #1836 (2025-08-09): Feature python callbacks
This PR implements all the event callbacks in the Python bindings and in the Dummy rig:

* rig_set_freq_callback() (fixes #481)
* rig_set_dcd_callback() (NOT IN Dummy)
* rig_set_mode_callback()
* rig_set_vfo_callback()
* rig_set_pltune_callback() (NOT IN Dummy)
* rig_set_ptt_callback() (needed a fix in dummy.c)
* rig_set_spectrum_callback() (NOT IN Dummy)

In `dummy.c` I'm invoking the callbacks directly instead of using hte following functions that depend on `HAVE_PTHREAD`, while the bindings do not depend on it:
```
src/event.h:30:int rig_fire_freq_event(RIG *rig, vf [... cut]
closed 2025-08-09 by N0NB