Hamlib/Hamlib #483: Add Barrett 950

ea3iav, 2020-12-31:
I have the ini file for omnirig and all comands used to control the rig with a terminal and serial connection 

https://drive.google.com/file/d/12fMTihVP57fKVj58pVcdtzQo_vv-ozkl/view?usp=drivesdk

https://drive.google.com/file/d/1hiYPA_Y_1suzQcf1o7-caQ06cyVXURpR/view?usp=drivesdk

ea3iav, 2020-12-31:
Great! I think that buildokg from source and including this file is way beyond my skills
Thank you

mdblack98, 2020-12-31:
Let's try this one...hopefully works...there was just one command I had to change from the 2050.
https://www.dropbox.com/s/il7d8suovzum6vp/wsjtx-2.3.0-rc2-win64.exe?dl=0

You can test it with rigctl-wsjtx
rigctl-wsjtx -m 32002 -r com1 -s 9600 
You get be able to get/set frequency, get/set mode, ptt, and splitSo these commands should work in rigctl-wsjtx -- the lower case are reads, uppercase are writes.
F 14074000fM USBmM LSBmM CWmM AMmM FMT 1   (turns on PTT but no power out)tT 0tsS 1 VFOBsS 0 VFOA

mdblack98, 2020-12-31:
FYI....That dropbox link was removed....should not have been posted for public consumption...

Mike W9MDB

https://www.dropbox.com/s/il7d8suovzum6vp/wsjtx-2.3.0-rc2-win64.exe?dl=0

ea3iav, 2021-01-01:
Sorry the link is broken. Thank you for your help

mdblack98, 2021-01-01:
I sent a new link to you privately.
Please send your email address.
Mike W9MDB

ea3iav, 2021-01-03:
Sorry I could download the file and try it
Informative commands work, such as f or m
But those in capitals are not working properly 😕
[image]

mdblack98, 2021-01-03:
Try this version and if you need to run rigctl again please add "-vvvvv -Z" to get more debug info.
I had to correct the set_freq command.
https://www.dropbox.com/s/ek62wx7yg8dusqv/wsjtx-2.3.0-rc2-win64.exe?dl=0

Mike W9MDB

ea3iav, 2021-01-05:
Thank you Mike
I tried new version

Whereas info commands are ok I still have problems with invalid parameter using "write commands"
For instance, when changing mode the command line freezes 

C:\WSJT\wsjtx\bin>rigctl-wsjtx -m 32002 -r com9 -s 9600 -vvvvv -Z
2021-01-05:00:34:06.239149: rigctl Hamlib 4.1~git
Last commit was Fri Jan 01 21:13:17 2021 +0000 SHA=a59bda
2021-01-05:00:34:06.239149: Report bugs to <hamlib-developer@lists.sourceforge.n
et>

2021-01-05:00:34:06.239149: rig_init called
2021-01-05:00:34:06.239149: initrigs4_barrett: _init called
2021-01-05:00:34:06.254749: rig_register called
2021-01-05:00:34:06.254749: rig_register: rig_register (32001)
2021-01-05:00:34:06.254749: rig_register called
2021-01-05:00:34:06.254749: rig_register: rig_register (32002)
2021-01-05:00:34:06.254749: initrigs4_barrett: _init back from rig_register
2021-01-05:00:34:06.254749: rig_init: rig has VFO_A
2021-01-05:00:34:06.254749: rig_init: rig has VFO_MEM
2021-01-05:00:34:06.254749: barrett_init version 20181111.0
2021-01-05:00:34:06.254749: rig_open called
2021-01-05:00:34:06.254749: port_open called
2021-01-05:00:34:06.254749: serial_open called
2021-01-05:00:34:06.254749: serial_setup called
2021-01-05:00:34:06.254749: serial_setup: tcgetattr
2021-01-05:00:34:06.254749: serial_setup: cfsetispeed=9600,0x000d
2021-01-05:00:34:06.254749: serial_setup: cfsetospeed=9600,0x000d
2021-01-05:00:34:06.254749: serial_setup: data_bits=8
2021-01-05:00:34:06.254749: serial_setup: parity=0
2021-01-05:00:34:06.254749: serial_setup: tcsetattr TCSANOW
2021-01-05:00:34:06.254749: serial_flush called
2021-01-05:00:34:06.254749: tcflush
2021-01-05:00:34:06.317150: rig_get_vfo called
2021-01-05:00:34:06.317150: rig_get_vfo: no get_vfo
2021-01-05:00:34:06.317150: vfo_fixup: vfo=VFOA
2021-01-05:00:34:06.317150: vfo_fixup: final vfo=VFOA
2021-01-05:00:34:06.317150: rig_open: No set_vfo function rig so default vfo = V
FOA
Opened rig model 32002, '950'
2021-01-05:00:34:06.317150: Backend version: 20181111.0, Status: Alpha
2021-01-05:00:34:06.317150: rigctl_parse: called, interactive=1

Rig command: f
2021-01-05:00:34:27.018791: rigctl_parse: cmd=f(66)
2021-01-05:00:34:27.018791: rigctl_parse: cmd==0x66
2021-01-05:00:34:27.018791: rigctl_parse: vfo_opt=0
2021-01-05:00:34:27.018791: rigctl_parse: debug1
2021-01-05:00:34:27.018791: rigctl_parse: debug5
2021-01-05:00:34:27.018791: rigctl_parse: debug10
2021-01-05:00:34:27.018791: rigctl_parse: vfo_opt=0
2021-01-05:00:34:27.018791: rig_get_freq called vfo=currVFO
2021-01-05:00:34:27.018791: vfo_fixup: vfo=currVFO
2021-01-05:00:34:27.018791: vfo_fixup: Leaving currVFO alone
2021-01-05:00:34:27.018791: get_cache_freq: vfo=currVFO, freq=0
2021-01-05:00:34:27.018791: elapsed_ms: start = 0,0
2021-01-05:00:34:27.018791: rig_get_freq: cache check age=1000000ms
2021-01-05:00:34:27.018791: rig_get_freq: cache miss age=1000000ms, cached_vfo=N
one, asked_vfo=currVFO
2021-01-05:00:34:27.018791: vfo_fixup: vfo=VFOA
2021-01-05:00:34:27.018791: vfo_fixup: final vfo=VFOA
2021-01-05:00:34:27.018791: rig_get_freq: no set_vfo so vfo=VFOA
2021-01-05:00:34:27.018791: barrett_get_freq: vfo=VFOA
2021-01-05:00:34:27.018791: barrett_transaction: cmd=IR
2021-01-05:00:34:27.018791: rig_flush: called for serial device
2021-01-05:00:34:27.018791: serial_flush called
2021-01-05:00:34:27.018791: tcflush
2021-01-05:00:34:27.018791: write_block called
2021-01-05:00:34:27.081191: write_block(): TX 3 bytes
[... 52 lines cut ...]
2021-01-05:00:35:06.636350: write_block called
2021-01-05:00:35:06.698750: write_block(): TX 11 bytes
2021-01-05:00:35:06.698750: 0000 4f 52 31 34 30 37 34 30 30 30 0d
OR14074000.
2021-01-05:00:35:06.698750: read_string called, rxmax=64
2021-01-05:00:35:06.729950: read_string(): RX 6 characters
2021-01-05:00:35:06.729950: 0000 13 45 30 0d 0a 11
.E0...
2021-01-05:00:35:06.729950: barrett_transaction: resultlen=6
2021-01-05:00:35:06.729950: barrett_transaction: retval=6
2021-01-05:00:35:06.729950: 0000 13 45 30 0d 0a 11
.E0...
2021-01-05:00:35:06.729950: barrett_transaction: removing xoff char
2021-01-05:00:35:06.729950: barrett_transaction: removing xon char
2021-01-05:00:35:06.729950: barrett_transaction: setting result
2021-01-05:00:35:06.729950: 0000 45 30
E0
2021-01-05:00:35:06.729950: barrett_transaction: returning result=E0
2021-01-05:00:35:06.729950: barrett950_set_freq: Expected OK, got 'E0'
2021-01-05:00:35:06.729950: rigctl_parse: vfo_opt=0
set_freq: error = Invalid parameter
2021-01-05:00:35:06.729950: rigctl_parse: retcode=-1
2021-01-05:00:35:06.729950: rigctl_parse: called, interactive=1

Rig command: 2021-01-05:00:35:06.729950: rigctl_parse: cmd==0x0a

mdblack98, 2021-01-05:
Another idea -- I could use say channels 441-450 as a 10-band channel memory.  So once the frequencies are set they rarely need to change unless you change modes from FT8 to FT4 for example which have different frequencies.  So you could band-hop on one mode without having to write to memory.
I kind of like that idea as it minimizes the # of channels used and minimizes EEPROM writes.
So I'm going to implement that.

Mike W9MDB

mdblack98, 2021-01-05:
So it appears that the only way to set frequency on the 950 is to program a memory channel and select it.
I don't suppose you have a schematic for the 950?  Writing to a memory location normally means writing to an EEPROM.  And there is a limited number of write cycles an EEPROM can handle depending on the specific EEPROM used and runs 100,000 to 1,000,000 or so.
If I program it to use memory location 1 and let's say you do 20 band changes per day that would be 5,000 days or a bit over 13 years and the EEPROM is likely to fail.I could also use 2021 as a base year and change the channel# used every year so with 400 channels that would last 400 years or more.

ea3iav, 2021-01-05:
Hi! The way Frequency control goes for instance in omnirig is to reprogram or change the frequency on the default channel. No need to even name the channel. 
So you connect the radio and send commands. They can be strings or single commands.
Here are the commands 
[](https://drive.google.com/file/d/1hiYPA_Y_1suzQcf1o7-caQ06cyVXURpR/view?usp=drivesdk)

mdblack98, 2021-01-05:
If you always use the default channel you will burn out the EEPROM a lot sooner.  That's a fairly old rig and quite possible the EEPROM may not have more then 100,000 writes to any one location.  You'd have to look up the EEPROM model number.
My solution of using channels 441-450 for the 10 bands will minimize the # of writes to EEPROM.  If you work FT8 all day for example you will never have to write to EEPROM.Switching to FT4 will require 1 write and then switching back to FT8 will require another write.  So only when you have to use a different frequency on a band will EEPROM get updated.
Mike W9MDB

ea3iav, 2021-01-05:
Hi! It is a 27C512 Eprom
I am thinking about doing some backups... as I use this method of constant rewrite with my nextion interface

https://youtu.be/3PT1YqP62LU

mdblack98, 2021-01-05:
That's an EPROM and not EEPROM.So I don't think that's where it keeps channel info.Is there another memory device in there?

 
 

https://youtu.be/3PT1YqP62LU

mdblack98, 2021-01-11:
The AT28C64-25SI is still available
https://www.digikey.com/en/products/detail/microchip-technology/AT28C64-25SI/398949

Did you get the last link I sent you to try?
Please contact me off list at mdblack98@yahoo.com

closed 2021-02-10 by mdblack98

MrSon888, 2022-01-07:
Hi, how did this go in the end? Have you considered adding the Barrett 4050?

mdblack98, 2022-01-07:
It all went quite well.  Hopefully the 4050 doesn't require channel writes to do simple frequency changes.