Hamlib/Hamlib #492: Quisk and rigctl hamlib 3.3

dg1vs, 2021-01-07:
I have a problem with hamlib, accessing a so called virtual serial port open from the software Quisk. Quisk opens 2 virtual serial ports /tmp/QuiskTTY0 and /tmp/QuiskTTY1. Both can be accessed via minicom and I was able to trigger request an getting the answers. The command-set is based on the FlexRadio ZZxx command set. So ZZFA works this way and I was able to get the frequency of VFO A back. 
Trying to access the virtual serial port via rigctl (“rigctl –model=2301 –rig-file=/tmp/QuiskTTY0”)  is resulting in rig_open : error = IO error.
To be honest I have no idea where the problem is. Maybe someone can point me in the right direction. In case, I will add extra printf in the source-code if needed. I'm running hamlib 3.3

Kind regards Karsten
73 de dg1vs

mdblack98, 2021-01-09:
Check you have permissions on /tmp/QuiskTTY1Maybe try "sudo rigctl"
Also...can you upgrade to hamlib 4.0 and test that?

Mike W9MDB

dg1vs, 2021-01-10:
Hi, thanks for the answers.
The permissions are ok, otherwise a connection with Minicom wouldn’t work.
At the beginning I tried with hamlib 4.0, with the same problem.

mdblack98, 2021-01-10:
Can you please test current hamlib with rigctl and "-vvvvv -Z" added to the command line and se the output?
Mike W9MDB

dg1vs, 2021-01-10:
just the 3.3 version for the 4.0 version i need to recompile.. needs some time... but maybe the it helps:
par_open: PPSETMODE "/tmp/QuiskTTY0": Inappropriate ioctl for device**

rigctl --model=2301 --rig-file=/tmp/QuiskTTY0 -vvvvv -Z
2021-01-10:16:00:06.436942: rigctl, Hamlib 3.3
2021-01-10:16:00:06.437017: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-01-10:16:00:06.437031: rig_init called
2021-01-10:16:00:06.441977: initrigs3_flexradio called
2021-01-10:16:00:06.442016: rig_register called
2021-01-10:16:00:06.442040: rig_register: rig_register (2301)
2021-01-10:16:00:06.442057: rig_register called
2021-01-10:16:00:06.442070: rig_register: rig_register (2303)
2021-01-10:16:00:06.442087: rig_register called
2021-01-10:16:00:06.442099: rig_register: rig_register (2304)
2021-01-10:16:00:06.442151: rig_open called
2021-01-10:16:00:06.442171: port_open called
2021-01-10:16:00:06.442185: par_open called
**2021-01-10:16:00:06.442270: par_open: PPSETMODE "/tmp/QuiskTTY0": Inappropriate ioctl for device**
rig_open: error = IO error

mdblack98, 2021-01-10:
You have a Flex radio?
There's a difference in model numbers between 3.0 and 4.0
So you need to do "rigctl -l | grep FlexOr something that identifies your rig for each.
It's trying to open a parallel port which is wrong.
Mike

dg1vs, 2021-01-10:
Yes and no. I have a FPGA based SDR and Quisk as Software on Linux side. Quisk offers the opportunity for a **CAT virtual serial port**. The port in my case is linked to a kind of terminal. I can open the virtual serial port with minicom and python with serial commands and then it works with Quisk with a set of ZZxx commands (e.g “ZZFA;” returns the frequency) 
The manual says use a kind of flexradio… 236 or 2301 for the 3.3 version.

dg1vs, 2021-01-10:
What I'm looking for is a rig which opens a serial port and speaks Flex commands, like ZZFA
Thanks, and kind regards Karsten

mdblack98, 2021-01-10:
The rig you select is a parallel port rig.
Try -m 236
Mike W9MDB

dg1vs, 2021-01-10:
Already done, i got a different problem

rigctl --model=236 --rig-file=/tmp/QuiskTTY0 -vvvvv -Z
2021-01-10:17:59:03.171685: rigctl, Hamlib 3.3
2021-01-10:17:59:03.171738: Report bugs to <hamlib-developer@lists.sourceforge.net>

2021-01-10:17:59:03.171747: rig_init called
2021-01-10:17:59:03.171770: initrigs3_kenwood called
2021-01-10:17:59:03.171793: rig_register called
2021-01-10:17:59:03.171818: rig_register: rig_register (213)
2021-01-10:17:59:03.171830: rig_register called
2021-01-10:17:59:03.171855: rig_register: rig_register (201)
2021-01-10:17:59:03.171873: rig_register called
2021-01-10:17:59:03.171884: rig_register: rig_register (225)
2021-01-10:17:59:03.171920: rig_register called
2021-01-10:17:59:03.171930: rig_register: rig_register (203)
2021-01-10:17:59:03.171944: rig_register called
2021-01-10:17:59:03.171967: rig_register: rig_register (204)
2021-01-10:17:59:03.171981: rig_register called
2021-01-10:17:59:03.171994: rig_register: rig_register (216)
2021-01-10:17:59:03.172010: rig_register called
2021-01-10:17:59:03.172033: rig_register: rig_register (224)
2021-01-10:17:59:03.172047: rig_register called
2021-01-10:17:59:03.172059: rig_register: rig_register (205)
2021-01-10:17:59:03.172070: rig_register called
2021-01-10:17:59:03.172079: rig_register: rig_register (207)
2021-01-10:17:59:03.172091: rig_register called
2021-01-10:17:59:03.172105: rig_register: rig_register (209)
2021-01-10:17:59:03.172119: rig_register called
2021-01-10:17:59:03.172131: rig_register: rig_register (210)
2021-01-10:17:59:03.172153: rig_register called
2021-01-10:17:59:03.172178: rig_register: rig_register (222)
2021-01-10:17:59:03.172211: rig_register called
2021-01-10:17:59:03.172224: rig_register: rig_register (214)
2021-01-10:17:59:03.172239: rig_register called
2021-01-10:17:59:03.172251: rig_register: rig_register (230)
2021-01-10:17:59:03.172264: rig_register called
2021-01-10:17:59:03.172277: rig_register: rig_register (221)
2021-01-10:17:59:03.172300: rig_register called
2021-01-10:17:59:03.172335: rig_register: rig_register (229)
2021-01-10:17:59:03.172361: rig_register called
2021-01-10:17:59:03.172386: rig_register: rig_register (238)
2021-01-10:17:59:03.172412: rig_register called
2021-01-10:17:59:03.172446: rig_register: rig_register (202)
2021-01-10:17:59:03.172471: rig_register called
2021-01-10:17:59:03.172484: rig_register: rig_register (211)
2021-01-10:17:59:03.172507: rig_register called
2021-01-10:17:59:03.172534: rig_register: rig_register (206)
2021-01-10:17:59:03.172550: rig_register called
2021-01-10:17:59:03.172574: rig_register: rig_register (208)
2021-01-10:17:59:03.172611: rig_register called
2021-01-10:17:59:03.172629: rig_register: rig_register (215)
2021-01-10:17:59:03.172647: rig_register called
2021-01-10:17:59:03.172670: rig_register: rig_register (226)
2021-01-10:17:59:03.172682: rig_register called
2021-01-10:17:59:03.172696: rig_register: rig_register (217)
2021-01-10:17:59:03.172711: rig_register called
2021-01-10:17:59:03.172726: rig_register: rig_register (233)
2021-01-10:17:59:03.172749: rig_register called
2021-01-10:17:59:03.172764: rig_register: rig_register (220)
2021-01-10:17:59:03.172781: rig_register called
2021-01-10:17:59:03.172805: rig_register: rig_register (223)
2021-01-10:17:59:03.172826: rig_register called
2021-01-10:17:59:03.172842: rig_register: rig_register (227)
2021-01-10:17:59:03.172911: rig_register called
2021-01-10:17:59:03.172925: rig_register: rig_register (234)
2021-01-10:17:59:03.172943: rig_register called
2021-01-10:17:59:03.172959: rig_register: rig_register (231)
2021-01-10:17:59:03.172974: rig_register called
2021-01-10:17:59:03.172988: rig_register: rig_register (239)
2021-01-10:17:59:03.173006: rig_register called
2021-01-10:17:59:03.173031: rig_register: rig_register (237)
2021-01-10:17:59:03.173058: rig_register called
2021-01-10:17:59:03.173074: rig_register: rig_register (228)
2021-01-10:17:59:03.173089: rig_register called
2021-01-10:17:59:03.173102: rig_register: rig_register (219)
2021-01-10:17:59:03.173125: rig_register called
2021-01-10:17:59:03.173150: rig_register: rig_register (232)
2021-01-10:17:59:03.173176: rig_register called
2021-01-10:17:59:03.173209: rig_register: rig_register (236)
2021-01-10:17:59:03.173243: rig_register called
2021-01-10:17:59:03.173277: rig_register: rig_register (240)
2021-01-10:17:59:03.173358: kenwood_init called
2021-01-10:17:59:03.173379: kenwood_init: if_len = 37
2021-01-10:17:59:03.173416: rig_open called
2021-01-10:17:59:03.173443: port_open called
2021-01-10:17:59:03.173468: network_open called
2021-01-10:17:59:03.174360: network_open: cannot get host "/tmp/QuiskTTY0": Unknown error
rig_open: error = Invalid configuration

mdblack98, 2021-01-10:
Try the TS-2000 -- if your rig is compatible with Flex than it should also be compatible with TS-2000 commands.
Is there a CAT protocol document for Quisk?
Mike W9MDB

dg1vs, 2021-01-10:
Bingo! Thanks a lot!

> Is there a CAT protocol document for Quisk?
No yet. Until now Quisk supports the following commands:
ZZAG
ZZAI
ZZFA
ZZFB
ZZIF
ZZMD
ZZMU
ZZPS
ZZRS
ZZSP
ZZSW
ZZTX
ZZVE

See also [https://pypi.org/project/quisk/] and [http://james.ahlstrom.name/quisk/]
There is already  a diskussion ongoing to extend the command set.

Kind regrads
Karsten
73 de dg1vs

dg1vs, 2021-01-10:
Thanks a lot

closed 2021-01-10 by dg1vs

mdblack98, 2021-01-10:
If you give 4.0 hamlib a try the PowerSDR rig may work.Not all the ZZ commands are implemented but many of the ones you show aren't any different than the Kenwood command

  2048  FlexRadio/ANAN         PowerSDR/Thetis         20201231.0      Stable      RIG_MODEL_POWERSDR

dg1vs, 2021-01-10:
Cool, I will test it.
73 de dg1vs

Mit freundlichen Grüßen
Dr. Karsten Schmidt

mdblack98, 2021-01-10:
If you have a problem we may have to add Quisk as a new rig.
Mike W9MDB

dg1vs, 2021-01-11:
Hi Mike,
trying the new hamlib 4.0 I ran in problems, with both the rig 2048
PowerSDr and rig 2014 TS-2000.
Looks like rigctl is trying to open a network port, but a serial port
is needed.

Karsten
73 de dg1vs

rigctl --model=2048 --rig-file=/tmp/QuiskTTY0 -vvvvv -Z
2021-01-11:20:05:20.082306: rigctl Hamlib 4.0
Last commit was Fri Dec 25 02:30:30 2020 +0000 SHA=519f82
2021-01-11:20:05:20.082364: Report bugs to <
hamlib-developer@lists.sourceforge.net>

2021-01-11:20:05:20.082424: rig_init called
2021-01-11:20:05:20.082507: initrigs4_kenwood called
2021-01-11:20:05:20.082538: rig_register called
2021-01-11:20:05:20.082564: rig_register: rig_register (2012)
2021-01-11:20:05:20.082585: rig_register called
2021-01-11:20:05:20.082597: rig_register: rig_register (2013)
2021-01-11:20:05:20.082612: rig_register called
2021-01-11:20:05:20.082628: rig_register: rig_register (2001)
2021-01-11:20:05:20.082648: rig_register called
2021-01-11:20:05:20.082666: rig_register: rig_register (2025)
2021-01-11:20:05:20.082697: rig_register called
2021-01-11:20:05:20.082717: rig_register: rig_register (2003)
2021-01-11:20:05:20.082731: rig_register called
2021-01-11:20:05:20.082747: rig_register: rig_register (2004)
2021-01-11:20:05:20.082760: rig_register called
2021-01-11:20:05:20.082773: rig_register: rig_register (2016)
2021-01-11:20:05:20.082786: rig_register called
2021-01-11:20:05:20.082798: rig_register: rig_register (2024)
2021-01-11:20:05:20.082814: rig_register called
2021-01-11:20:05:20.082829: rig_register: rig_register (2005)
2021-01-11:20:05:20.082845: rig_register called
2021-01-11:20:05:20.082859: rig_register: rig_register (2007)
2021-01-11:20:05:20.082873: rig_register called
2021-01-11:20:05:20.082886: rig_register: rig_register (2009)
2021-01-11:20:05:20.082900: rig_register called
2021-01-11:20:05:20.082913: rig_register: rig_register (2010)
2021-01-11:20:05:20.082926: rig_register called
2021-01-11:20:05:20.082938: rig_register: rig_register (2022)
2021-01-11:20:05:20.082951: rig_register called
2021-01-11:20:05:20.082965: rig_register: rig_register (2014)
2021-01-11:20:05:20.082979: rig_register called
2021-01-11:20:05:20.082994: rig_register: rig_register (2030)
2021-01-11:20:05:20.083009: rig_register called
2021-01-11:20:05:20.083022: rig_register: rig_register (2021)
2021-01-11:20:05:20.083039: rig_register called
2021-01-11:20:05:20.083054: rig_register: rig_register (2029)
2021-01-11:20:05:20.083068: rig_register called
2021-01-11:20:05:20.083084: rig_register: rig_register (2043)
2021-01-11:20:05:20.083099: rig_register called
2021-01-11:20:05:20.083114: rig_register: rig_register (2044)
2021-01-11:20:05:20.083129: rig_register called
2021-01-11:20:05:20.083145: rig_register: rig_register (2045)
2021-01-11:20:05:20.083160: rig_register called
2021-01-11:20:05:20.083176: rig_register: rig_register (2047)
2021-01-11:20:05:20.083191: rig_register called
2021-01-11:20:05:20.083207: rig_register: rig_register (2038)
2021-01-11:20:05:20.083222: rig_register called
2021-01-11:20:05:20.083237: rig_register: rig_register (2002)
2021-01-11:20:05:20.083252: rig_register called
2021-01-11:20:05:20.083265: rig_register: rig_register (2011)
2021-01-11:20:05:20.083276: rig_register called
2021-01-11:20:05:20.083292: rig_register: rig_register (2006)
2021-01-11:20:05:20.083308: rig_register called
2021-01-11:20:05:20.083323: rig_register: rig_register (2008)
2021-01-11:20:05:20.083338: rig_register called
[... 28 lines cut ...]
2021-01-11:20:05:20.083760: rig_register: rig_register (2032)
2021-01-11:20:05:20.083773: rig_register called
2021-01-11:20:05:20.083783: rig_register: rig_register (2036)
2021-01-11:20:05:20.083797: rig_register called
2021-01-11:20:05:20.083810: rig_register: rig_register (2048)
2021-01-11:20:05:20.083823: rig_register called
2021-01-11:20:05:20.083835: rig_register: rig_register (2040)
2021-01-11:20:05:20.083848: rig_register called
2021-01-11:20:05:20.083861: rig_register: rig_register (2041)
2021-01-11:20:05:20.083878: rig_register called
2021-01-11:20:05:20.083890: rig_register: rig_register (2046)
2021-01-11:20:05:20.083934: kenwood_init called, version
20201214/2020161217
2021-01-11:20:05:20.083952: kenwood_init: if_len = 37
2021-01-11:20:05:20.083965: rig_open called
2021-01-11:20:05:20.084001: rig_open: using network address
/tmp/QuiskTTY0
2021-01-11:20:05:20.084017: port_open called
2021-01-11:20:05:20.084028: network_open called
2021-01-11:20:05:20.084043: network_open version 1.0
2021-01-11:20:05:20.084061: network_open: hoststr=/tmp/QuiskTTY0,
portstr=
2021-01-11:20:05:20.085058: network_open: cannot get host
"/tmp/QuiskTTY0": Unknown error
rig_open: error = Invalid configuration

mdblack98, 2021-01-11:
Do a pull and it should be fixed now.  Was treating /tmp as a hostname.
Mike W9MDB