Hamlib/Hamlib #499: rig.h missing

tony1tf, 2021-01-09:
Trying to compile Minos2 and got an error:
In file included from ../../mqt/MqtBase/serialdata.cpp:1:
../../mqt/MqtBase/serialdata.h:7:10: fatal error: hamlib/rig.h: No such file or directory
 #include "hamlib/rig.h"

I havn't been able to find this file - is this a bug in this build? - help please

mdblack98, 2021-01-09:
Sounds like you didn't do a "make install"
Mike W9MDB

N0NB, 2021-01-09:
* On 2021 09 Jan 11:23 -0600, tony1tf wrote:
> Trying to compile Minos2 and got an error:
> In file included from ../../mqt/MqtBase/serialdata.cpp:1:
> ../../mqt/MqtBase/serialdata.h:7:10: fatal error: hamlib/rig.h: No such file or directory
>  #include "hamlib/rig.h"

There is a subtle difference between using angle brackets and double
quotes in an #include directive.  In this case,

#include <hamlib/rig.h>

should always be used so the normal system include directories are
searched.

The use of double quotes implies a local header file that is often
located in the same directory as the source file.  Some variables are
set in the Hamlib build system to expand this behavior to to search the
Hamlib source tree but this is not portable to third party software
building against a properly installed Hamlib.

> I havn't been able to find this file - is this a bug in this build? - help please

It should be found in /usr/include/hamlib/rig.h if installed as a system
package in most Linux distributions.  If Hamlib was compiled and
installed locally, then it's probably in /usr/local/include/rig.h.  As
each of these are considered system directories, the C preprocessor
should find it in either location.

73, Nate

-- 

"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."

Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

tony1tf, 2021-01-09:
Hi Mike
Thanks for that astute observation. I'm a bit of a newbie when it comes to these big compiles. I backtracked through my commands, and you are absolutely correct. After I had sent in the issue, I did find the file, and I was just in the process of wondering why the compilation didn't find it. My computer is happily compiling Minos2 now. Mind you, it's an awful lot of code just to get the latest Minos logging program onto the RaspberryPi desktop on an old mini-pc. I dont really need 99% of it, so it should be possible somewhere to cut down the amount of code.
73
Tony, G3OVH

closed 2021-01-09 by mdblack98